mstrens/oXs_on_RP2040 #75: DIY5104 FrSky X20S FBUS Protocol

Satcomix, 2023-04-19:
Hello Mstrens,
I just installed the new test version 2.2.2 to check the DIY5104 sensor.
System FrSky X20S with FBUS protocol.
I think it's either the query speed (too slow) or I've activated too many sensors, I almost only get SENSOR LOST.
With the other sensors I can see the query point very well, but not with the DIY5104 GPS Cumulative Distance.
Greeting,
Torsten

mstrens, 2023-04-19:
I reduce the priority of some sport/fbus fields.
Perhaps this will solve the issue?
It is in 2.2.3

Satcomix, 2023-04-19:
Hello Mstrens,
With version 2.2.3, my handheld no longer receives any information from the DIY5104. The Display show ---
Reducing the priority didn't work.
br,
Torsten

mstrens, 2023-04-19:
What are the sensors that are activated?
If you do not define I2C pins (and do not activate FVP/FVN) , do you still have the issue?

Satcomix, 2023-04-19:
All possible Sensors are activate for the tests.
processing cmd

Cmd to execute:   

Version = 2.2.3 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    4  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255    6    7 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Fbus(Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 353 , 448 , -2735
     Gyro offsets         X, Y, Z = 12 , -161 , -196
Aispeed sensor is detected using MS4525
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Foreseen GPS type is Ublox (configured by oXs) :GPS is detected and has a fix
Failsafe uses predefined values
     Chan 1...4  = 992 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
processing cmd

Cmd to execute:   FV

GPS Latitude = xx.xxxxxxxxxx degree
GPS Longitude = x.xxxxxxxxx degree
GPS Groundspeed = 4 cm/s
GPS Heading = 278.760000 degree
GPS Altitude = 4701 cm
GPS Num sat. = 118
GPS Date J M A = 19 4 23 
GPS Time H M S = 11 1 37 
GPS Pdop = 134 
GPS Home bearing = 207 degree
GPS Home distance = 0 m
Volt 1 = 1576 mVolt
Current (Volt 2) = 1669 mA
Volt 3 = 1614 mVolt
Volt 4 = 269 mVolt
Capacity (using current) = 34 mAh
Vspeed = -7 cm/s
Baro Rel altitude = -33 cm
Pitch = -9 degree
Roll = -1 degree
RPM = 998 Hertz
Ads 1 1 = 119 mVolt
Ads 1 2 = 119 mVolt
Ads 1 3 = 119 mVolt
Ads 1 4 = 119 mVolt
Ads 2 1 = 118 mVolt
Ads 2 2 = 119 mVolt
Ads 2 3 = 118 mVolt
Ads 2 4 = 119 mVolt
Airspeed = 80 cm/s
Compensated Vspeed = -27 cm/s

Satcomix, 2023-04-19:
If I disable I2C bus =255
Sometimes a query pops up on the DIY5104, but most of the time it doesn't, and i get sensor lost.

mstrens, 2023-04-19:
Do you have the issue also if you use FVP command. I expect not.
I think I understand the issue.
Currently, the cumulative distance is made available only when a new value is calculated.
This happens only if the GPS position change.
I will adapt this.

Satcomix, 2023-04-19:
When I enter FVP, same problem.
I think you're right about the change in the GPS values, since my GPS is lying quietly on the table right now.

mstrens, 2023-04-19:
I put a version 2.2.4 to fix it (I hope)

Satcomix, 2023-04-19:
I think, it works. I activate I2C again. No Sensor lost.
DIY5103 and 5104 have the same values.
processing cmd

Cmd to execute:   

Version = 2.2.4 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    4  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255    6    7 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Fbus(Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 353 , 448 , -2735
     Gyro offsets         X, Y, Z = 12 , -161 , -196
Aispeed sensor is detected using MS4525
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Foreseen GPS type is Ublox (configured by oXs) :GPS is detected and has a fix
Failsafe uses predefined values
     Chan 1...4  = 992 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
processing cmd

Cmd to execute:   FV

GPS Latitude = xx.xxxxxxxxxx degree
GPS Longitude = x.xxxxxxxxx degree
GPS Groundspeed = 8 cm/s
GPS Heading = 331.920000 degree
GPS Altitude = 4848 cm
GPS Num sat. = 120
GPS Date J M A = 19 4 23 
GPS Time H M S = 11 27 21 
GPS Pdop = 136 
GPS Home bearing = 266 degree
GPS Home distance = 1 m
Volt 1 = 1576 mVolt
Current (Volt 2) = 1670 mA
Volt 3 = 1614 mVolt
Volt 4 = 266 mVolt
Capacity (using current) = 14 mAh
Vspeed = -4 cm/s
Baro Rel altitude = 37 cm
Pitch = -8 degree
Roll = -1 degree
RPM = 998 Hertz
Ads 1 1 = 118 mVolt
Ads 1 2 = 118 mVolt
Ads 1 3 = 118 mVolt
Ads 1 4 = 118 mVolt
Ads 2 1 = 118 mVolt
Ads 2 2 = 118 mVolt
Ads 2 3 = 118 mVolt
Ads 2 4 = 118 mVolt
Airspeed = 20 cm/s
Compensated Vspeed = -18 cm/s
Gps cumulative distance = 0

closed 2023-04-19 by Satcomix