mstrens/oXs_on_RP2040 #77: Spektrum SRXL2 Protocol

Satcomix, 2023-05-03:
Hello Mstrens,
I noticed something about the SRXL2 protocol regarding the RGB LED.
When using for the first time without a connection to the handheld (off), the LED lights up/flashes red, OK
With SETFAILSAIFE switched on, it lights up/flashes blue after switching off the transmitter, OK.
But when running, **it should be solid/flashing green**. Although 8 PWM channels are transmitted with SRXL2 (SBUS+TLM) and telemetry is received, the **LED flashes blue**.
Greeting,
Torsten

Satcomix, 2023-05-05:
Hello Mstrens,
I think it would be advisable to exchange the current version **2.1.4-Main** for **2.2.11-Test**, since there are problems with 2.1.4 with the offset from 125Mhz to 133Mhz.
In version **2.2.12 2 there are still problems** with the RGB LED and with the spectrum SRXL2 the problem with the PWM values.
**Thank you for your effort and work.**
Greeting,
Torsten

mstrens, 2023-05-05:
I just updated main branch from test branch( 2.2.13)

I also made a new version in test 2.2.14 in order to fix the issue with led color in srxl2 (to be tested).

The issue about PWM is that SRXL2 does not use the same conversion rules for PWM.
E.g. in SRXL2 the mid position is coded as 1024 (decimal value) while in Sbus (=Frsky) it is at (172+1811)/2= 911.
I can add a special conversion to convert SRXL2 value to sbus value (e.g 1024 to 911).
Still the values you noticed in you first post does not seems 100% OK.

mstrens, 2023-05-05:
It is not 100% ok because if I do following linear mapping:
342 => -100% => 1000us
1697 => +100% => 2000us
Then 1024 does not gives exactly 0% and it gives about 1503us instead of 1500usec

Satcomix, 2023-05-05:
> It is not 100% ok because if I do following linear mapping: 342 => -100% => 1000us 1697 => +100% => 2000us Then 1024 does not gives exactly 0% and it gives about 1503us instead of 1500usec

Hello Mstrens,
Thanks for the answer.
I would suggest that the center is exactly 1500us and plus/minus 100% is slightly above like FrSky 988us and 2012us.
I will make some Tests with Version 2.2.14
br,
Torsten

Satcomix, 2023-05-05:
In version 2.2.14 the LED is green now,ok. 
But it also lights up green when I **turn off the transmitter**, usually with SETFAILSAFE it lights up blue.

Satcomix, 2023-05-05:
Version 2.2.14 SRXL2 Spektrum
CH1 -100%
CH6 -100%
CH8 +100%
Rest 0%
Failsafe uses predefined values
     Chan 1...4  =  1094  1520  1520  1520
     Chan 5...8  =  1520  1094  1520  1520
     Chan 9...12 =   881   881   881   881
     Chan 13...16=   881   881   881   881

Cmd to execute:   PWM
PWM values us (sbus) 1... 8   1094(  342)  1520( 1024)  1520( 1024)  1520( 1024)  1520( 1024)  1094(  342)  1520( 1024)  1946( 1706)
PWM values us (sbus) 9...16    881(    0)   881(    0)   881(    0)   881(    0)   881(    0)   881(    0)   881(    0)   881(    0)

mstrens, 2023-05-05:
Now 2.2.15.
I hope PWM values are better.
I do not see why led stay green when Tx is off.
Can you try this version. I added a debug msg.
When it is connected, it should print "0" on the PC.
When failsafe is activated (no msg received from receiver), it should print "2".
Question: it could be that when receiver loose his connection with Tx that it continues to generate channels values on the SRXL2 bus. if so, oXs can't detect the failsafe. This could be checked making capture of the frame with a logic analyser (or adding some more debug message to oXs)

Satcomix, 2023-05-05:
Debug Message: Can not open Terminal!
0
0
0
0
0
0
But I can measure:
-100%=1093us
+-0% = 1408us
+100%=1719us

mstrens, 2023-05-06:
I made some changes for srxl2 (same version nr) in order to:
- get better PWM values 
- print a message each time oXs get a frame with rc channels values.
in SRXL2 there are 2 types of frames with rc channels values:
0 = normal
1 = failsafe.
Could you say what happens when the receiver lost the connection with handset: Does oXs get frame with failsafe value (1) or does it stop receiving frame (no debug msg)

Satcomix, 2023-05-06:
Hello Mstrens,
Unfortunately I can't do any tests today because I don't have access to the equipment at the moment.
But what I saw yesterday evening:
Transmitter is on and oXs_RP2040 with connection to SRXL2 receiver: LED green, ok.
Transmitter switched off: LED green, not ok
Receiver separated from oXs_RP2040: LED blue??? i think red???
br,
Torsten

Satcomix, 2023-05-07:
Hello Mstrens,
I did some tests with Version 2.2.15-2 and the Spektrum SRXL2 protocol today.
The PWM values ​​on the GPIO PWM and SBUS_out now match and show values ​​from **1000us/-100% to 2000us/+100%, 0%=1500us.**
When displaying the RGB LED under **SETFAILSAFE** I get the following displays:
Transmitter off and Receiver connected to oXs SRXL2: **LED red**
Transmitter on and Receiver at oXs: **LED green**
After normal function power off the Transmitter and Receiver is still connected: **LED green???**
Transmitter off and disconnect Receiver from oXs: **LED blue**
**Debug messages:**
Transmitter on and receiver connected to oXs: **fs=0**
Transmitter off and Receiver connected to oXs: **fs=1**
br,
Torsten

mstrens, 2023-05-08:
I put a version 2.3.0 on github test.
Hoping it solves the led issue for srxl2

Satcomix, 2023-05-08:
Hello Mstrens,
I make some tests with Version 2.3.0_TEST
No Debug messages appear.
Now everything works as it should with the SRXL2 protocol, whether PWM, SBUS_out, RGB LED.
After power down the transmitter, now the LED get blue with SETFAILSAFE.
PWM Values:
1000us= -100%=Terminal 1000us  (191)
1500us=     0% =Terminal 1500us  (992)
2000us=+100%=Terminal 2000us (1792)
Thank you for your work and effort.
br,
Torsten

Satcomix, 2023-11-21:
Hello Mstrens,
After Arnd noticed something on RCG, I did some tests with the DX8 and SPM4651T. I then asked Pierre777 to also carry out tests with the SRXL2 protocol.
Same result!
I only receive Airspeed and Altitude, and Arnd and Pierre mostly only receive
TRSS 31
FdeA 7
FLss 0
Hold 0
Something must have changed.
greetings,
Torsten

mstrens, 2023-11-21:
@Satcomix 
So you do not receive Vspeed?
And altitute: is it gps altitude or baro altitude?
If you use the FVP command do you get more?
Note: I checked the code and noticed that when Volt1 or Volt2 where defined but not RPM, TMP1 and TMP2, then Volt, current and capacity where not sent. I fixed this in 2.10.10

Satcomix, 2023-11-21:
Hello Mstrens,
I know what the problem is with some radios:
The firmware only allows certain IDs.
I just made my own build (renamed tele_device_esc into pbox) with Ver.2.10.10 and now see more telemetry data.
With srxl2.cpp only vario_s, gps_binary, airspeed, esc and RX_MAH are created.
I have just created the PBOX and now I also see a Voltage1/ Voltage2 and Capacity1+2.

mstrens, 2023-11-21:
does it mean that I do not have to make more changes to oXs code?

Satcomix, 2023-11-21:
What I also just noticed is that when I run FVP, the telemetry transmission stops every 1-2 seconds.
Buffer overflow in the srxl2 protocol?

mstrens, 2023-11-21:
I do not know if there is a buffer overflow.
oXs provides a frame to the Rx each time the Rx asks for a frame.
It could be that the Rx request to often.

Satcomix, 2023-11-21:
Hello Mstrens,
With my old DX8 Gen1 DSMX SRXL2 protocol I can read the following values.
PWM CH1-8
Volt1,
Volt2,
Volt3,
RPM,
airspeed,
Baro Altitude,
So the SRXL2 Protocol should work.
Of course, the oXs_RP2040 user may have to create his own build if the transmitter has older firmware and only other telemetry IDs/sensors are known in this firmware.
Maybe there is an option to expand the 5 sensors in the srxl2.cpp, e.g.
FlightPackV,
PowerBox/PBox
GForce,
GPS BCD,
Greetings,
Torsten

Satcomix, 2023-11-23:
Hello Mstrens,
I have a few questions about the SRXL2 Protocol.
How can I create more sensors in the priority list regarding min and max pooling?
The SRXL2 TX buffer only has 5 (0-4) cases created.
The case 4 RX MAH is not activated, if I comment it out (activate) the telemetry no longer works.
What do I have to do and how do I best proceed?
Since I own a DX8Gen1, I wanted to at least try the values ​​for:
RxV
Volts
Amps
Temp
RPM
PBox
GForce
GPS (BCD, i think)
to show me.
I get the values ​​for AirSpeed ​​and Altitude.
I was also able to activate the PBox and see RPM there as Battery1(max 100V=100Umin), V1 as Battery2, V3 as Capacity1 and V2 as Capacity2.
If I could display all the sensors, it would also help owners of older radios.
greetings,
Torsten

mstrens, 2023-11-23:
Spektrum has a very complex telemetry system. The worse of all protocols I have seen.
Each telemetry frame contains 22 bytes (If think).
They defined about 100 types of frames. Each type contains several fields. Several types of frame contains partly the same fields.
So there is a great redundancy.
If I add e.g. the 2 GPS (BCD) frames, then oXs will send the GPS data twice (once with the "binary" frame and once with the 2 GPS "BCD" frames). Furthermore I use currently a general flag to send GPS frame only when there are new GPS data. This becomes even more complex when there are 3 frames for GPS.
I also do not know what the handset will do when it receives the same data in 2 different types of frame. Is it something in the handset to select the types of frames to be processed/discarded?

Satcomix, 2023-11-23:
> Spektrum has a very complex telemetry system. The worse of all protocols I have seen. Each telemetry frame contains 22 bytes (If think). They defined about 100 types of frames. Each type contains [...]

Hello Mstrens,
I may have expressed myself a bit unclearly.
Not everything should be transferred, as today there are mainly newer radios with newer firmware.
A way to allow users to comment out(activate) and comment(deactivate) certain functions/sensors.
I also managed to swap the ESC, which I don't find in my radio, for the PBox and by adjusting certain parameters in srxl2.cpp and srxl_sensors.h I can now set the values ​​for V1(Voltage.) in the PBox on my radio ), read V2(Ampere), Temp1 and Temp2 precisely.
Of course, this assumes that the user can compile and create builds themselves.
greetings,
Torsten

mstrens, 2023-11-23:
Based on the doc of spektrum, I think that powerbox frame contains 
uint16_t	volt1;																// Volts, 0.01v
	uint16_t	volt2;																// Volts, 0.01v
	uint16_t	capacity1;															// mAh, 1mAh
	uint16_t	capacity2;															// mAh, 1mAh
	uint16_t	spare16_1;
	uint16_t	spare16_2;
	uint8_t	spare;
I do not know how you can use this frame with a spektrum handset to get e.g. current and temp with correct scale and units.
Does the spektrum handset allows you to change scale and units of the different fields?

mstrens, 2023-11-23:
Spektrum defines several frames to transmit only one or two fields e.g.
//////////////////////////////////////////////////////////////////////////////
//
//							VOLTAGE
//
//////////////////////////////////////////////////////////////////////////////
//
typedef struct
{
	uint8_t		identifier;														// Source device = 0x01
	uint8_t		sID;															// Secondary ID
	uint16_t		volts;															// 0.01V increments
} __attribute__((packed)) STRU_TELE_HV;

//////////////////////////////////////////////////////////////////////////////
//
//							TEMPERATURE
//
//////////////////////////////////////////////////////////////////////////////
//
typedef struct
{
	uint8_t		identifier;														// Source device = 0x02
	uint8_t		sID;															// Secondary ID
	int16_t		temperature;													// Temperature in degrees Fahrenheit
} __attribute__((packed)) STRU_TELE_TEMP;

I could use them but:
- I am not sure they are supported by all handset
- I am afraid I have to fill the rest of the frame (to get 22 bytes) with 0. It could be that the rx sent the 22 bytes to the handset and so it consumed perhaps a lot of the bandwidth over the RF signal (and so reduce the number of fields transmitted per sec)

mstrens, 2023-11-23:
Do you know the types of frame that are supported by old handset/receiver?

Satcomix, 2023-11-23:
Hello Mstrens,
Thank you for your answer and explanation.
In order to test the SRXL2 protocol for other users with my old DX8Gen1 radio, I had to come up with something because the old firmware means I can only work with certain telemetry IDs.
This doesn't matter for newer systems, which take the data from oXs directly.
The telemetry data that is transmitted can be found in the four spectrum sensors (case 0-4, 4 RX_MAH is not active).
ESC for example includes V1, V2, Temp1, Temp2 and RPM, but i have no ESC Sensor in my handset, only PBox, so i must change it.
Vario, GPS_binary, Airspeed are also in the Spektrum cases.
You can change these, like I did, and I changed the scale directly in the .cpp.
I haven't changed the unit, but for checking it doesn't matter whether it says 23mAh or 23°C

Satcomix, 2023-11-23:
> Do you know the types of frame that are supported by old handset/receiver?

All types for my handset, are in the sensors.h, but older Radios can not work with internal IDs, when you use the old XBUS system.
With the new SRXL2 protocol you can use also the internal sensor IDs.

Satcomix, 2023-11-23:
> Spektrum defines several frames to transmit only one or two fields e.g. ////////////////////////////////////////////////////////////////////////////// // // VOLTAGE // [...]

You can explain it well with these two:
tele_Device_Voltage ID 0x01 // High voltage sensor (INTERNAL)
In the older XBUS variant from Spektrum, the connection for voltage, temperature and RPM was attached directly to the receiver, hence the “internal sensor” which could not be assigned elsewhere. Of course the same applies to ID 0x02 Temp.
These sensors are usually not directly connected to the SRXL2 receiver. This means I can use the IDs as long as the recipient does not use these IDs internally.
In my case I can only set the following sensors on my handset display.
Volts/FlightPackV
RPM
Temp
Amps
GPS_BCD
Airspeed
Altitude
RxVoltage
GForce
PowerBox

mstrens, 2023-11-24:
I just tried to add gps_bcd frames to oXs (2.10.12).
There is a new #define in config.h to activate this format.
In the version on github, this option is currently activated but I plan to disable it by default in  the future (binary is more efficient)
When used, it replaces the gps binary format.

I did not tested it.
I do not fill the gps time currently

Satcomix, 2023-11-24:
Hello Mstrens,
Thank you very much.
I will test the version 2.10.12 this evening(18:00 Berlin Time) and will you give a feedback.
greetings,
Torsten

Satcomix, 2023-11-24:
Hello Mstrens,
I downloaded Ver.2.10.12-test, but I see 2.10.11 in the terminal.
On GH in the src folder I see in config.h Ver.2.10.12.
On the SPEKTRUM handset I only see zeros for GPS.
Airspeed and altitude are displayed correctly.
Greetings,
Torsten

mstrens, 2023-11-24:
perhaps I forgot to compile after changing the version nr.
I compile now and put it again on github.
Still I did not change the code. So it could be that it displays only 0 like before.

Note: I have plan to generate the frame only when there is a 3D fix.

Satcomix, 2023-11-24:
No improvement with 2.10.12
Only zeros on GPS
Ver.2.10.12
SRXL2 Protocol
GPS Groundspeed = 5 cm/s
GPS Heading = 0.440000 degree
GPS Altitude = 3721 cm
GPS Num sat. = 116
GPS Date J M A = 24 11 23 
GPS Time H M S = 18 46 55 
GPS Pdop = 130 
GPS Home bearing = 159 degree
GPS Home distance = 0 m

mstrens, 2023-11-24:
Yes I use LOC and STATS frames.

I put on github 2.10.13 where oXs should display on the PC the data being part of LOC frame.
Do you get messages with this version?

Satcomix, 2023-11-24:
I just looked again in the srxl2.cpp and the srxl2.h, there is still GPS_binary ID26 as the telemetry sensor.
The radios won't accept that.
You would have to create two GPS sensors in srxl2.cpp with GPS_LOC and GPS_STATS(ID 16 and 17, like GPS_binary.
These two would then also have to be listed as sensors in the srxl2.h.
I will test 2.10.13.

Satcomix, 2023-11-24:
Ver.2.10.13

GPS LOC alt=356 lat=51073536  long=6430328  course=849   Hdop=14
GPS LOC alt=356 lat=51073536  long=6430328  course=849   Hdop=14
GPS LOC alt=356 lat=51073536  long=6430328  course=849   Hdop=14
GPS LOC alt=356 lat=51073536  long=6430328  course=849   Hdop=14
GPS LOC alt=355 lat=51073536  long=6430328  course=1446   Hdop=14
GPS LOC alt=355 lat=51073536  long=6430328  course=1585   Hdop=14
GPS LOC alt=355 lat=51073536  long=6430327  course=1652   Hdop=14
GPS LOC alt=355 lat=51073536  long=6430327  course=1652   Hdop=14
GPS LOC alt=356 lat=51073536  long=6430327  course=1652   Hdop=14
GPS LOC alt=355 lat=51073536  long=6430327  course=1652   Hdop=14
GPS LOC alt=355 lat=51073536  long=6430327  course=1652   Hdop=14
GPS LOC alt=356 lat=51073537  long=6430327  course=1652   Hdop=14
GPS LOC alt=356 lat=51073537  long=6430327  course=1652   Hdop=15
GPS LOC alt=356 lat=51073537  long=6430326  course=1575   Hdop=15
GPS LOC alt=357 lat=51073537  long=6430326  course=1575   Hdop=15
GPS LOC alt=358 lat=51073538  long=6430325  course=1575   Hdop=15
GPS LOC alt=359 lat=51073538  long=6430325  course=1612   Hdop=15
GPS LOC alt=360 lat=51073538  long=6430325  course=1612   Hdop=14
GPS LOC alt=361 lat=51073539  long=6430325  course=1612   Hdop=14
GPS LOC alt=362 lat=51073539  long=6430325  course=1612   Hdop=14
GPS LOC alt=362 lat=51073539  long=6430325  course=1612   Hdop=14
GPS LOC alt=363 lat=51073540  long=6430324  course=1612   Hdop=14
GPS LOC alt=363 lat=51073540  long=6430324  course=1612   Hdop=14
GPS LOC alt=364 lat=51073540  long=6430324  course=1612   Hdop=14
GPS LOC alt=365 lat=51073541  long=6430324  course=1612   Hdop=14
GPS LOC alt=365 lat=51073541  long=6430324  course=1612   Hdop=14
GPS LOC alt=365 lat=51073541  long=6430324  course=1612   Hdop=14
GPS LOC alt=366 lat=51073542  long=6430324  course=1612   Hdop=14
GPS LOC alt=366 lat=51073542  long=6430323  course=1612   Hdop=14
GPS LOC alt=367 lat=51073542  long=6430323  course=1612   Hdop=14
GPS LOC alt=369 lat=51073543  long=6430323  course=1612   Hdop=14
GPS LOC alt=369 lat=51073543  long=6430323  course=1612   Hdop=14
GPS LOC alt=370 lat=51073543  long=6430322  course=1612   Hdop=14

mstrens, 2023-11-24:
This seems OK.
After having filled the values, I copy them in the buffer that is transmitted.
In 2.10.14 I now print the buffer that will be sent by oXs.

Satcomix, 2023-11-24:
Config parameters are OK
Press ? + Enter to get help about the commands
GPS LOC alt=421 lat=51073525  long=6430389  course=3427   Hdop=18
GPS LOC alt=421 lat=51073525  long=6430389  course=3427   Hdop=18
GPS LOC alt=421 lat=51073524  long=6430388  course=3512   Hdop=18
GPS LOC alt=421 lat=51073523  long=6430388  course=3485   Hdop=18
GPS LOC alt=422 lat=51073522  long=6430388  course=3485   Hdop=18
srxl2 buffer filled  A6  80  16  10   0   0  22   4  22  35   7  51  88   3  43   6  85  34  18   0  9A  8D 
GPS LOC alt=422 lat=51073521  long=6430388  course=3473   Hdop=18
GPS LOC alt=423 lat=51073521  long=6430388  course=3473   Hdop=18
GPS LOC alt=423 lat=51073521  long=6430388  course=3502   Hdop=18
srxl2 buffer filled  A6  80  16  10   0   0  23   4  21  35   7  51  88   3  43   6   2  35  18   0  55  28 
GPS LOC alt=423 lat=51073520  long=6430388  course=3502   Hdop=18
GPS LOC alt=423 lat=51073520  long=6430388  course=3502   Hdop=22
GPS LOC alt=423 lat=51073518  long=6430388  course=3488   Hdop=22
GPS LOC alt=423 lat=51073516  long=6430389  course=3488   Hdop=22
srxl2 buffer filled  A6  80  16  10   0   0  23   4  16  35   7  51  89   3  43   6  88  34  22   0  55  28 
GPS LOC alt=423 lat=51073516  long=6430389  course=3488   Hdop=22
GPS LOC alt=423 lat=51073516  long=6430389  course=3488   Hdop=22
GPS LOC alt=423 lat=51073515  long=6430389  course=3488   Hdop=22
GPS LOC alt=423 lat=51073514  long=6430389  course=3488   Hdop=22
GPS LOC alt=423 lat=51073514  long=6430389  course=3488   Hdop=22
GPS LOC alt=423 lat=51073512  long=6430389  course=3583   Hdop=22
srxl2 buffer filled  A6  80  16  10   0   0  23   4  12  35   7  51  89   3  43   6  83  35  22   0  97  B1 
GPS LOC alt=423 lat=51073510  long=6430389  course=3583   Hdop=22
GPS LOC alt=423 lat=51073509  long=6430390  course=2597   Hdop=22
GPS LOC alt=423 lat=51073509  long=6430390  course=2597   Hdop=26
GPS LOC alt=423 lat=51073509  long=6430390  course=2018   Hdop=26
GPS LOC alt=423 lat=51073509  long=6430390  course=2018   Hdop=26
srxl2 buffer filled  A6  80  16  10   0   0  23   4   9  35   7  51  90   3  43   6  18  20  26   0  55  28

mstrens, 2023-11-24:
It could be that I have to swap the bytes of each field.
I will try to do it.

Satcomix, 2023-11-24:
Normally, if this is a GPS BCD code, I could just change GPS_Loc and I should recognize the correct data.
I try it, takes 5 minutes.

mstrens, 2023-11-24:
Here version 2.10.15 where I swap the bytes of each fields.
This is the first really change I made in the code (except adding dbg msg)

Satcomix, 2023-11-24:
Of course you changed something, the sensors were previously in .cpp and . h not available :-))

Config parameters are OK
Press ? + Enter to get help about the commands
GPS LOC alt=0 lat=32350751  long=99034306  course=0   Hdop=32
GPS LOC alt=0 lat=32350751  long=99034306  course=0   Hdop=18
GPS LOC alt=0 lat=32350751  long=98034306  course=0   Hdop=26
GPS LOC alt=0 lat=32350751  long=98034306  course=0   Hdop=26
GPS LOC alt=0 lat=32350751  long=98034306  course=0   Hdop=18
GPS LOC alt=0 lat=33350751  long=98034306  course=0   Hdop=18
srxl2 buffer filled  A6  80  16  10   0   0   0   0  51   7  35  33   6  43   3  98   0   0  18   0  55  28 
GPS LOC alt=0 lat=33350751  long=98034306  course=0   Hdop=18
GPS LOC alt=0 lat=33350751  long=98034306  course=0   Hdop=18
GPS LOC alt=0 lat=33350751  long=98034306  course=0   Hdop=18
srxl2 buffer filled  A6  80  16  10   0   0   0   0  51   7  35  33   6  43   3  98   0   0  18   0  D1  D1 
GPS LOC alt=0 lat=33350751  long=98034306  course=0   Hdop=18
GPS LOC alt=0 lat=34350751  long=98034306  course=0   Hdop=18
GPS LOC alt=0 lat=34350751  long=98034306  course=0   Hdop=18
GPS LOC alt=0 lat=34350751  long=98034306  course=0   Hdop=18
srxl2 buffer filled  A6  80  16  10   0   0   0   0  51   7  35  34   6  43   3  98   0   0  18   0  D1  D1 
GPS LOC alt=0 lat=34350751  long=97034306  course=0   Hdop=18
GPS LOC alt=0 lat=34350751  long=97034306  course=0   Hdop=16
GPS LOC alt=0 lat=34350751  long=97034306  course=0   Hdop=16
GPS LOC alt=0 lat=34350751  long=97034306  course=0   Hdop=16
GPS LOC alt=0 lat=34350751  long=97034306  course=0   Hdop=16
GPS LOC alt=0 lat=34350751  long=97034306  course=0   Hdop=16
srxl2 buffer filled  A6  80  16  10   0   0   0   0  51   7  35  34   6  43   3  97   0   0  16   0  BA  C4 
GPS LOC alt=0 lat=34350751  long=97034306  course=0   Hdop=16
GPS LOC alt=0 lat=34350751  long=97034306  course=0   Hdop=16
GPS LOC alt=0 lat=34350751  long=97034306  course=0   Hdop=16
GPS LOC alt=0 lat=34350751  long=97034306  course=0   Hdop=16
GPS LOC alt=0 lat=34350751  long=97034306  course=0   Hdop=16
GPS LOC alt=0 lat=34350751  long=97034306  course=0   Hdop=16
srxl2 buffer filled  A6  80  16  10   0   0   0   0  51   7  35  34   6  43   3  97   0   0  16   0  BA  C4 
GPS LOC alt=0 lat=34350751  long=97034306  course=0   Hdop=18
GPS LOC alt=0 lat=34350751  long=97034306  course=0   Hdop=18
GPS LOC alt=0 lat=34350751  long=97034306  course=0   Hdop=18
GPS LOC alt=0 lat=34350751  long=96034306  course=0   Hdop=18
GPS LOC alt=0 lat=34350751  long=96034306  course=0   Hdop=18
GPS LOC alt=0 lat=35350751  long=96034306  course=0   Hdop=18
srxl2 buffer filled  A6  80  16  10   0   0   0   0  51   7  35  35   6  43   3  96   0   0  18   0  3E  CA

mstrens, 2023-11-24:
I found a bug.
I had forgotten to copy the "gpsflags' byte.
Now 2.10.16
I hope it is OK now.

Satcomix, 2023-11-24:
At handset only zero at GPS

srxl2 buffer filled  A6  80  16  10   0   0   0   0  51   7  34  71   6  43   3  94   0   0  15  3B  6F  EE 
GPS LOC alt=0 lat=71340751  long=94034306  course=0   Hdop=15
GPS LOC alt=0 lat=71340751  long=94034306  course=0   Hdop=15
GPS LOC alt=0 lat=71340751  long=94034306  course=0   Hdop=15
GPS LOC alt=0 lat=70340751  long=93034306  course=0   Hdop=15
GPS LOC alt=0 lat=70340751  long=93034306  course=0   Hdop=16
srxl2 buffer filled  A6  80  16  10   0   0   0   0  51   7  34  70   6  43   3  93   0   0  16  3B  B9  21 
GPS LOC alt=0 lat=70340751  long=93034306  course=0   Hdop=16
GPS LOC alt=0 lat=70340751  long=93034306  course=0   Hdop=16
GPS LOC alt=0 lat=70340751  long=93034306  course=0   Hdop=16
GPS LOC alt=0 lat=70340751  long=93034306  course=0   Hdop=15
GPS LOC alt=0 lat=70340751  long=93034306  course=0   Hdop=16
GPS LOC alt=0 lat=70340751  long=93034306  course=0   Hdop=16
srxl2 buffer filled  A6  80  16  10   0   0   0   0  51   7  34  70   6  43   3  93   0   0  16  3B  FA  D6 
GPS LOC alt=0 lat=70340751  long=93034306  course=0   Hdop=15
GPS LOC alt=0 lat=70340751  long=93034306  course=0   Hdop=16
GPS LOC alt=0 lat=70340751  long=93034306  course=0   Hdop=16
GPS LOC alt=0 lat=70340751  long=93034306  course=0   Hdop=15
GPS LOC alt=0 lat=70340751  long=93034306  course=0   Hdop=15
srxl2 buffer filled  A6  80  16  10   0   0   0   0  51   7  34  70   6  43   3  93   0   0  15  3B  B9  21 
GPS LOC alt=0 lat=70340751  long=93034306  course=0   Hdop=15
GPS LOC alt=0 lat=70340751  long=93034306  course=0   Hdop=15
GPS LOC alt=0 lat=70340751  long=93034306  course=0   Hdop=15
GPS LOC alt=0 lat=70340751  long=93034306  course=0   Hdop=15
srxl2 buffer filled  A6  80  16  10   0   0   0   0  51   7  34  70   6  43   3  93   0   0  15  3B  3D  D8 
GPS LOC alt=0 lat=70340751  long=93034306  course=0   Hdop=15
GPS LOC alt=0 lat=70340751  long=93034306  course=0   Hdop=15
GPS LOC alt=0 lat=70340751  long=93034306  course=0   Hdop=15
srxl2 buffer filled  A6  80  16  10   0   0   0   0  51   7  34  70   6  43   3  93   0   0  15  3B  4E  61 
GPS LOC alt=0 lat=70340751  long=93034306  course=0   Hdop=15
GPS LOC alt=0 lat=70340751  long=93034306  course=0   Hdop=15
GPS LOC alt=0 lat=70340751  long=93034306  course=0   Hdop=15
GPS LOC alt=0 lat=70340751  long=93034306  course=0   Hdop=15
GPS LOC alt=0 lat=70340751  long=93034306  course=0   Hdop=15
GPS LOC alt=0 lat=69340751  long=93034306  course=0   Hdop=17
srxl2 buffer filled  A6  80  16  10   0   0   0   0  51   7  34  69   6  43   3  93   0   0  17  3B  CA  50

mstrens, 2023-11-24:
This seems me OK. I do not understand why you do not get values on handset.
Now in 2.10.17 I also display the STATS frame (just to be sure).

Satcomix, 2023-11-24:
I'll continue testing tomorrow morning.
I had a little success this week and was able to see data on the GPS, it wasn't correct, but at least something was being transferred and displayed. I'll take a look at the .cpp and other srxl2 files tomorrow.
greetings,
Torsten

Config parameters are OK
Press ? + Enter to get help about the commands
GPS LOC alt=0 lat=5350751  long=90034306  course=0   Hdop=15
GPS LOC alt=0 lat=5350751  long=90034306  course=0   Hdop=15
GPS LOC alt=0 lat=5350751  long=90034306  course=0   Hdop=15
GPS LOC alt=0 lat=5350751  long=90034306  course=0   Hdop=15
GPS LOC alt=0 lat=5350751  long=90034306  course=0   Hdop=15
GPS LOC alt=0 lat=5350751  long=90034306  course=0   Hdop=15
srxl2 buffer LOC filled  A6  80  16  10   0   0   0   0  51   7  35   5   6  43   3  90   0   0  15  3B  5B  17 
GPS LOC alt=0 lat=5350751  long=91034306  course=0   Hdop=15
srxl2 buffer STATS filled  A6  80  16  10   0   0   0   0  FF  FF  FF  FF  17   0   0   0   0   0   0   0  5B  17 
GPS LOC alt=0 lat=5350751  long=91034306  course=0   Hdop=15
GPS LOC alt=0 lat=5350751  long=91034306  course=0   Hdop=15
srxl2 buffer LOC filled  A6  80  16  10   0   0   0   0  51   7  35   5   6  43   3  91   0   0  15  3B  5B  17 
GPS LOC alt=0 lat=5350751  long=91034306  course=0   Hdop=15
GPS LOC alt=0 lat=5350751  long=91034306  course=0   Hdop=15
GPS LOC alt=0 lat=5350751  long=91034306  course=0   Hdop=15
srxl2 buffer STATS filled  A6  80  16  10   0   0   0   0  FF  FF  FF  FF  17   0   0   0   0   0   0   0  DA  60 
GPS LOC alt=0 lat=5350751  long=91034306  course=0   Hdop=16
srxl2 buffer LOC filled  A6  80  16  10   0   0   0   0  51   7  35   5   6  43   3  91   0   0  16  3B  5B  17 
GPS LOC alt=0 lat=5350751  long=91034306  course=0   Hdop=16
GPS LOC alt=0 lat=5350751  long=91034306  course=0   Hdop=16
GPS LOC alt=0 lat=5350751  long=91034306  course=0   Hdop=16
GPS LOC alt=0 lat=5350751  long=91034306  course=0   Hdop=16
GPS LOC alt=0 lat=4350751  long=91034306  course=0   Hdop=16
srxl2 buffer LOC filled  A6  80  16  10   0   0   0   0  51   7  35   4   6  43   3  91   0   0  16  3B  8A  FD 
GPS LOC alt=0 lat=4350751  long=91034306  course=0   Hdop=16
srxl2 buffer STATS filled  A6  80  16  10   0   0   0   0  FF  FF  FF  FF  16   0   0   0   0   0   0   0  5B  17 
GPS LOC alt=0 lat=4350751  long=92034306  course=0   Hdop=16
GPS LOC alt=0 lat=4350751  long=92034306  course=0   Hdop=16
GPS LOC alt=0 lat=4350751  long=92034306  course=0   Hdop=16
GPS LOC alt=0 lat=4350751  long=92034306  course=0   Hdop=16
GPS LOC alt=0 lat=4350751  long=92034306  course=0   Hdop=17
srxl2 buffer LOC filled  A6  80  16  10   0   0   0   0  51   7  35   4   6  43   3  92   0   0  17  3B   E  CC 
GPS LOC alt=0 lat=4350751  long=92034306  course=0   Hdop=17
srxl2 buffer STATS filled  A6  80  16  10   0   0   0   0  FF  FF  FF  FF  15   0   0   0   0   0   0   0  91  1F 
GPS LOC alt=0 lat=4350751  long=92034306  course=0   Hdop=17
GPS LOC alt=0 lat=4350751  long=92034306  course=0   Hdop=17
GPS LOC alt=0 lat=4350751  long=92034306  course=0   Hdop=17
GPS LOC alt=0 lat=4350751  long=92034306  course=0   Hdop=17
GPS LOC alt=0 lat=4350751  long=92034306  course=0   Hdop=16
srxl2 buffer LOC filled  A6  80  16  10   0   0   0   0  51   7  35   4   6  43   3  92   0   0  16  3B   E  CC

mstrens, 2023-11-24:
Very strange: STATS frame should have an ID =17 and here it is 16. I do not yet understand why.

mstrens, 2023-11-24:
I found the bug about ID=17 for STATS
Now 2.10.18

Satcomix, 2023-11-25:
Good Morning Mstrens.
Test with version 2.10.19-test
First smal successes!!!
The time is now displayed, although it's about 7 hours too short and it doesn't count any further, but I can see hours, minutes and seconds.
The sats are displayed, so with 3DFix the number is correct.
I see the following values ​​on the display.
Speed: 0.0km/h
OLD: ---------m
Heading: 0°
Lat: 00.000000
Lon: 00.000000
Time: 01:47:45
Sats: 19
I can't yet interpret these ads:
0.0
0.0
0m
I think
Good Morning Mstrens.
Test with version 2.10.19-test
First successes!!!
The time is now displayed, although it's about 7 hours too short and it doesn't count any further, but I can see hours, minutes and seconds.
The sats are displayed, so with 3DFix the number is correct.
Altitude apparently it is not being transferred yet, or the value is too high.
I see the following values ​​on the display.
Speed: 0.0km/h
Alt: ---------m
Heading: 0°
Lat: 00.000000
Lon: 00.000000
Time: 01:47:45
Sats: 19
I can't yet interpret these ads:
0.0
0.0
0m
I think the three values are: course, bearing from home and distance from home or cumulative distance.

mstrens, 2023-11-25:
It is still very strange.
Time is part of SATS frame. It is just the field I do not fill with gps data but with a dummy value FFFFFFFF (in cleanflight code it is the code for unknown value).
I just see also that cleanflight fill also 6 bytes with FF at the end of the frame in order to have a payload of 16 bytes.
I added it to 2.10.21 and I display again the 2 frames.

Still it seems that you do not get the data from LOC frame. I do not see why.

Satcomix, 2023-11-25:
The result is the same with version 2.10.21.
3D Fix Sats works.
Everything else is zero and at altitude dashes.
Time: 01:47:45 no change

Satcomix, 2023-11-25:
I just looked at the files.
I believe the problem is with the simultaneous transmission of GPS_binary and GPS_Loc/Stats
I'll change something and then do a build.

mstrens, 2023-11-25:
I do not think so.
There is a #define in config.h that is used to say that it is bcd or binary but not both at the same time

Satcomix, 2023-11-25:
You're right, I had just commented out (deactivated) certain things, but that didn't work either.

mstrens, 2023-11-25:
2.10.22.
I found the same bug for LOC as for STATS (stupid copy/paste).
Do you get LOC and STATS frame on the PC display?

Satcomix, 2023-11-25:
Hello Mstrens,
I just changed in srxl2.cpp the name GPS_binary in version 2.10.10 to GPS_Loc.
Then I get a display for Heading, Lat and Lon.
Heading: from 100° to 1600°
Lat: DD:846373
Lon 16F:700006
Time: 00:00:00
With Lat and Lon the values ​​before : i.e. DD change to 77- A7, and with 16F to 1E2 - B2

mstrens, 2023-11-25:
Did you tried 2.10.22?

Satcomix, 2023-11-25:
Ver. 2.10.22
Display:
Speed: 0.0km/h  0.0
Alt: 0.1m            0.1
Heading: 0.0°    16987m
Lat: 42.584585
Lon: 81.057176    Sat:
Time: 01:47:45      17

Terminal:
srxl2 buffer LOC filled  A6  80  16  10  16   0   0   0  51   7  35  66   6  43   3  72   0   0  13  3B  94  B2 
srxl2 buffer STATS filled  A6  80  16  10  17   0   0   0  FF  FF  FF  FF  17   0  FF  FF  FF  FF  FF  FF  C0  72 
srxl2 buffer LOC filled  A6  80  16  10  16   0   0   0  51   7  35  67   6  43   3  72   0   0  13  3B  94  B2 
srxl2 buffer STATS filled  A6  80  16  10  17   0   0   0  FF  FF  FF  FF  16   0  FF  FF  FF  FF  FF  FF  FA  35 
srxl2 buffer LOC filled  A6  80  16  10  16   0   0   0  51   7  35  67   6  43   3  72   0   0  13  3B  94  B2 
srxl2 buffer STATS filled  A6  80  16  10  17   0   0   0  FF  FF  FF  FF  16   0  FF  FF  FF  FF  FF  FF  D1  D1 
srxl2 buffer LOC filled  A6  80  16  10  16   0   0   0  51   7  35  67   6  43   3  72   0   0  13  3B  74  C5 
srxl2 buffer STATS filled  A6  80  16  10  17   0   0   0  FF  FF  FF  FF  16   0  FF  FF  FF  FF  FF  FF  D1  D1 
srxl2 buffer LOC filled  A6  80  16  10  16   0   0   0  51   7  35  67   6  43   3  72   0   0  13  3B  75  5E 
srxl2 buffer STATS filled  A6  80  16  10  17   0   0   0  FF  FF  FF  FF  16   0  FF  FF  FF  FF  FF  FF  94  B2 
srxl2 buffer LOC filled  A6  80  16  10  16   0   0   0  51   7  35  68   6  43   3  72   0   0  13  3B  94  B2 
srxl2 buffer STATS filled  A6  80  16  10  17   0   0   0  FF  FF  FF  FF  15   0  FF  FF  FF  FF  FF  FF  E3  ED 
srxl2 buffer LOC filled  A6  80  16  10  16   0   0   0  51   7  35  68   6  43   3  72   0   0  13  3B  94  B2 
srxl2 buffer STATS filled  A6  80  16  10  17   0   0   0  FF  FF  FF  FF  16   0  FF  FF  FF  FF  FF  FF  70  F3 
srxl2 buffer LOC filled  A6  80  16  10  16   0   0   0  51   7  35  69   6  43   3  73   0   0  13  3B  94  B2 
srxl2 buffer STATS filled  A6  80  16  10  17   0   0   0  FF  FF  FF  FF  16   0  FF  FF  FF  FF  FF  FF  94  B2 
srxl2 buffer LOC filled  A6  80  16  10  16   0   0   0  51   7  35  69   6  43   3  73   0   0  13  3B  E3  ED 
srxl2 buffer STATS filled  A6  80  16  10  17   0   0   0  FF  FF  FF  FF  16   0  FF  FF  FF  FF  FF  FF  94  B2 
srxl2 buffer LOC filled  A6  80  16  10  16   0   0   0  51   7  35  69   6  43   3  73   0   0  12  3B  BF  56

mstrens, 2023-11-25:
The values you get for long and lat are strange.
I made 2.10.23 where I removed swapping the bytes.
Perhaps it helps

[22 more replies]

closed 2023-11-27 by Satcomix