mstrens/oXs_on_RP2040 #78: Support question: free up bandwidth

Allerhopp, 2023-05-04:
I would like to exclude VSpeed from sending to free up bandwidth. Compensated VSped is sufficient for me. Instead I would like to increase the rate of other sensors like airspeed. 
What is the best way to exclude single sensors? I understand that I then have to compile myself.

Satcomix, 2023-05-04:
Hello Allerhopp,
When you try it this way, like the DIY_YAW value.

    sportFieldId[PITCH] = DIY_PITCH;
    sportFieldId[ROLL] = DIY_ROLL;
    //sportFieldId[YAW] = DIY_YAW;
    sportFieldId[RPM] = RPM_FIRST_ID;
    sportFieldId[ADS_1_1] = DIY_ADS_1_1;
br,
Torsten

Allerhopp, 2023-05-05:
Looks good. Thank you!

mstrens, 2023-05-08:
I put on github a version 2.3.0 (in test branch) that is supposed to let the user set up the priorities of telemetry fields for Sport and Fbus.
In config.h file, there are some new parameters.

Here an extract :
// The #define below allow to setup the relative priority of field transmissions
// Value should be in range 3...250 
// A field with a lower value will be transmitted more often than another field with a higher value  
// Still value 0 means that this field may not be transmitted
// The values are relative.
//     E.g. if oXs has 3 fields to transmit; one with priority 5 and 2 with priority 50
//        the one with priority 5 will be transmitted about 10 X more often than the 2 others   
// There is no need to set the value to 0 for fields not available (e.g. GPS fields if GPS pins are undefined) 
#define P_LATITUDE 50
#define P_LONGITUDE 50
#define P_GROUNDSPEED 50
#define P_HEADING 80
#define P_ALTITUDE 80
#define P_NUMSAT 200
...

Satcomix, 2023-05-08:
Hello Mstrens,
In order to understand everything a little more precisely with the transmission of the telemetry values, I have a question that you have discussed with Mike Blandfort before, i think.
There are IDs from 0 to 27 for the S.PORT/FBUS protocol and every 12ms a value with ID and AppId is queried and transmitted.
How many values ​​were there again that fit into a frame without problems?
I have something of about 70 in mind and the possible sensor lost message should only be transmitted up to 50, right?
I can't find the post in the RCG anymore.
Greetings,
Torsten

mstrens, 2023-05-08:
If I remember well Sport pooling is at 11ms (and not 12ms). (not 100% sure)
Mike said that only about 75% of the pool requests should be replied (because the RF has a limited bandwidth). 
oXs on pro mini could replied up to 4 or 5 Ids but oXs_on RP2040 replies only one to one of the 28 IDs.
If oXs is the only one device on the bus, at the max , it will reply to 1 pool request on 2 so it will use max 50%
If there 1 other device, it will reply to 1 pool request on 3 so at 33%.
Etc...

Satcomix, 2023-05-08:
Thank you for the answer, I still had those things in the back of my mind somewhere.
br,
Torsten

jasc76, 2023-06-19:
can this be put in the console ?

mstrens, 2023-11-21:
In principe, oXs adjust automatically the bandwidth depending on the fields to be transmitted.
I think it works.

closed 2023-11-21 by mstrens