mstrens/oXs_on_RP2040 #79: fuel gauge as with OxS

onki69, 2023-05-08:
In OpenXSensor it was possible to connect a flow meter that provides pulses when fuel flow through it.
A very common sensor is this one:
https://www.voelkner.de/products/24929/B.I.O-TECH-e.K.-Durchfluss-Sensor-FCH-M-PP-LC-FCH-M-PP-LC-Betriebsspannung-Bereich-5-24-V-DC-Messbereich-0.015-0.8-l-min.html?offer=ee5d91568f0a212fbda527a611fde25b

Do you plan to implement the fule gauge capabilities from OxS into the 2040 project as well?

I don't fly gas engines but my friends would be happy to have a fuel gauge.

mstrens, 2023-05-08:
At this stage, I do not plan to implement fuel gauge into oXs_on RP2040.
I do not have anymore a sensor to test it.
It also requires quite many changes because oXs has to accept commands from the Tx (lua script or Jeti menu) in order to:
- calibrate the sensor
- define the tank capacity
- reset the % when tank is filled.
I suggest to use a pro mini version for this (not so common) feature.

onki69, 2023-05-08:
OK - thanks for your feedback.