mstrens/oXs_on_RP2040 #80: OXS 2040 with Jeti EX-Bus

Klippeneck, 2023-05-11:
Hello mstrens,
just played version 2.3.0 on my testboard, with protocol Jeti J it works with MS5611 and Beitan GPS220.
As a test I only changed the protocol to Jeti EX Bus, then I don't see the sensor in the transmitter.
Receiver input was set to EX bus and sensor configured accordingly.
Has anyone else done tests with Jeti? Attached is the config file.
Greetings Jürgen
Version = 2.3.0 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   28  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   29  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   26  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   27  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255  255  255 (V1 / V4 = 26, 27, 28, 29)

Protocol is Jeti (Exbus)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is not detected
Airspeed sensor is not detected
    Vspeed compensation channel = 0
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox (configured by oXs) :GPS is detected but has not (yet) a fix
Failsafe type is HOLD

Config parameters are OK
Press ? + Enter to get help about the commands

mstrens, 2023-05-11:
I am sorry but the jeti exbus protocol has not yet been tested. So I am not surprised it does not work.
I am waiting for someone with a logic analyser to make a capture of the exbus signal to understand what is going wrong.

KeesBlokland, 2023-05-12:
Hi Michel, I've only just started playing with this project, so I am not fully up to date yet. (trying to wade through RC-Groups and RC-Network posts) I've been testing this part as well, as soon as I get my fresh protocol analyzer (should be there in a few days) I will let you know.  Attached doc shows my progress to date. As far as I can see, the prepared output looks ok.  

I hope I can contribute to this part of this great project! 

regards,
Kees

[RP2040-no-EXbus-yet.pdf](https://github.com/mstrens/oXs_on_RP2040/files/11466283/RP2040-no-EXbus-yet.pdf)

KeesBlokland, 2023-05-19:
I am at the point where I need to acquire some serious debugging skills. I have been trying to get some help from ChatGPT, and even though that is fun, it does not always give correct answers. (no surprise there, it is a text tool trying to do things it does not understand) However 'we' managed to create working code to process the CRC stuff so that it gets to the 'Send Telemetry' function. But there it stops for the moment. 

[RP2040-no-EXbus-3.pdf](https://github.com/mstrens/oXs_on_RP2040/files/11514695/RP2040-no-EXbus-3.pdf)

mstrens, 2023-05-19:
In order to debug, I use 2 ways.
- first I add some "printf" commands in the code in order to get information on the PC (via USB cable). This allows to know if the  program run into some part of the code. It also allows to know the value of some variable and so to know if it is normal or not.
- I use a logic analyser to look at the signals received and generated by oXs. This is important e.g. on a bidirectional bus (like Exbus) to check if the replies of oXs fit into the time windows expected by the protocol (avoiding e.g. conflict between an oXs reply and the next request/data  sent by the receiver. 

I think that oXs can read the data sent by the receiver because:
- oXs does not give debug messages (via printf) that the CRC of the received frames is wrong
- oXs gives debug messages with the content of the telemetry frame it created.

If the Tx does not give telemetry data, it means that there is a bug
- or in the content of the telemetry frames being generated
- and/or in the timing those frames are put on the exbus.
The best tool to check this is a logic analyser.

KeesBlokland, 2023-05-19:
Thank you, effectively I use those 2 methods as well, I usually look at both the data on the logic analyzer and compare it to what I see on the USB dongle. I see that the data on the USB always matches the data on the USB dongle. Therefore I mostly use the USB dongle for logging the data, it is a bit more convenient. 

I do use print statements as much as I can. Example:
 
bool exbusProcessNextInputByte( uint8_t c){
    // check if valid
    printf("Process incoming byte: 0x%02X\n", c);

I see the incoming packets.  The output of HTerm (USB dongle) and the output on my terminal right now is:

 _2e e0 2e e0 2e 92 24_                      //this is the start of the terminal output. I removed some cr's in the terminal output for readability. 

**3d 1 8 25 3a 0 c7 6d**         //    my telemetry request, as seen on the input to exbusProcessNextInputByte

_3e                                               // next 3 lines as they appear on terminal.
Process incoming byte: 0x3E
exbus rx buffer with wrong check digit
3 28 25 31 20_    //end of terminal output, with removed cr's for readability.

Some how it consistently starts processing the next 3E sequence, but skipping the 3D sequence. 

Exactly the same happens when I enable the Simulate RX_EXbus code. 

Sofar I have not seen any evidence that it even starts to use exbusCreateSendTelemetry();
from: _printf("calling exbusCreateSendTelemetry\n");
                        exbusCreateSendTelemetry();_

My next step is trying to find out why it skips the 3D0108 but tries to process the 3E sequence. 
And yes' I'll look at the logic analyzer output, to see at what stage in the message the 3D01 appears, compared to the 3E etc. 

an interesting problem for sure!

mstrens, 2023-05-19:
When I:
- uncomment the line #define SIMULATE_RX_EXBUS
 and 
- add     printf("exbus creating tlm frame\n"); at the beginning of function exbusCreateSendTelemetry()

then I get on USB:
exbus decoding Rc channels
exbus creating tlm frame
exbus decoding Rc channels
exbus creating tlm frame
exbus decoding Rc channels
exbus creating tlm frame
exbus decoding Rc channels
exbus creating tlm frame

 This shows that the 2 types of frames coming from the receiver are recognized by oXs

KeesBlokland, 2023-05-19:
Strange indeed!

Tomorrow morning I’ll try your published version for testing, just to exclude the possibility of me building buggy code.

KeesBlokland, 2023-05-20:
Good morning all. 
I am looking at the Logic analyzer input (Ex-bus), using the unmodified oXs.uf2 as published here. 
After acquiring a bit of data, stop the Tx, and start looking for 0x3D packets. 

- First thing to notice is that the Telemetry requests arrive exactly every 10 mSec. (green marker 1)
- After the the complete request has been received, you see the 4 milliSec gap, where TLM data could go. (Between purple marker 2)

And then the first byte that arrives is 3E, which sort of matches my earlier observation. (Process incoming byte: 0x3E)

Next: build code again, and verify that my build matches what I see here. when it matches, start adding debug print statements.

[image]
[image]
File captured with Logic 2 from Saleae:
[Ex-bus-tlm_req.zip](https://github.com/mstrens/oXs_on_RP2040/files/11521727/Ex-bus-tlm_req.zip)

KeesBlokland, 2023-05-20:
Hi, when you say: 
    uncomment the line #define SIMULATE_RX_EXBUS
    and
    add printf("exbus creating tlm frame\n"); at the beginning of function exbusCreateSendTelemetry()

then I get on USB:
exbus decoding Rc channels
exbus creating tlm frame

With SIMULATE_RX_EXBUS enabled, I only get 'exbus rx buffer with wrong check digit' on my terminal. 

my .uf2 and exbus.cpp: 
[oXs.zip](https://github.com/mstrens/oXs_on_RP2040/files/11521845/oXs.zip)
[exbus-cpp.zip](https://github.com/mstrens/oXs_on_RP2040/files/11521850/exbus-cpp.zip)

[image]

mstrens, 2023-05-20:
do you use the version in test branch?
When you ac

mstrens, 2023-05-20:
When you activate the simulation, oXs should not be connected to the receiver (otherwise there could be a merge of simulation and receiver signal)

mstrens, 2023-05-20:
perhaps that tlm did not work because oXs dis not sent reply to jetibox request.
In a new version 2.4.1 in test branch, I now added support for jetibox frame.

Hoping it could help.

mstrens, 2023-05-20:
Sorry, in 2.4.1 I forgot to remove the option to simulate a jeti RX.
This could explain the messages (conflict between receiver and simulation).
I now put version 2.4.2 on github where I removed this simulation option.

Perhaps is it ok now

KeesBlokland, 2023-05-21:
Just installed your 2.4.2 version on my board. When tx is on, I see the following two messages in the  VSCode terminal. 

exbus decoding Rc channels
exbus creating tlm frame.

but no actual messages on the ex-bus yet. 

Q: I am using Visual Studio Code v 1.78.2 and GCC 12.1.0 arm-none-eabi. However, even though the code compiles without errors, I do not seem to be able to get the same results from my uf2 that you get. Can you share your setup?

Q: The topic seems to be closed, is there another place we can continue the story?

Enclosed the capture file I just made. 
[ex-bus-2.4.2.zip](https://github.com/mstrens/oXs_on_RP2040/files/11523797/ex-bus-2.4.2.zip)

mstrens, 2023-05-21:
I do not know why thye topic was closed. I reopen it.

I am also using VS code 1.78.2 but for GCC I am on 10.3.1 (this is probably the version that the windows rapsberry installer has installed).
The 2 messages that you get are the same as the one I get (but as I have no Jeti TX/RX I have to activate the simulation with the #define). Take care not to activate the simulation when oXs is connected to the RX.

With the version you compiled yourself, when the TX is on, do you get the same messages as with my version?
When the Tx is off, I expect that the RX set the bus in High impedance and it could be that oXs get noise (= dummy char)

May I suggest making first tests with a config that only activate Volt1 (so you can set the VOLT2, ...I2C and GPS pins on 255).
Compared with the version non EXbus, the EXbus version can sent more than 15 telemetry fields. For fields having an index above 16, I use a special way to fill the data. I did not find this way in the doc but in the code made by other people. I expect it is valid but I am not 100% sure.
Another difference between non Exbus and Exbus is the fact that in Exbus I have some gaps in the sequence of field index. Again I expect it is valid because I saw other code that are build in the same way. 
If needed, I could make a test version that would only sent one tlm field with an index = 1 (just to be sure) 

I did not yet looked at your capture. I will do it in the next hours.

mstrens, 2023-05-21:
I put on github a version 2.4.3 where:
- I send the tlm frame faster after the request
- I send the list of fields more often.

Not sure it will help!

mstrens, 2023-05-21:
The jeti EXbus protocol is still in testing phase.
As far I know the Tx does not discover the sensors and I do not know why.

I also did not really take care up to now about the color of the led for Jeti Exbus protocol. I would have to check this part. of the code.
I still expect that the led should blink (after the setup delay). This is done to show that the program is still running the main loop (and so is not blocked somewhere).
Please note also that some users got RP2040-zero modules with a led that has 2 inverted colors. To take care of this, it is possible to activate an option in config.h to invert the colors.

KeesBlokland, 2023-05-21:
re led: it’s only correct when you change the setting in config.h 
after that, the behavior is correct.

This is valid for my Waveshare board, so you need to play until you get the correct colours.

KeesBlokland, 2023-05-21:
I still see that the first byte processed is the 3E, not the 3D. Don’t want to post too much noise, so I will try to get to the bottom of that. 

If it is helpful I can post a longer capture of actual exbus traffic. A couple of meg should be no problem.

Let me know if you need it.

regards, Kees

mstrens, 2023-05-21:
In Exbus protocol, the Rx sent 2 frames after each other.
The first one starts with a 3E 03. It provides the Rc channels.
It is (often) immediately followed by a 3D 01 that is the telemetry request.
oXs replies to the 3D 01.

Each frame sent by the receiver has a packetId. It is strange that this packetId does not change in most case.
I expect that the receiver reuses the same packetId as long it does not receive a valid reply.

I will check again the way I fill the frame for a reply.
If I do not find the bug, I will probably make a version where I just sent dummy frames copied from a official jeti sensor.

Do you have an official Jeti sensor? If so, I would like to get a capture with a logic analyser of the exbus signal to compare with what I am sending.

mstrens, 2023-05-21:
I put on githu a version 2.4.4 where I change the position of the first char used to calculate a CRC inserted in the telemetry frame.
I hope this solve the issue.
Let me know.

mstrens, 2023-05-21:
Thanks for the feed back.
I now put version 2.4.5 on github in order to
- remove the debug messages
- sent the field names less often (once every 256 frames instead of once every 3); this allow to sent more often the fields values.
Note : I do not understand why you can't compile yourself. It is strange.

Satcomix, 2023-05-21:
> 1. Using the version you compiled, the sensor can display RP2040 flashing red, but it will flash once every 1 second on the remote control display screen. The time to find all the sensors in the remote control is also very slow, which takes about 1 minute;
> 
>     2. Using my own compiled version, the sensor cannot display properly on the remote control.
>        [image]

Hello aepub,
I think you have a problem with the folders.
I have attached a complete .txt printout of the build folder and running a Ver.2.4.5 build from my PC with Win10Pro
Maybe it helps.
Best regards,
Torsten
[VSCode_oXs_Build.txt](https://github.com/mstrens/oXs_on_RP2040/files/11524660/VSCode_oXs_Build.txt)

KeesBlokland, 2023-05-21:
just put 2.4.5 on the chip, fired up the tranny and…
[image]

looks like you did it!

Satcomix, 2023-05-21:
> My compilation environment is installed by default（ https://github.com/raspberrypi/pico-setup-windows/blob/master/docs/tutorial.md ）The code for the main branch can be used normally after compilation, but the only difference is that the current test compilation is different from yours, and I don't understand why. The attachment shows me using vscode to open the folder oXs_ on_ RP2040 test and compiled output text information. [aepub-build-oxs.txt](https://github.com/mstrens/oXs_on_RP2040/files/11524804/aepub-build-oxs.txt)

Hello aepub,
I see some differences to my SDK build folders.
I build on C:/ and save to C:/
When you install Pico setup, on C:/User/Documents/Your_Folder_CMake_RP2040.
This folder contains 4 folders:
1:  Your_Folder_oXs_RP2040,
2: pico-examples
3: pico-extras
4: pico playground
Normally in 1: Your_Folder_oXs_RP2040 you have 4 folders.
1: .vscode
2: build
3: oXs_on_RP2040-main
4: oXs_on_RP2040 test
In this folder are the unzipped copys of the GitHub -test or -main .zip folder.

In 4: oXs_on_RP2040 test you have the folders:
vscode, build, oXs_on_RP2040 test
br,
Torsten

mstrens, 2023-05-22:
@Klippeneck
I do not see immediately why GPS lat jump.
I put a version 2.4.8 on github that should sent debug messages to the USB.
There should be 2 types of lines:
- one with the original GPS lat value from the Ublox and the other with the GPS lat as formatted for Jeti
- the other with all telemetry frames (txt or data). Those lines contain only an extract of the frame (txt or data part)
Looking at the message will perhaps help to find the bug.
Can you try it and send me a copy of the messages (at least after GPS got the fix)

Klippeneck, 2023-05-22:
Hello mstrens,
I have installed SW 2.4.8, but when I enter FV for a fix, I get the following:
fv
processing cmd
Cmd to execute:   FV

GPS Latitude = 48.0543008 degree
GPS Longitude = 8.7253440 degree
GPS Groundspeed = 6 cm/s
GPS Heading = 243.750000 degree
GPS Altitude = 82623 cm
GPS Num sat. = 107
GPS Date J M A = 22 5 23 
GPS Time H M S = 17 9 52 
GPS Pdop = 230 
GPS Home bearing = 64 degree
GPS Home distance = 21 m
Vspeed = -2 cm/s
Baro Rel altitude = 72 cm
Gps cumulative distance = 30
Vspeed compensation = 1.10

I am not entirely clear about the USB, is the sensor operated on the receiver for this purpose? 
and in parallel via USB of the RP2040 with a terminal program the output is recorded on USB? I would then also have to supply the sensor with voltage via USB.

Satcomix, 2023-05-22:
Hello Klippeneck,
The command FV in terminal shows you the FieldView of the telemetry fields, that are transmitted from the oXs_RP2040 to the handheld/transmitter.
FV=FieldView not GPS Fix.
br,
Torsten

mstrens, 2023-05-22:
When RP2040 is connected to PC via USB,  I would recommend (not sure it is really required) to power the RP2040 (and sensors) only via USB. So I suggest to disconnect the red wire from the Exbus JR connector.

Klippeneck, 2023-05-22:
Hello mstrens,
Frame= 1  11  0  40 
Frame= 1  11  0  40 
Frame= 61  69  0  1  11  0  40 
Frame= 61  69  0  1  11  0  40 
lat orig=1ca47e84  jeti=300cb6
Frame= 19  b6  c  30  0  29  18  aa  8  20  31  7  20  41  0  0  51  6  3  1  1  11  0  40 
Frame= 1  11  0  40 
Frame= 1  11  0  40 
Frame= 12  19  41  6c  74  6d 
Frame= 61  69  0  4  12  2  0  20  1  11  0  40 
lat orig=1ca47e84  jeti=300cb6
Frame= 19  b6  c  30  0  29  18  aa  8  20  31  a  20  41  0  0  51  6  3  1  1  11  0  40 
Frame= 1  11  0  40 
Frame= 1  11  0  40 
Frame= 61  69  0  1  11  0  40 
Frame= 61  69  0  1  11  0  40 
lat orig=1ca47e84  jeti=300cb6
Frame= 19  b6  c  30  0  29  18  aa  8  20  31  8  20  41  0  0  51  6  3  1  1  11  0  40 
Frame= 1  11  0  40 
Frame= 1  11  6  40 
Frame= 61  69  0  1  11  6  40 
Frame= 61  69  0  1  11  6  40 
lat orig=1ca47e81  jeti=300cb6
Frame= 19  b6  c  30  0  29  18  aa  8  20  31  7  20  41  0  0  51  6  3  1  4  12  2  0  20  1  11  6  40 
Frame= 1  11  6  40 
Frame= 1  11  6  40 
Frame= 61  69  0  1  11  6  40 
Frame= 11  33  56  73  70  65  65  64  6d  2f  73 
lat orig=1ca47e83  jeti=300cb6
Frame= 19  b6  c  30  0  29  18  aa  8  20  31  2  20  41  0  0  51  6  3  61  69  0  1  1  11  6  40 
Frame= 1  11  6  40 
Frame= 1  11  6  40 
Frame= 61  69  0  1  11  6  40 
Frame= 1  11  6  40 
lat orig=1ca47e86  jeti=300cb6
Frame= 19  b6  c  30  0  29  18  aa  8  20  51  6  3  61  69  0  1  1  11  6  40 
Frame= 31  3  20  41  0  0  4  12  3  0  20  1  11  6  40 
Frame= 1  11  6  40 
Frame= 61  69  0  1  11  6  40 
Frame= 61  69  0  1  11  6  40 
lat orig=1ca47e8a  jeti=300cb6
Frame= 19  b6  c  30  0  29  18  aa  8  20  31  4  20  41  0  0  51  6  3  1  1  11  6  40 
Frame= 1  11  6  40 
Frame= 1  11  6  40 
Frame= 61  69  0  1  11  6  40 
Frame= 61  69  0  1  11  6  40 
Frame= 0  19  6f  58  73  20 
lat orig=1ca47e8c  jeti=300cb6
Frame= 19  b6  c  30  0  29  18  aa  8  20  31  2  20  41  0  0  51  6  3  1  4  12  2  0  20  1  11  6  40 
Frame= 1  11  6  40 
Frame= 61  69  0 
Frame= 1  11  6  40 
Frame= 61  69  0  1  11  6  40 
lat orig=1ca47e8d  jeti=300cb6
Frame= 19  b6  c  30  0  29  18  aa  8  20  31  3  20  41  0  0  51  6  3  1  1  11  6  40 
Frame= 1  11  6  40 
Frame= 1  11  6  40 
Frame= 61  69  0  1  11  6  40 
Frame= 61  69  0  1  11  6  40 
lat orig=1ca47e8e  jeti=300cb6
Frame= 19  b6  c  30  0  29  18  aa  8  20  31  3  20  41  0  0  51  6  3  1  1  11  6  40 
Frame= 1  11  6  40 
Frame= 4  12  3  0  20  1  11  6  40 
Frame= 61  69  0  1  11  6  40 
Frame= 61  69  0  1  11  6  40 
Frame= 1  39  47  70  73  20  4c  61  74  b0 
lat orig=1ca47e8d  jeti=300cb6
Frame= 19  b6  c  30  0  29  18  aa  8  20  31  5  20  41  0  0  51  6  3  1  1  11  6  40 
Frame= 1  11  6  40 
Frame= 61  69  0  1  11  6  40 
Frame= 61  69  0  1  11  6  40 
lat orig=1ca47e8c  jeti=300cb6
Frame= 19  b6  c  30  0  29  18  aa  8  20  31  1  20  41  0  0  51  6  3  1  1  11  6  40 
Frame= 1  11  6  40 
Frame= 1  11  6  40 
Frame= 61  69  0  4  12  3  0  20  1  11  6  40 
Frame= 61  69  0  1  11  6  40 
lat orig=1ca47e8c  jeti=300cb6
Frame= 19  b6  c  30  0  29  18  aa  8  20  31  2  20  41  0  0  51  6  3  1  1  11  6  40 
Frame= 1  11  6  40 
Frame= 1  11  6  40 
Frame= 61  69  0  1  11  6  40 
Frame= 61  69  0  1  11  6  40 
lat orig=1ca47e8d  jeti=300cb6

mstrens, 2023-05-22:
The debug messages seems OK.
The first frame line following a lat orig line contains always the same value B6 C 30 0 (that matches the jeti value - in reverse order).

So, I do not understand why the latitude value displayed on the TX changes.

KeesBlokland, 2023-05-23:
> The debug messages seems OK. The first frame line following a lat orig line contains always the same value B6 C 30 0 (that matches the jeti value - in reverse order).
> 
> So, I do not understand why the latitude value displayed on the TX changes.

..maybe we are pushing the limits of what Jeti can handle! But before jumping to that conclusion, it’s time for more debuggins. Weather is lousy here, so it’s a good day for it.

mstrens, 2023-05-23:
you can try different things:
- change the parameters via usb commands to let oXs transmit only GPS data (set volt1,..SCL, SDA on 255)
- let oXs transmit all fields with fixed values using the commands FVP and FVN. Note: please use new version 2.4.9 when you use those commands because I just fixed a bug (protocol was changed to J when those commands where used)

KeesBlokland, 2023-05-23:
Here you go: Version = 2.4.9 

with only GPS:

[devttyACM0_2023_05_23.08.57.10.775.zip](https://github.com/mstrens/oXs_on_RP2040/files/11540558/devttyACM0_2023_05_23.08.57.10.775.zip)
[ex-bus-2.4.9.zip](https://github.com/mstrens/oXs_on_RP2040/files/11540559/ex-bus-2.4.9.zip)

mstrens, 2023-05-23:
In the video, I see that LON is also jumping. It seems that the other data (alt, speed) are not jumping (they just change slowly).
It is also abnormal that the GPS cumulative distance is negative.

When you activate FVP, are LAT and LONG also jumping?

KeesBlokland, 2023-05-23:
I guess the number of Sats being shown as 117, means I have a 3D fix (added 100) and the real value is 17? 
on 2.4.9 I just made these screencaptures, after only enabling Lat/Lon and Sats in the display and received tlm. 
[Jeti_Tx_Screencap.zip](https://github.com/mstrens/oXs_on_RP2040/files/11540941/Jeti_Tx_Screencap.zip)
Screen 005.bmp shows my real coordinates.)

mstrens, 2023-05-23:
Indeed : when number of sat is greater than 100, it means that GPS has a fix. Number of sat is then the last 2 digits.
It is strange that the format for Latitude change from GPS to date to degree.
I must have a bug somewhere but I do not yet know where.
Anyway, this fact will help.

KeesBlokland, 2023-05-23:
Just so we do the same things:  (and again thanks to all participating in this adventure. Things are moving so fast I hardly have time for a cup of coffee!)  

> In the video, I see that LON is also jumping. It seems that the other data (alt, speed) are not jumping (they just change slowly). It is also abnormal that the GPS cumulative distance is negative.  
> When you activate FVP, are LAT and LONG also jumping?

All sensors disabled, only GPS active. 

Q: so even though FPV forced all values, the live GPS will replace the one you preloaded? 
I will do another one with one with FVP, and disconnect the output from the GPS, so that the preloaded values are not messed up. 

From what I see it appears the values do not always end up in the correct display field. 

On my terminal I get:
`Cmd to execute:   FVP

GPS Latitude = 89.1234560 degree
GPS Longitude = 178.1234560 degree
GPS Groundspeed = 2468 cm/s
GPS Heading = 179.120000 degree
GPS Altitude = 135721 cm
GPS Num sat. = 23
GPS Date J M A = 16 4 23 
GPS Time H M S = 34 19 49 
GPS Pdop = 123 
GPS Home bearing = 179 degree
GPS Home distance = 234 m
Volt 1 = 7321 mVolt
Current (Volt 2) = 89321 mA
Volt 3 = 15321 mVolt
Volt 4 = 14321 mVolt
Capacity (using current) = 34321 mAh
Temp 1 (Volt 3) = 136 degree
Temp 2 (Volt 4) = 148 degree
Vspeed = 258 cm/s
Baro Rel altitude = 246821 cm
Pitch = 89 degree
Roll = 78 degree
Yaw = 67 degree
RPM = 369 Hertz
Ads 1 1 = 11111 mVolt
Ads 1 2 = 12121 mVolt
Ads 1 3 = 13131 mVolt
Ads 1 4 = 13141 mVolt
Ads 2 1 = 21212 mVolt
Ads 2 2 = 22222 mVolt
Ads 2 3 = 23232 mVolt
Ads 2 4 = 24242 mVolt
Airspeed = 15151 cm/s
Compensated Vspeed = 167 cm/s
Sbus hold counter = 98
Sbus failsafe counter = 12
Gps cumulative distance = 135791
Vspeed compensation = 1.10
Internal telemetry fields are now filled with POSITIVE dummy values
To get real values again, you have to power down
lat orig=351f2907  jeti=591cef
Frame= 19  ef  1c  59  0  29  ef  1c  b2  20  31  78  23  41  0  0  51  4d  5  61  17  0  1  a1  dc  42  d4  e4  22  40  e4  11  86  0  b1  d9  5b  c1  f1  57  f1  88  0  1  10  94  0  4  12  6a  60  20  1  11  2  41  1  13  59  0  1  14  4e  0  4  16  7c  56  0  1  17  21  2  1  18  a7  40 
`
There is no other output after this line, until I stop and start monitoring. 
Display on the Tx is what I see after a cold start of the RP2040, just watching the display. (video's are too big, hope this works too. )

[Tx_screen_after_powerdown_all.zip](https://github.com/mstrens/oXs_on_RP2040/files/11541254/Tx_screen_after_powerdown_all.zip)

mstrens, 2023-05-23:
I found a bug in a table defining the length of the GPS cum distance.
Version 2.4.11 is on github.
Perhaps it help.

KeesBlokland, 2023-05-23:
It surely is: Stable display with 14 sats at present! Cumulative probably adds up all the little random movements, so it will be a while before it shows anything. I'll start making a sensor for airborne tests!

I'll start adding more sensors again. 

Thanks again!

KeesBlokland, 2023-05-23:
2.14.12: So far so good!

mstrens, 2023-05-23:
OK. 
If it is confirmed that it works, I will later on remove the remaining debug msg about exbus frames being generated

Klippeneck, 2023-05-23:
Hello mstrens,
SW 2.4.12 looks good, all telemetry values seem OK.
Only GPS Heading I could not test, I will do tomorrow in flight if the weather plays along.
Jürgen 
[image]

mstrens, 2023-05-25:
I found the bug for cumulative dist. It was 100 too small.
It is fixed in 2.5.1 (in test)

KeesBlokland, 2023-05-25:
Thanks!, I'll try. 
Found another small one: 
I just happen to have a bunch of BMP280's, it is not visible on the Transmitter, but is seen when I do a FV.
(for my application it is a backup for the GPS altitude)
 
I think the issue is in this bit: Baro1 is selected two times, probably must be **baro3.baroinstalled**. 

setupExbusList(..
`if (( ( ms4525.airspeedInstalled || sdp3x.airspeedInstalled) && ( baro1.baroInstalled || baro2.baroInstalled || baro1.baroInstalled) )  ||  activateAllFields) {
        exbusFieldList[exbusMaxFields++] = AIRSPEED_COMPENSATED_VSPEED ; 
    }
`
I also ran into the problem that i can no longer make a good uf2, get loads of exbus errors. Looking into that now.

KeesBlokland, 2023-05-25:
Q: How much work would it be, if it is at all possible, to add another I2C port? 

This is probably something for the 'oh, would it not be nice' list.

Satcomix, 2023-05-25:
> `Q:` How much work would it be, if it is at all possible, to add another I2C port?
> 
> This is probably something for the 'oh, would it not be nice' list.

Hello Kees,
I2C is a bus system, why another connection? 
All modules work on the same bus.
If you want to swap the I2C bus, from I2C1 to I2C0 it's easy.
In VsCode in the tab: replace in files simply enter i2c1 and replace with i2c0. After that just do a build, done.
Greeting,
Torsten

mstrens, 2023-05-25:
I fix the bug about detecting BMP280. The bug was there in the 2 jeti protocols.
New version is on github.

Indeed why do you want another I2C bus?

KeesBlokland, 2023-05-25:
> I fix the bug about detecting BMP280. The bug was there in the 2 jeti protocols. New version is on github.
> 
> Indeed why do you want another I2C bus?

..lol, you are right, I did not have my proper coffee during the morning startup procedure!!

KeesBlokland, 2023-05-26:
> The uf2 file compiled by oneself is still abnormal in Exbus mode. https://www.rcgroups.com/forums/showpost.php?p=51022293&postcount=1269

Just something to try? (Clean reconfigure, clean rebuild, clean)

[image]

Satcomix, 2023-05-26:
> > The uf2 file compiled by oneself is still abnormal in Exbus mode. https://www.rcgroups.com/forums/showpost.php?p=51022293&postcount=1269
> 
> Just something to try? (Clean reconfigure, clean rebuild, clean)
> 
> [image]

Hello
Mstrens,
aepub,
Kees,
I'm starting to think that an update of Windows or Pico Visual Studio Code is responsible for the problems, because I also had problems this morning and first deleted all build folders.
I have now installed everything again and created a build of version 2.5.2-test, which you could **please check** to see whether it works with your JETI system. On S.PORT and FBUS i have no problems with my own build.
br,
Torsten
[oXs.zip](https://github.com/mstrens/oXs_on_RP2040/files/11573889/oXs.zip)

Klippeneck, 2023-05-26:
Hello mstrens,
thanks for the change regarding Cum Dist., now it is correct.
During today's test flight I noticed that the telemetry value GPS Heading was constantly set to 0. 
1 hour flight, 33609 Items, but it is always zero. This seems to be another bug.

mstrens, 2023-05-26:
There was indeed a bug for GPS heading in exbus.
I hope I fix it now (loaded on github)

Klippeneck, 2023-05-26:
Hello mstrens,
With SW 2.5.3, the sensor no longer brings telemetry values, registration in the transmitter works.

mstrens, 2023-05-26:
I changed an option back in cmakelist.txt about getting compilation warning.
It is on github (2.5.4).
Not sure if this help.

Satcomix, 2023-05-26:
> I changed an option back in cmakelist.txt about getting compilation warning. It is on github (2.5.4). Not sure if this help.

Hello Mstrens,
I did the 2.5.2 build in debug mode in VScode and it worked with JETI the same way as with FBUS and S.PORT. So why does something have to be changed now in cmakelist.txt, it worked all the months.
Greeting,
Torsten

mstrens, 2023-05-26:
When I compiled, I did not get the warnings.
I suspected that some compilation warnings should have been taken into account to solve the issue that some members had when they compiled them self.
So I first made a change in cmakelist.txt in order to activate the warnings (I added a line "add_compile_options(-Wall)").
I had a look at some warnings and fixed some of them (about unused variables). It was version 2.5.3
It seems that the telemetry fields are not provided anymore in exbus.
So I made now 2.5.4 where I remove the line for warnings hoping it is OK again.

Klippeneck, 2023-05-26:
Hello mstrens,
both 2.5.3 and 2.5.4 do not come up with telemetry values.
Version 2.5.2 was the best working version so far, except that GPS haeding was missing.
Jürgen

mstrens, 2023-05-26:
I now put 2.5.5 where I reactivated an unused variable in exbus.
I do not know if this will help.

If it does not help could you make a capture with the logic analyser with this 2.5.5 version.

KeesBlokland, 2023-05-27:
ai, yes, something is broken. 
2.5.3 is for me the last one that gives me GPS. Did not spot that, because I was working on roll&pitch in the workshop where I do not have gps reception. I was just looking at the output of the serial monitor. 
Me bad ;-(
I'm going to build another setup to put in my plane, so that I can use the testing version on the workbench again. Don't know if I manage that this weekend though!

I will try to get a capture later.  
Have a good weekend everybody, the sun is shining, looks like lots of flying today!

Kees

[19 more replies]

closed 2023-12-06 by Klippeneck