mstrens/oXs_on_RP2040 #81: Pitch and Roll to PWM channel funktion

Satcomix, 2023-05-16:
Hello Mstrens,
Ver.2.3.10
VScode config.h
#define PITCH_CONTROL_CHANNEL 16
#define PITCH_RATIO_CHANNEL 6  
#define PITCH_RATIO  100  
#define PITCH_MAX 100  
#define PITCH_MIN -100 
#define ROLL_CONTROL_CHANNEL 15
#define ROLL_RATIO_CHANNEL 5 
#define ROLL_RATIO  100 
#define ROLL_MAX 100 
#define ROLL_MIN -100 

Version = 2.3.10 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    4  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255    9   10
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Fbus(Frsky)

Cmd to execute:   PWM
PWM values us (sbus) 1... 8   1500(  992)  1500(  992)  1500(  992)  1500(  992)  2012( 1811)  2012( 1811)  1500(  992)  1500(  992)
PWM values us (sbus) 9...16   1500(  992)  1500(  992)  1500(  992)  1500(  992)  1500(  992)  1500(  992)  1500(  992)  1500(  992)

Channel 15 =GPIO 9 for Pitch with channel 5 for adjust from Transmitter Poti1(Ch5) =+100%
Channel 16=GPIO10 for Roll with channel 6 for adjust from Transmitter Poti2 (Ch6) = +100%
Have I misconfigured or understood something?
br,
Torsten

mstrens, 2023-05-16:
This seems me ok except that Channel 15 = GPIO 9 would be used for Roll (and not pitch as you said)
Same error for channel 16 (would be used for pitch )

Satcomix, 2023-05-16:
> This seems me ok except that Channel 15 = GPIO 9 would be used for Roll (and not pitch as you said) Same error for channel 16 (would be used for pitch )

Hello Mstrens,
I swapped the two terms roll and pitch
but it shouldn't matter for the function, right?
When I move the MP6050 I get no movement on GPIO9(Ch15) and 10(Ch16).
I will make some more tests.
br,
Torsten

mstrens, 2023-05-16:
I have perhaps some more bugs.
If you change the value sent by the handset (without moving the mp6065) does the servo move?

pierrotm777, 2023-05-16:
As mstrens said change Roll channel to 14, not 15 .

Satcomix, 2023-05-16:
> Did you have done a MPUCAL ?

Hello Pierre,
with this MP6050 I do not have to carry out an MPUCAL. It is factory calibrated to zero.

Satcomix, 2023-05-16:
0 1500 1500 -1 -1
0 1500 1500 -1 -1
0 1500 1500 -1 -1
0 1500 1500 -1 -1
0 1500 1500 -1 -1
0 1500 1500 -1 -1
the MP6050 has full function, i make some tests with debug output!

Satcomix, 2023-05-16:
0 1500 1500 992.000000 0.000000
0 1500 1500 992.000000 0.000000
0 1500 1500 992.000000 0.000000
0 1500 1500 992.000000 0.000000
0 1500 1500 992.000000 0.000000
0 1500 1500 992.000000 0.000000

VScode config.h
#define PITCH_CONTROL_CHANNEL 16
#define PITCH_RATIO_CHANNEL 15
#define PITCH_RATIO 100
#define PITCH_MAX 100
#define PITCH_MIN -100
#define ROLL_CONTROL_CHANNEL 14
#define ROLL_RATIO_CHANNEL 13
#define ROLL_RATIO 100
#define ROLL_MAX 100
#define ROLL_MIN -100

Channel13 = Poti Roll
Channel15 = Poti Pitch

0 changes as soon as I move the MPU6050.
992 changes as soon as I turn Poti Pitch from 172 to 1811
for Poti Roll i can see no function in debug message.
br,
Torsten

mstrens, 2023-05-16:
currently there is no debug for roll (because it is just a copy/paste); I first try to solve pitch issues.
The last value in msg should change from -200 up to +200 when the pot for pitch ratio change
As long the last value is 0 , it is "logical" that there is no compensation.
I will look at the code again.

Satcomix, 2023-05-16:
In a previous version, the 1500us changed when I moved the MP6050, but not in this version. It always stayed at 1500us

mstrens, 2023-05-16:
It is because the calculated ratio (based on the channel) was always 0 (bug).
There is a new version to fix this.

pierrotm777, 2023-05-16:
Pitch is again OK !

pierrotm777, 2023-05-16:
Ratio go from 199 to -199

Satcomix, 2023-05-16:
Hello Mstrens,
The pitch_control and the pitch_ratio works well!
Very good function!
br,
Torsten

Satcomix, 2023-05-19:
Hello Mstrens,
I still have a question about the Pitch_Ratio and Roll_Ratio, which is controlled by the transmitter with -100% ---0%--- +100% via two PWM channels.
What value does the ratio refer to?
1. on the angle value coming from the MP6050,-90degree-+90degree?
2. on the PWM value at the GPIO port,988-2012?
3. to an internal value in the oXs script,172-1811?
I was assuming that with a value of +50%(exact+48.8%) ratio, the PWM value should only go up to about 1750us, or with -50%(exact-48.8%) ratio it would be about 1250us, but the values ​​go from about 1000 to 2000.
br,
Torsten

mstrens, 2023-05-19:
the ratio is a parameter that allows to take care of the mechanical.
Imagine e.g. that when the plane is horizontal and the servo is at neutral position (PWM = 1500 usec), the camera point to the horizon.
No imagine that your servo moves 60° when PWM signal is about 2000 us (+100%).
Depending on your mechanical, it could be that, when the servo moves by 60°, the camera points in a direction that is only e.g. 30° above the horizon. 
This means that if the TX channel sent a PWM 1500us (neutral) but the plane has the nose down by 30°, oXs has to move the servo in such a way that the camera points 30° up and so oXs must generate a PWM of about 2000 us (500 us more than the value it get from the Tx channel).
The ratio allows to set up the relation between  pitch (or roll) and the PWM correction to apply.

Satcomix, 2023-05-19:
Hello Mstrens,
Thanks for the detailed answer.
The last sentence explained everything, it is the ratio between the sent PWM signal (e.g. CH16) and the correction PWM signal to be assumed (MP6050 to CH16).
Another question:
Where can I find the **physical IDs**, of the individual telemetry groups(GPS, Baro,...) , of the oXs sensor in the script, or does oXs use the same IDs as FrSky, e.g. GPS ID4=0x83
I know the **application ID**s.
Thanks very much,
Greeting,
Torsten

mstrens, 2023-05-19:
OXs on RP2040 uses only one physical ID.
It is defined in config.h in this line.
#define SPORT_DEVICEID    DATA_ID_FAS  // this line defines the physical ID used by sport.

Satcomix, 2023-05-19:
Hello Mstrens,
The only one I've found before is:
#define SPORTDEVICEID **0xE4**
And FAS is 0x22 ???
A list of telemetry ID`s I found:
// Default sensor data IDs (Physical IDs + CRC)
#define DATA_ID_VARIO 0x00 // 0
#define DATA_ID_FLVSS 0xA1 // 1
#define DATA_ID_FAS 0x22 // 2
#define DATA_ID_GPS 0x83 // 3
#define DATA_ID_RPM **0xE4** // 4
#define DATA_ID_SP2UH 0x45 // 5
#define DATA_ID_SP2UR 0xC6 // 6

Thank you,
br,
Torsten