mstrens/oXs_on_RP2040 #83: not shure where to ask this not realy an issue

secretman28, 2023-06-11:
can you add a discution tab for talk without going here ?
i think it could be usefull 

my queestiion is the # we setup in the config is the pin number or the gpio? 
i got it on the pico but not seam to do anything

tank's

mstrens, 2023-06-11:
You can also put your questions on RCGroups in this topic.
https://www.rcgroups.com/forums/showthread.php?4130529-openXsensor-on-RP2040-board

In the file config.h there is no # to define pins.
You have to select the gpio's being used via a USB/serial terminal once the program has been flashed.
The numbers refers to GPIO's and not pin number.
It is via the same way that you have to select the protocol being used (and most common parameters)

secretman28, 2023-06-11:
ok tanks i didint know it has a rcgroup thread . 
ok for the gpio and not pin# 
 
yes all work with serial #1 i will look more in to it tonight tanks for your fast answer

secretman28, 2023-06-12:
omg, all is good now, my bad, i use a elrs rx that was in sbus output, no wonder why it got messup my head, sry for all 
all work perfect 16ch :) TANKS TO DE DEV !!

closed 2023-06-12 by secretman28