mstrens/oXs_on_RP2040 #85: negative values of Current

aeropic, 2023-06-16:
Current is computed with a formula like:
Curr = V*SCALEx - OFFSETx

The current sensors output a Vcc/2 value when 0 A is flowing. Basically OFFSETx compensate this Vcc/2. But there could be some noise at sensor level leading to negative values of computed current.

The S-PORT protocol does not accept negative values (at least for Current telemetry). Under EDGE-TX negative values are displayed with a huge positive value...

To fix this,  I made a modification of "voltage.cpp" source file to clamp negative values to zero in the getVoltages function (the file is attached) :
...
if (mVoltPerStep[cntInit] !=0) {
value = ( ((float) sumVoltage[cntInit]) / (( float) SUM_COUNT_MAX_VOLTAGE) * mVoltPerStep[cntInit]) - offset[cntInit];
// Volt3 and Volt 4 can be used as temperature or voltage depending on value of config.temperature
// volt 2 is used for current and consumed capacity is then calculated too

// as negative currents are not allowed in S-PORT keep only positive values if cntInit = 1 (current)
if (cntInit == 1) {
   if (value < 0) value = 0;
}
...

I tested this and it works like a charm.
Please feel free to introduce this change in a future release.

And thanks a lot for this excellent project.

Regards
Alain
[src.zip](https://github.com/mstrens/oXs_on_RP2040/files/11767369/src.zip)

mstrens, 2023-06-16:
I made a correction on github in version 2.5.9 in order to avoid having negative values for current in Sport and Fbus protocols.
Still I did not made the change in voltage.cpp but only in sport.cpp and fbus.cpp.
So when you use the command FV , you can still negative values that let you know that you have probably to adjust the offset.

aeropic, 2023-06-16:
Oops I'm in 2.5.0...
I'll download the 2.5.9. Indeed it is better to keep the change at telemetry level, it's ggod to see the raw value with FV. That's perfect thanks

aeropic, 2023-06-16:
I'm not familiar with github, where is located the 2.5.9 ? When I download the project I get the 2.5.0...

mstrens, 2023-06-16:
In github you can have several branches.
Per default you get "Main".
There is a drop list and you can select other branches (when they exist).
Here a direct link
https://github.com/mstrens/oXs_on_RP2040/tree/test

aeropic, 2023-06-16:
thanks a lot, I downloaded and tested it: working perfectly.
Many many thanks for your reactivity and the quality of this project! It's awesome...

closed 2023-07-01 by aeropic