mstrens/oXs_on_RP2040 #86: Not an issue, just a request for more info

Billy60, 2023-06-19:
Hi Michel
I have made a few Arduino based penXvarios with some success. However, This version seems more simple to tune but could you please show a 'miki' kind of page ?
* I need only RPM and Temp telemetry readings. it seems the device could do it but in your telemetry list of signals they don't appear...
* What are the values of the peripherals: the Temp sensor should be : NTC ? of what value ? PTC ? special chip ?
* the inputs pins will be provided with PWM signals  or  positive pulses (PWM behind a capacitor) ?
Thanks a lot

mstrens, 2023-06-19:
The list of fields depends on the protocol being activated.
It is explained in a file doc/fields_per_protocol.txt

For temperature you need a chip that produce a voltage based on the temperature like the TMP36.
The relation between temperature and voltage has to be linear. So a NTC is not good.
You can connect 1 or 2 chips (on Volt3 and Volt4 pins). You have to declare (via a USB command) the number or temperature to be measured. You also have to adjust scale and offset (for volt 3 /volt 4) according the chip you use).

For RPM, oXs has to get pulses. The voltage of the pulse has to vary from 0V (min) to 3.3V (max). If you have a module that already provide such pulses it is ok to use it. If not it is safe to make some filter with a capacitor, one resistor and 2 diodes (to avoid having to low or to high voltage on the pin)

Billy60, 2023-06-20:
Thank you Michel,
The RPM pulses will enter on which type of pin ? would it be the PWM one ?
Have a good day

mstrens, 2023-06-20:
RPM is not PWM related.
You can choose the pin you want for RPM.
You do it entering a command via USB/serial just like you must do to select the protocol (or other pin functions).
Defining a pin for RPM ( with another value than 255) means that oXs will define it as input, will count the pulses and will transmit a telemetry field with RPM.

Billy60, 2023-06-20:
Great, you are the best ! 
Thanks a lot for all your help and for those fantastic projects.

Billy60, 2023-06-27:
Good morning Michel
I have uploaded your oxs.uf2 file and do get the green led flashing nicely.
Hover whenever I try to get into it with Thonny, I get the following message:
WARNING:root:Unexpected echo. Expected b'%Run -c $EDITOR_CONTENT\r\n', got b'processing cmd\r\n\r\nCmd to '
What is this error ?

Thanks a lot !

Billy60, 2023-06-27:
Thanks Michel, however I still only want to use the device as it is and would never modify it.
Only, I don't get into the normal configuration of the pin numbers etc. I only get this error message.
Do I have to go through all this hardly understandable sdk install and all those steps which are far beyond my abilities ?
I would be very happy to have it run the simple way !

mstrens, 2023-06-27:
To upload the program into the RP2040, you just have to 
- press the boot button while you connect the rp2040 to the PC via USB (this will enter it bootloader mode)
- a usb drive appear on your pc
- do a drag and drop of oXs.uf2 file to this drive.

The program will be automatically uploaded and flashed.

Then you have to adapt the setup to you need. 
Without pressing the boot button, connect the RP2040 to the PC via USB.
Start on your pc a serial terminal program. It can be e.g. the one from arduino IDE or another one like candle.
Set the com port to 115200 baud and send commands to the serial port.
Press e.g. 
? + enter (question mark and ENTER)
The Pc should display the full list of command that are available.
E.g. to change the protocol enter : protocol=E
You can define the pins being (for baro, GPS, ...) used in the way.

Billy60, 2023-06-27:
Thank you a lot Michel
I had done all those steps properly but was having a hard time with
Terminal softwares.
The only one which worked was Termite, all the Powershell, Arduino one,
TerraTerm, Thonny did not work.
And I still don't know why ...
But now it is flying, thank you for your developments and help !!

very best regards

Guillaume

closed 2023-11-21 by mstrens