mstrens/oXs_on_RP2040 #87: Digital output instead of PWM

onki69, 2023-06-20:
Is there a possibility to have a digital output instead of an PWM output.
For instance a puls width (coming from the bus) below 1500µs corresponds to digital 0 while anything above this is a digital 1.

Might be useful for function modellers to switch light or other non PWM based stuff.

Satcomix, 2023-06-20:
Hello Onki,
Hello Mstrens,
I think there are enough other ways to switch a function via one of the 16 PWM outputs, since the current carrying capacity of an RP2040 is not very high either.
Maybe you are interested in a system developed by me from 6 to 24-channel PWM / or digital-IO / or PWM-sequences based on Pololu Mini Maestro controllable via a switch widget in Ethos.
Best regards,
Torsten

mstrens, 2023-06-22:
Did you already had a look at the "sequencer" in openXsensor" on arduino pro mini.
This allows to use a channel to drive some digital pins.
It also allows to define sequences of switching of the pins (e.g. to do led blinking) based on the channel.
It is quite flexible but requires to configure some parameters.
If there is interest, I could look to implement this functionality into oXs on RP2040

pierrotm777, 2023-07-04:
Add a sequencer into the RP2040 would be a very nice feature.

mstrens, 2023-09-18:
Sequencer has been added to oXs

mstrens, 2023-10-07:
Sequencer seems to be OK  (in test branch currently)

closed 2023-10-07 by mstrens