mstrens/oXs_on_RP2040 #89: PWM output with FS-IA6B I-BUS protocol

aeropic, 2023-07-04:
Hi,
I'm not sure how to set upt hte configuration of OxS RP2040 and FlySky FS-IA6B.
I already made it work with telemetry configured on the letf "SENS" port of the Rx conected to pin 0 of RP2040.
I connected the right port "SERVO" to pin 9 as PRImary input of RP2040.

Now, having set my TX (TX16S) in mode PWM+IBUS, I do not see any change in PWM outputs (C1=8) PWM remains at (881 (...0)) .

What am I doing wrong ? Has the RP2040 already been tested with this RX for servos functions ?

thanks
Alain

mstrens, 2023-07-04:
When IBUS protocol is selected, oXs expect:
- to get Rc channels on PRI (or SEC) pin with a Sbus signal (so not a Ibus)
- to transmit the telemetry data on the TLM pin using the Ibus specifications (oXs reply to request from t receiver on this pin)

Are you sure that your receiver is generating a Sbus signal?
I do not have Flysky hardware so I can't test it.
Normally Sbus decoding to PWM has been tested and works.

aeropic, 2023-07-04:
Ah OK, more complex set up I thought :-)
I'm not sure my FS-IA6B is able to generate SBUS even if the configuration of the TX (TX16S 4in1 module) allows to select either
- PWM + IBUS
- PWM+ SBUS
I tried with both modes and it does not work.
from this page it seems SBUS is OK... https://www.rcgroups.com/forums/showthread.php?2853042-iBus-and-sBus-Compatible

I successfully hooked an arduino nano to the IBUS "servo" port of my RX with the TX set in IBUS too and was able to receive "IBUS servo packets" with a very simple code (I join it)
[10CHReceiver.zip](https://github.com/mstrens/oXs_on_RP2040/files/11947325/10CHReceiver.zip)

aeropic, 2023-07-04:
What is fun is that when setting my TX in PWM+SBUS instead of PWM+IBUS, the above arduino code is still working fine, I get the correct servo values following the sticks inputs...

mstrens, 2023-07-04:
I had a short look at the arduino code and it seems that it decodes only Ibus rc channels.
It does not have code to read Sbus signal.
So if you still get correct servo values, it means that you receiver is not generating Sbus signal but still Ibus when you change the setting on the TX.
I do not know why.

aeropic, 2023-07-04:
That's weird indeed that the Rx does not outpu SBUS when the Tx is supposed to be configured in this mode.

Would it be difficult to port the arduino code to RP, I can debug ? I tried to open SBUS.cpp but I got quickly lost ;-)

pierrotm777, 2023-07-04:
Is it possible that the receiver don't invert the Sbus signal as frsky does it on it's receivers?

mstrens, 2023-07-04:
If the Ibus arduino project (see above) decodes the signal, it means that the signal is not a Sbus signal (inverted or not).
Sbus has many more differences compared to Ibus.

mstrens, 2023-07-04:
Do you know if it is possible to get 2 Ibus signals (on 2 different pins)  on a Flysky receiver.
If so, I can keep TLM pin to manage telemetry and then use PRI pin to decode the Rc channels.
This would require less changes.

Otherwise (if there is only one pin to manage Ibus on rx), I would have to change more code (using a logic like for Fbus (frsky) or Exbus (jeti).  

I did not find on the web a full documentation about Ibus (combining Rc channels frames and telemetry frames).
Do you know a link to such a doc.

pierrotm777, 2023-07-04:
I have a fs-ia6b, I shall try the Sbus output as soon as possible.

pierrotm777, 2023-07-04:
May be this lib?
https://github.com/bmellink/IBusBM

aeropic, 2023-07-04:
> Do you know if it is possible to get 2 Ibus signals (on 2 different pins) on a Flysky receiver. If so, I can keep TLM pin to manage telemetry and then use PRI pin to decode the Rc channels. This would require less changes.

Yes sure, FS-IA6B or FS-IA10B Rx get two ports, one for telemetry sensors called "SENS", one for servos serial called "SERVO"
> 
>
> I did not find on the web a full documentation about Ibus (combining Rc channels frames and telemetry frames). Do you know a link to such a doc.

Same for me... But I found the arduino code which shows the serial protocol used for IBUS servo part and does not seem to much complex ! My feeling is that telemetry is clearly separated from servos but I'm not sure. I'm sure that :
- OXS works fine for telemetry part (TLM pin connected on the left connector)
- OXS does not work (for me) on the right connector neither in IBUS nor SBUS whatever is present on the left connector (I tried with or without TLM declared and connected)
- the arduino code works from the right SERVO connector, TX16S set in IBUS or SBUS.

[image]

aeropic, 2023-07-04:
In the library link found by Pierrot I read this : The TGY-IA6B has 2 iBUS pins: one for the servos (only output) and one for the sensors/telemetry (which uses a half-duplex protocol to switch between output and input to poll for sensor data).

This confirms that the 2 ports are available and independant one of the other !

mstrens, 2023-07-04:
I read it too now.
I started writing some code to manage Rc channels form Ibus on a separate pin (so keeping telemetry on another pin)

aeropic, 2023-07-04:
Whooahoo you're fast. Ready to test :-)
Please keep me informed !

aeropic, 2023-07-04:
You're very fast thanks for your support.
I tested it and I have the feeling it's not (yet) working.

What I did:
- tested servo in PWM+IBUS : no value change afte command PWM
- tested servo in PWM+SBUS : idem
- tested TLM + SERVO in IBUS : TLM is still working but nothing at servo level.

Could it be a hardware issue ? Fried port on RP2040? (the Rx is OK and still working with arduino)

Note that when adding TLM=0 in the configuration I received a set of:
sbusQueue try add error
sbusQueue try add error
sbusQueue try add error
sbusQueue try add error
sbusQueue try add error
sbusQueue try add error
sbusQueue try add error
sbusQueue try add error
sbusQueue try add error
sbusQueue try add error
sbusQueue try add error
sbusQueue try add error
sbusQueue try add error
sbusQueue try add error

Version = 2.6.2 
sbusQueue try add error
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
sbusQueue try add error
sbusQueue try add error
Primary channels input    =    9  (PRI     = 5, 9, 21, 25)
sbusQueue try add error
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
sbusQueue try add error
sbusQueue try add error
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
sbusQueue try add error
GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
sbusQueue try add error
GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
sbusQueue try add error
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ...,

final config
=====
Cmd to execute:   

Version = 2.6.2 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    9  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =  255  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =  255  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =    8  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27  255  255 (V1 / V4 = 26, 27, 28, 29)

Protocol is ibus(Flysky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , -15.150000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , -38400.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is not detected
Acc/Gyro is not detected
Airspeed sensor is not detected
    No Vspeed compensation channel defined; oXs uses default settings
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox (configured by oXs) :GPS is not (yet) detected
Led color is normal (not inverted)
Failsafe type is HOLD

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
processing cmd

Cmd to execute:   PWM
PWM values us (sbus) 1... 8    881(    0)   881(    0)   881(    0)   881(    0)   881(    0)   881(    0)   881(    0)   881(    0)
PWM values us (sbus) 9...16    881(    0)   881(    0)   881(    0)   881(    0)   881(    0)   881(    0)   881(    0)   881(    0)

mstrens, 2023-07-04:
I do not  see immediately the bug.
I added some more debug messages.
Please try version 2.6.3 to see the messages

aeropic, 2023-07-04:
quite verbose :-)

I get this log
=======
t= 5
get= 5
put= DC
get= DC
put= 5
get= 5
get= DC
get= 5
get= DC
put= DC
get= 5
put= 5
get= DC
get= 5
get= DC
put= 43
get= 5
put= F3
get= 43
get= F3
put= 8020
put= 40
get= 8020
put= 5
get= 40
get= E8
get= 5
put= 5
get= E6
put= DC
get= 5
get= DC
get= 3
put= DC
put= 5
get= DC
get= 5
put= 5
get= C6
put= DC
get= 5
put= 5
get= DC
put= EE
get= 5
get= EE
get= 5
put= 3
get= DC
get= 3
get= DC
get= 5
put= DC
put= 5
put= DC
get= DC
get= 5
put= DC
get= DC
get= 5
get= DC
get= 5
get= DC
get= 5
put= 5
get= DC
[... 1067 lines cut ...]
put= DC
get= 3
put= 5
get= DC
put= C6
get= 5
put= 5
get= C6
get= 5
get= DC
get= 5
put= EE
put= 5
get= EE
get= 5
put= 3
get= DC
put= DC
get= 3
put= 5
get= DC
get= 5
get= DC
put= 5
get= 5

mstrens, 2023-07-04:
I put other messages (version 2.6.4)
It is supposed to display the PWM values contained in the Rc channel frame

aeropic, 2023-07-04:
weird, I do not get any log...
Even after RETURN
===========
Version = 2.6.4 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    9  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =  255  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =  255  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =    8  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27  255  255 (V1 / V4 = 26, 27, 28, 29)

Protocol is ibus(Flysky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , -15.150000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , -38400.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is not detected
Acc/Gyro is not detected
Airspeed sensor is not detected
    No Vspeed compensation channel defined; oXs uses default settings
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox (configured by oXs) :GPS is not (yet) detected
Led color is normal (not inverted)
Failsafe type is HOLD

Config parameters are OK
Press ? + Enter to get help about the commands

 >>

mstrens, 2023-07-04:
I printed the message only if a check sum was OK
I now made a version 2.6.5 where I print before checking the checksum.
I hope it will help

aeropic, 2023-07-04:
cool, frames are received. I placed logs into the txt file. I tried to move some sticks to see values changing...
[log.txt](https://github.com/mstrens/oXs_on_RP2040/files/11952268/log.txt)

aeropic, 2023-07-04:
I've to stop now, resuming tomorrow !!!!
Good night

aeropic, 2023-07-05:
The checksum is indeed blocking the RcChannels decoding. I suppressed the checksum verification:
 // >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
            if  ( runningIbusFrame[1] = 0X40 )    // byte 1 must be 0X40 for a RC channel frame
                  { 
                ibusDecodeRcChannels();
            }
I made a test this morning and I get this :
Frame= 20 40 e8 5 e6 5 dc 3 dc 5 c6 5 dc 5 ee 5 dc 3 dc 5 dc 5 dc 5 dc 5 dc 5 dc 5 43 f3 
Rc= 1512 58885 1510 56325 988 56323 1500 50693 1478 56325 1500 60933 1518 56325 988 56323 
Frame= 20 40 e8 5 e6 5 dc 3 dc 5 c6 5 dc 5 ee 5 dc 3 dc 5 dc 5 dc 5 dc 5 dc 5 dc 5 43 f3 
Rc= 1512 58885 1510 56325 988 56323 1500 50693 1478 56325 1500 60933 1518 56325 988 56323 
Frame= 20 40 e8 5 e6 5 dc 3 dc 5 c6 5 dc 5 ee 5 dc 3 dc 5 dc 5 dc 5 dc 5 dc 5 dc 5 43 f3 
Rc= 1512 58885 1510 56325 988 56323 1500 50693 1478 56325 1500 60933 1518 56325 988 56323 
Frame= 20 40 e8 5 e6 5 dc 3 dc 5 c6 5 dc 5 ee 5 dc 3 dc 5 dc 5 dc 5 dc 5 dc 5 dc 5 43 f3 
Rc= 1512 58885 1510 56325 988 56323 1500 50693 1478 56325 1500 60933 1518 56325 988 56323 
Frame= 20 40 e8 5 e6 5 dc 3 dc 5 c6 5 dc 5 ee 5 dc 3 dc 5 dc 5 dc 5 dc 5 dc 5 dc 5 43 f3 
Rc= 1512 58885 1510 56325 988 56323 1500 50693 1478 56325 1500 60933 1518 56325 988 56323 
Frame= 20 40 e8 5 e6 5 dc 3 dc 5 c6 5 dc 5 ee 5 dc 3 dc 5 dc 5 dc 5 dc 5 dc 5 dc 5 43 f3 
Rc= 1512 58885 1510 56325 988 56323 1500 50693 1478 56325 1500 60933 1518 56325 988 56323 
Frame= 20 40 e8 5 e6 5 dc 3 dc 5 c6 5 dc 5 ee 5 dc 3 dc 5 dc 5 dc 5 dc 5 dc 5 dc 5 43 f3 
Rc= 1512 58885 1510 56325 988 56323 1500 50693 1478 56325 1500 60933 1518 56325 988 56323 
Frame= 20 40 e8 5 e6 5 dc 3 dc 5 c6 5 dc 5 ee 5 dc 3 dc 5 dc 5 dc 5 dc 5 dc 5 dc 5 43 f3 
Rc= 1512 58885 1510 56325 988 56323 1500 50693 1478 56325 1500 60933 1518 56325 988 56323 
Frame= 20 40 e8 5 e6 5 dc 3 dc 5 c6 5 dc 5 ee 5 dc 3 dc 5 dc 5 dc 5 dc 5 dc 5 dc 5 43 f3 
Rc= 1512 58885 1510 56325 988 56323 1500 50693 1478 56325 1500 60933 1518 56325 988 56323 
Frame= 20 40 e8 5 e6 5 dc 3 dc 5 c6 5 dc 5 ee 5 dc 3 dc 5 dc 5 dc 5 dc 5 dc 5 dc 5 43 f3 
Rc= 1512 58885 1510 56325 988 56323 1500 50693 1478 56325 1500 60933 1518 56325 988 56323 
Frame= 20 40 e8 5 e6 5 dc 3 dc 5 c6 5 dc 5 ee 5 dc 3 dc 5 dc 5 dc 5 dc 5 dc 5 dc 5 43 f3 
Rc= 1512 58885 1510 56325 988 56323 1500 50693 1478 56325 1500 60933 1518 56325 988 56323 
Frame= 20 40 e8 5 e6 5 dc 3 dc 5 c6 5 dc 5 ee 5 dc 3 dc 5 dc 5 dc 5 dc 5 dc 5 dc 5 43 f3 
Rc= 1512 58885 1510 56325 988 56323 1500 50693 1478 56325 1500 60933 1518 56325 988 56323 
Frame= 20 40 e8 5 e6 5 dc 3 dc 5 c6 5 dc 5 ee 5 dc 3 dc 5 dc 5 dc 5 dc 5 dc 5 dc 5 43 f3 
Rc= 1512 58885 1510 56325 988 56323 1500 50693 1478 56325 1500 60933 1518 56325 988 56323 

one value over 2 seems reflecting a stick position, but moving one stick whatever the stick has some impacts on all values...
I hope this helps !

aeropic, 2023-07-05:
Look it seems the other values (??) are just big endians reversed:
56325 = 0xDC05 while 0x5DC = 1500

Could it be that the checksum too is reversed ?

mstrens, 2023-07-05:
No.
I just forgot to increase low and high variable by 2 instead of 1 at each iteration.
But there where another error. The loop must be 14X instead of 16X.
I will soon put a new version.
I make a kind of simulation first.

mstrens, 2023-07-05:
I put version 2.6.6 on github (test branch)
I think that it is OK at least up to the check about the checksum.

aeropic, 2023-07-05:
Amazing: V266 is working perfectly (nearly)
I get only 14 channels
Rc= 1511 1500 1034 1500 1478 1500 1512 988 988 2011 1500 1500 1500 1500 
Rc= 1511 1500 1034 1500 1478 1500 1512 988 988 2011 1500 1500 1500 1500 
Rc= 1511 1500 1034 1500 1478 1500 1512 988 988 2011 1500 1500 1500 1500 
Rc= 1511 1500 1034 1500 1478 1500 1512 988 988 2011 1500 1500 1500 1500 

All of them working fine

aeropic, 2023-07-05:
I get 14 PWM values
Cmd to execute:   PWM
PWM values us (sbus) 1... 8    991(  177)   991(  177)   991(  177)   991(  177)   991(  177)   991(  177)   991(  177)   988(  172)
PWM values us (sbus) 9...16    988(  172)   994(  182)   991(  177)   991(  177)   991(  177)   994(  181)   881(    0)   881(    0)

I plugged a servo
did C1=8 as my servo pin is 8 and I don't see it moving even if the displayed values are OK.

I think you can remove the debug line, it will be easier to check the PWM

aeropic, 2023-07-05:
I removed the debug print...
the PWM channels don't change with stick motion. They are stuck to this :

Cmd to execute:   PWM
PWM values us (sbus) 1... 8    991(  177)   991(  177)   991(  177)   991(  177)   991(  177)   991(  177)   991(  177)   988(  172)
PWM values us (sbus) 9...16    988(  172)   994(  182)   991(  177)   991(  177)   991(  177)   994(  181)   881(    0)   881(    0)
processing cmd

Cmd to execute:   PWM
PWM values us (sbus) 1... 8    991(  177)   994(  182)   991(  177)   988(  172)   991(  177)   991(  177)   991(  177)   988(  172)
PWM values us (sbus) 9...16    988(  172)   994(  182)   991(  177)   991(  177)   991(  177)   994(  181)   881(    0)   881(    0)

aeropic, 2023-07-05:
If Idebug print ibusRCChannels
//  printf("%i ", (int) temp );
        ibusRcChannels[i] = (uint16_t) ((((float) (temp - TO_PWM_MIN)) * ratioPwmToSbus)+0.5) +  FROM_SBUS_MIN ; // convert in Sbus value 16 bits)
        printf("%i ", (int) ibusRcChannels[i] );

I get a very limited range from 172 to 182

Rc= 177 177 182 177 177 177 177 182 172 182 177 177 177 182 
Rc= 177 177 182 177 177 177 177 182 172 182 177 177 177 182 
Rc= 177 177 182 177 177 177 177 182 172 182 177 177 177 182

mstrens, 2023-07-05:
There is a typo fout in the way ratioPwmToSbus is calculated. There must be a - instead of a /.
I will fix it in a few min

mstrens, 2023-07-05:
version 2.6.7 is uploaded

aeropic, 2023-07-05:
Yeah Michel you did it, the servo is moving !
**Huge thanks**

Please insert too the fix of negative currents in IBUS (if not already done) (see issue #88) so that we will get a perfect I-BUS version.

That's really great, you're awesome

mstrens, 2023-07-05:
The change for the negative currents was already done.
I now just added (for safety) some checks on the Rc channels PWM values received in ibus frame (in order to avoid dummy sbus values during the PWM to Sbus conversion)

aeropic, 2023-07-05:
perfect ! I'll close both issues... thanks again

closed 2023-07-05 by aeropic