mstrens/oXs_on_RP2040 #92: ELRS Serial Terminal Issue (QT Py)

JVideocassetta, 2023-07-17:
I am having some issues getting any output on serial output after configuring the RP2040 board (QT Py). I also have an ELRS receiver attached and connected to a transmitter. 

I have attempted multiple different baud rates, including the 420000 that my ELRS receiver was flashed for. I have not had any luck and was curious about the wiring explained in the ReadMe section. I currently have 5V, GND, RX to TX, and TX to RX. 

I am little bit confused by what is meant by the primary and secondary pin wiring for ELRS below. Is the standard wiring I have above not what should be used for ELRS?

<html>
<body>

protocol | PRI pin is connectected to | SEC pin is connected to | TLM pin is connected to | Comment
-- | -- | -- | -- | --
C(ELRS) | (TX from Rx1) | (TX from Rx2) | (RX from RX1) |  
S(Frsky sport) | (Sbus from Rx1) | (Sbus from Rx2) | (Sport from RX1 or Rx2) | (1)
F(Frsky Fbus) | Fbus from Rx1 | (Sbus from Rx2) | Not used | (2)
J(Jeti ex) | (Sbus from Rx1) | (Sbus from Rx2) | (Ex from Rx1 or Rx2) |  
E(Jeti Exbus) | Exbus from Rx1 | (Sbus from Rx2) | Not used | (2)
H(Hott) | (Sbus from Rx1) | (Sbus from Rx2) | ???(tlm from RX1 or Rx2) | (1)
M(Multiplex) | (Sbus from Rx1) | (Sbus from Rx2) | ???(tlm from RX1 or Rx2) | (1)
I(Flysky Ibus) | (Sbus from Rx1) | (Sbus from Rx2) | ( Ibus from RX1 or Rx2) | (1)
L(Spektrum Srxl2) | Srxl from Rx1 | Not used | Not used | (2)
2(Futaba Sbus2) | Sbus2 from Rx1 | (Sbus2 from Rx2) | Sbus2 from Rx1 via 1Kohm | (3)

</body>
</html>

Thank you for any help!

mstrens, 2023-07-17:
It is possible to connect oXs to 2 receivers. This allows to have redundancy. If one receiver lost the signal from the transmitter but the other receiver still get it, oXs can still continue to generates the signal.
This explain why there are 2 pins (to be defined) named PRI and SEC. 
If you use only one receiver, connect the TX pin from the receiver to a pin that you define as PRI (= primary).
To define a pin as PRI you have to enter a command like PRI=5 to oXs using the USB connection and a serial terminal.
In this example 5 is in fact the number of the GPIO being connected. For PRI, it is possible to use gpio 5, 9, 21 or 25.

The TLM pin is used by oXs to send back telemetry data to the receiver. It has to be connected to the RX pin of the ELRS receiver if you want to get telemetry data. To do so, you have to assign a pin (GPIO number)  to it using a command like TLM=6. For TLM you can use any GPIO of the RP2040 (so any value between 0 and 29).

If you want that oXs generates some signals (one Sbus and/or a PWM for some servos), you have also to define the GPIO's to be used using commands like e.g. SBUS=1 and of C4=7 (to generate a PWM signal on gpio 7 based on channel 4).

Configure your ELRS receiver to use the same baudrate as oXs (or the opposite).

Perhaps can you post here the setup you used for oXs to check if it is OK.
To get it just press ENTER on the serial terminal and make a cut and paste.

JVideocassetta, 2023-07-17:
I appreciate your response, thank you for the clarification.

The current issue I am having right now is that I am not getting any output on terminal after pressing enter. I am using Putty on a Windows machine and I have confirmed the port is correct (COM4 in my case). I'm using the default 8n1 serial settings and have not had any luck with various baud rates including 115200, 9600, 420000, or 400000. As confirmation that I am on the correct port, when connected to the terminal (no output), resetting the RP2040 board causes my terminal to lose connection and I need to reopen it after reacquiring the connection.

There did not appear to be any complications when configuring the RP2040 board with the oXs.uf2 file.

mstrens, 2023-07-17:
I do not know why but I also had some issues with some serial terminal program.
can you try the serial terminal that is part of arduino IDE. I already used it.
A good alternative (I use this currently) is the serial monitor extension in vscode. 
Baudrate is 115200.
8N1 is OK.

JVideocassetta, 2023-07-18:
That turned out to be successful, thanks for the advice! I first attempted picocom on a linux machine and had the same issue, but I was able to get the arduino IDE to work. 

I am now starting to play with some of the settings and I already have a PWM channel outputting and coming in on a raspberry PI, thank you! Software is super easy to use and saved me a lot of time!

For the PWM channels (I haven't tried it), would it be possible to assign these channels to pins 16-29? The RP2040 board I use is a little bit limited and does not have access to pins 0-2 or 7-15 (more or less).

```
Cmd to execute: 

Version = 2.5.0 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =   20  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =  255  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =  255  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =    3  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255  255  255 (V1 / V4 = 26, 27, 28, 29)

Protocol is CRSF (=ELRS)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is not detected
Acc/Gyro is not detected
Airspeed sensor is not detected
    No Vspeed compensation channel defined; oXs uses default settings
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox (configured by oXs) :GPS is not (yet) detected
Led color is normal (not inverted)
Failsafe type is HOLD

Config parameters are OK
Press ? + Enter to get help about the commands
```

mstrens, 2023-07-18:
RP2040 has 8 internal PWM modules (0, 1, ...7); each module has 2 submodule (A, B). Each can generate a PWM signal.
So there is a maxl of 16 PWM signals.
In the datasheet, they a named 0A, 0B, 1A, 1B, ...., 7A, 7B
Each signal can only be routed to one of 2 pins
Here an extract of the datasheet (page 525); sorry but the formatting is lost and make it difficult to read here)
GPIO 0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15
PWM Channel 0A 0B 1A 1B 2A 2B 3A 3B 4A 4B 5A 5B 6A 6B 7A 7B

GPIO 16 17 18 19 20 21 22 23 24 25 26 27 28 29
PWM Channel 0A 0B 1A 1B 2A 2B 3A 3B 4A 4B 5A 5B 6A 6B 
It means that e.g. channel 0A can output the pwm signal on pin 0 and/or pin 16, Channel 0B on pin 2 and 17...
Still 7A and 7B can only be output on pin 14 and 15.

If you compile you own version of oXs and make some changes in the code, it would be possible to use other pins that 0/13 for some PWM.
Still I do not plan to change oXs for this request.
If you want to make you own version of oXs, I can explain where to change the code.

Otherwise, why don't you use a RP2040-Zero module. It is small and cheap and provide access to gpio 1/15

JVideocassetta, 2023-07-19:
I appreciate the info once again, I'm able to move a pin and get access to four channels which should be enough for my application. If not , I'll be sure to look into a larger RP2040 board.

Thanks for all of the assistance!

romoloman, 2023-07-23:
Have a look also to RP2040-TINY board, same size as RP2040-zero but:
1) one side of the PCB is completely flat, (Useful tu stack another pcb over it).
2) doesn't carry BUTTONS on board, so no risk to get them pressed by mistake.
3) doesn't carry USB-C connector on board
It still has the limit of max 5.5V as power supply (max is 6)

mstrens, 2023-07-24:
Indeed, this RP2040-Tiny could be a good alternative.

closed 2023-11-21 by mstrens