mstrens/oXs_on_RP2040 #94: seam to find a bug

secretman28, 2023-08-15:
on the 16 ch expander if elrs/crsf receiver is feeding seam to not get last 2 channel 15-16, if i switch elrs receiver to output sbus 16 work 
can it be in the oxs code somewhere?

mstrens, 2023-08-16:
I checked the oXs code and I do not see any reason why PWM on channels 15-16 should not work in ELRS protocol.
Generating the PWM signals is the same for both protocols.
The difference between the 2 protocols (ELRS/SBUS) in just in the way the data are received.

If you want I can make a debug version that would display on the PC the raw values received by oXs from ELRS for channels 15 and 16.  With this version you could check if the values change when the handset change those channels.

secretman28, 2023-08-16:
yes im interested i try to make rc plane people to get on elrs i need this to work im not a coder a can follow the Code a bit but... loll

mstrens, 2023-08-16:
Here a version 2.7.9
It is supposed to print a line on the PC (via USB serial terminal) each time RC data are received on ELR protocol on the prim pin.
Each line should contains 22 bytes (= a RC channel frame)
It is very difficult to understand the meaning of the bytes because the RC channels are packed (coded on 11 bits and so each channel takes more than one byte and less than 2 bytes).
Anyway, when Rc channel 16 changes, at least one of the 2 least bytes (byte 21 or/and 22) should change.
When RC channel 15 changes byte 20 and/or 21 should change.

[oXs.zip](https://github.com/mstrens/oXs_on_RP2040/files/12360490/oXs.zip)
,

jmxp69, 2023-08-16:
Here is a video of the issue: https://www.youtube.com/watch?v=CAHO2vdnPrA

jmxp69, 2023-08-16:
[image]
Debug output. All 16 channels are assigned to same stick. FYI: It works correctly with SBUS in from the RX.

jmxp69, 2023-08-16:
Receiver is Happymodel EPx with ELRS 3.3.0. TX is EdgeTX 2.8.4 with RM Ranger Micro on ELRS 3.3.0. Channel monitor confirms 16 channel output and simply switching RX output to SBUS all 16 channels show movement in SBUS output  and on PWM.

jmxp69, 2023-08-16:
[image]
Here is the configuration with CRSF

secretman28, 2023-08-16:
ya last 2 stay the same mmmm

mstrens, 2023-08-16:
Thanks for the feedback.
The messages given by oXs on the PC show indeed that the value of channel 16 (and I presume also channel 15) doe snot change.
I suspect that the bug is in ELRS firmware.
To be sure, best would be to look directly at the data provided by the ELRS receiver on his Tx pin.
This can be done at least in several ways:
- with a logic analyser (salea or clone)
- with a usb/serial module (like a FTDI).
If you do not have such a device, but have some arduino board (or rp2040) I could write a small program just to display the raw data generated by ELRS receiver.
Let me know what you can do.

secretman28, 2023-08-16:
ok tanks

jmxp69, 2023-08-16:
Issue opened with ELRS https://github.com/ExpressLRS/ExpressLRS/issues/2363

secretman28, 2023-08-17:
ok seam its not oXs its from elrs mmm need to find a way to fix that
haha

secretman28, 2023-08-18:
ok for the people, crsf serial use ch 15 and 16 for lq and rssi dbm, i find where to do the code change but still need to test sbus work #1 oXs work good tanks i will repost if something new tanks for your time !!

closed 2023-08-18 by secretman28

secretman28, 2023-08-20:
can you come chat with me ? i got some question to ask you
 discord secretman3986