mstrens/oXs_on_RP2040 #95: Enhancement Protocol Converter Mode

m0p3d, 2023-08-22:
It would be very beneficial for elrs adaption to have the possibility to get already owned hardware working with CRSF.

Would be nice to connect e.g. Graupner or FrSky Telemetry to oXS and get CRSF equivalent Telemetry as output.

mstrens, 2023-08-22:
So, what you ask is to have a module that get telemetry data from Graupner.Frsky or other RC companies and that converts the telemetry data into the ELRS format.
In principe this is possible but I do not see this as being part of oXs project.
Note: the ELRS telemetry is quite limited (about the type and number of fields). It is not so open as e.g. Sport (Frsky).
The primary goals of oXs is to build his own sensors at a much lower price than those from original rc companies.

m0p3d, 2023-08-22:
Thank you for investigating and your awesome project. Confirm u understood right my request.