mstrens/oXs_on_RP2040 #98: GPS telemetry variable name

f2knpw, 2023-09-15:
First of all I would like to thank you for this incredible firmware. It works so well !!!

I have just a very little suggestion. When connecting a Ublox M8N GPS everything is ok excepted that the lon/ lat variables (when discovered by EdgeTx on my Tx16s) do have exactly the same name "GPS".
This is not critical but I had to change their names to be able to use them into a lua script !
And whenever you "re discover" telemetry, their names are reset to "GPS" !

By the way, I have published a tutorial on hackaday to use this lib on my TX16s. Here is its link although for sure useless for you ! https://hackaday.io/project/192752-variometer-gps-tracker-remoteid-for-rc-glider

JP

mstrens, 2023-09-18:
Which protocol (Sport, ELRS, ...) do you use?

f2knpw, 2023-09-18:
Hi,

I am using FlSky2A on my FS-IA6B receiver.
My TX16s is equiped with internal 4in1 module.
[image]

[image]

f2knpw, 2023-09-18:
So with your firmware the protocol is "Flysky ibus"
[image]

mstrens, 2023-09-18:
In flysky (Ibus) protocol, oXs does not send the name of the field but a code.
The codes are defined like this:
IBUS_SENSOR_TYPE_GPS_LAT          = 0x80, //4bytes signed WGS84 in degrees * 1E7
IBUS_SENSOR_TYPE_GPS_LON          = 0x81, //4bytes signed WGS84 in degrees * 1E7
 
So oXs uses 2 different codes for long and lat.
I expect that it is EdgeTX that converts the code into a free text.
I suggest that you ask on EdgeTx forum because I expect the issue is at this level.

f2knpw, 2023-09-18:
issue tranfered to EdgeTx here : https://github.com/EdgeTX/edgetx/issues/4062

frankiearzu, 2023-09-21:
Just FYI, made the change for flySky_Ibus protocol.  Firmware can be downloaded at the end of the EdgeTX ticket: https://github.com/EdgeTX/edgetx/issues/4062

@mstrens : In your comment previously "IBUS_SENSOR_TYPE_GPS_LAT = 0x80, //4bytes signed WGS84 in degrees * 1E7".. does this means that we have to divide the LAT by 0x1E7 to get the correct value??  or the value is already a valid decimal notation??

mstrens, 2023-09-21:
The data transmitted over radio frequency are just integer (no decimals). It is the TX that convert it in float with decimals. The number of decimals is defined in a table on Tx side (and depends on the type of data).

mstrens, 2023-09-21:
Sorry, it is a typo error. The comment should say 1E7 (and not 0X1E7).
1E7 is the scientist representation of 10000000.
I will correct the comment in next version

frankiearzu, 2023-09-21:
My consusion!!!   when i see a number with A..E automatically my mind thing of Hex!

frankiearzu, 2023-09-21:
@mstrens 
Another Question.  The package info for ibus has SensorID and InstanceID. The instance ID seems to be different for every sensor in @f2knpw  image example.
Not an exper in Ibus, but in Spektrum, the instance is the ID of the device (usually 0).   So if you have only 1 GPS device, all the sensors associated with that device will have the same InstanceID  (different sensor ID or course).

This is for example if you have 2 devices who generate the set of sensors. like a double motor ESC you have 2 RPMs, etc. and the instance is how you differenciate wich one comes from what device  (first ESC is InstanceID=0, second ESC is InstanceID=1).

[image]