mstrens/oXs_on_RP2040 #99: Jeti EX Bus with PWM

Klippeneck, 2023-09-18:
Hello mstrens,
I am currently testing with the pierrotm 777 board.
RC is Jeti, receiver REX 10, E2 configured on EX bus.
RP 2040 SW is 2.70. Telemetry with MS 5611 and GPS works fine.
I wanted to use the PWM outputs for channels 8-12.
Configuration seems to be correct, but the servos have only approx. 20%
of the usual servo travel. Measured the servo signal varies only from 1,0 to
1.2 ms. According to the transmitter display it is 1-2ms. Can this be configured somewhere?
Jürgen

mstrens, 2023-09-18:
I found a bug.
I put a version 2.8.1 on github in test branch.
I did not tested this version.
Can you test if the bug is solved.
If not I should add some debug messages.

Klippeneck, 2023-09-18:
Hello mstrens,
I have just tested the test software 2.8.1,
Servos no longer move at all, constant 0.95ms at the outputs.
Jürgen

mstrens, 2023-09-18:
I already found a bug. I made version 2.8.2
Still not sure, this will solve the issue.
You can try it.
I will try to further debug this to morrow.

Klippeneck, 2023-09-19:
PWM output with Jeti EX bus works for all 16 channels.
Jürgen

closed 2023-09-19 by Klippeneck