mstrens/oXs_on_RP2040 #101: Support Hobbywing ESC Telemetry

m0p3d, 2023-09-22:
Any Chance to support readout of Hobbywing ESC Data/Telemetry
(Current, Consumption, Voltage, RPM)

mstrens, 2023-09-22:
I am currently testing a sd card logger to be associate to oXs.
When it is done, I can look to implement reading Hobbywing ESC.
Still, I do not have such a device. So I can't test the code.

I also saw that there are several versions of Hobbywing protocols.
Is it good enough to implement the latest (V5 I think)?

m0p3d, 2023-09-22:
V4 120A , V4 80A here ready to test

https://github.com/dgatf/msrc

Think its easy and most copy&paste

mstrens, 2023-09-25:
I prepared a draft version in test branch (2.8.5).
It is only part of the needed code but it should already allow to see if oXs can get some data from ESC Hobbywing V4.
In the config parameters (to be filled with USB/SERIAL command) there is now a parameter ESC to fill with the gpio being used to get data from ESC. Take care that the signal may not exceed 3.3V. I do not know if Hobbywing signal is 3.3 or 5V.
In this first version:
- I read only the voltage and current (no rpm, no consumption)
- I use fix parameters (#define) to scale voltage and current (later I should add new parameters in config or reuse existing scale0, scale1). I should probably also take care of an offset at least for current.
- I do not transmit the voltage and current to the receiver. I only print the values on the PC via USB/SERIAL
- later on, I would add check that V1, V2, TEMP1, TEMP2 and RPM pin are not defined when ESC pin is defined and I would transmit the ESC values.

Could you already test this version and say if you see voltage and current values on the PC display.
Probably that the values will not be 100% correct because scales are not the valid one for your ESC.

m0p3d, 2023-09-26:
Please instruct me how to wire - would like to use GPIO 5 for the Telemetry signal. Where to connect BEC (because no Mass=floating Signal)- be aware BEC is at 7.4V (for the msrc i used stepdown module 7.4V to 5V to power the RP2040)
Never used power via usb while ESC/BEC on

mstrens, 2023-09-26:
as said, current version does not yet provide telemetry data to the receiver.
It is just a test version to see if oXs can read the frames provided by the ESC.
For this test the RP2040 must be connected to the PC via the USB in order to let the PC display the messages with voltage and current.
As the RP2040 may not be powered by 2 sources, you have (for this test) to:
- connect RP2040 to the PC via USB
- connect RP2040 to ESC with only 2 wires :  ground ESC <=> ground RP2040 and ESC telemetry signal <=> one pin of RP2040 
The ESC has also to be connected to the power supply, the RX and the motor.
.You are free to select your self the RP2040 pin connected to the ESC telemetry signal.
To let oXs knows this pin( and so let know that a ESC is connected), you have to enter on the PC a command via a serial terminal (just like for other gpio's used for sensors). The command here is ESC=x where x = the gpio number. 
If you just press ENTER in the serial monitor, oXs should display the current config (so you can see if the ESC pin is well registered.
If you press ?+ENTER in the serial monitor, oXs should display all the commands that can be introduced (=help).
When the ESP pin is defined, oXs is supposed to print on the PC one line with voltage an current every 20 msec.

m0p3d, 2023-09-26:
22,87V Multimeter =

Config parameters are OK
Press ? + Enter to get help about the commands
Esc Volt=43718   current=0
Esc Volt=43718   current=0
Esc Volt=43718   current=0
Esc Volt=43735   current=0
Esc Volt=43718   current=0
Esc Volt=43718   current=0
Esc Volt=43718   current=0
Esc Volt=43718   current=0
Esc Volt=43718   current=0
Esc Volt=43718   current=0
Esc Volt=43718   current=0
Esc Volt=43718   current=0
Esc Volt=43718   current=0
Esc Volt=43718   current=0
Esc Volt=43718   current=0
Esc Volt=43718   current=0
Esc Volt=43735   current=0
Esc Volt=43718   current=0
Esc Volt=43718   current=0
Esc Volt=43718   current=0
Esc Volt=43718   current=0
Esc Volt=43718   current=0
Esc Volt=43853   current=0
Esc Volt=43718   current=0
Esc Volt=43718   current=0
Esc Volt=43718   current=0
Esc Volt=43718   current=0
Esc Volt=43718   current=0

mstrens, 2023-09-26:
Thanks.
That is not so bad for a first test.
So, it seems that oXs can read the data from ESC.
The value for Volt is supposed to be in millivolt.
The value for current remains 0 as long as Throttle is less that 1/4 of the max. So it is probably not abnormal for this test.

For Voltage, to get the right value, I have to apply a scale factor that depends on ESC model (depends on the number of cells supported by the ESC when I look at msrc doc. Currently I have used a "V-DIV" = 21 that is supposed to be used for ESC 5_12 cells.
I will change the code, in such a way that I will not apply anymore a fixed V-DIV and I will:
- let oXs apply (multiply) by the value that the user can fill in SCALE1. So I do not have to add new parameters.
- in the same way, I will use SCALE2 and OFFSET2 parameter to let the user adjust the current.
- I also imagine to reuse the existing RPM multiplicator for RPM.
 
I imagine to transmit Volt, current, consumption, RPM, temp FET and temp BEC in the current telemetry fields available.
In order to avoid conflict I will add some code to check that pin V1, V2, RPM and number of TMP are not defined when ESC pin is defined.

I could also look to support hobbywingV3.
Question: does this really make sense. If all new ESC made by Hobbywing are V4 or V5, it is probably not useful anymore to support V3 and it make code more complex.

I plan to write those changes in the coming days.

mstrens, 2023-09-26:
I put a version 2.8.7 in test branch on github.
It is supposed to:
- provide telemetry to RX for Voltage, current, RPM, temp1 and temp2 (so not yet consumption).
- take care of SCALEVOLT1 for volt,
- take care of SCALEVOLT2 and OFFSET2 for current
- take care of RPM multiplicator for RPM.
Those parameter have to choosen to get realistic value. If I know the parameters that you used with msrc, I could make some calculations to propose the oXs values to be used.

m0p3d, 2023-09-27:
Readouts- Serial Debug
V4 60A
22,8V 43583
22,1V 42263
15,4V 29422

V4 120A
22,8 43633
22,1 42280

mstrens, 2023-09-27:
I presume that the values you just reported are still with the previous version.
Best would be to try the latest version (that does not provide values on the PC via usb but on the handset).
It will then be easy to find the SCALE1 to apply based on your measurements.
E.g. if for XX.X Volt you get a value of YY.YY volt when SCALE1 = 1.0, then SCALE1 should be changed to XX.X / YY.YY

If you can't make measurements (to compare real value with reported value) , then I could make some calculations based on the msrc parameters that I see in his code but this will perhaps be less accurate.

m0p3d, 2023-09-27:
Im not able to get SPort (FrSky) to work/discover any sensors. TLM = 0 and Connect Sport from RX to RP2040 Pin 0 is all i need to do ?

mstrens, 2023-09-27:
Set TLM=X where X is the GPIO that you will connect to Sport pin of the RX; X can e.g. be 0
Set PROTOCOL=S
I suggest that you also set V3=26 (in order to ask oXs to measure and transmit volt3 on pin 26 even if the value has no sense). It is just a way to get at least one telemetry data. So if there is a bug in my code for Hobbywing v4, you should at least know that the rest of the program is OK.

mstrens, 2023-09-28:
FYI, I put version 2.8.8 on github.
I made some changes for Hobbywing V4 (e.g. I reactivate printing the ESC data via usb for debugging).

I added support for Kontronik ESC.
So you have now to:
- enter a command to specify the type of ESC; the command is ESC_TYPE=HW4 (or =KON)
- the command to specify the ESC gpio is now ESC_PIN=xx (instead of ESC=xx)

m0p3d, 2023-09-29:
Confirmed works for -RPM, TEMP1,2  ,Current
Strange VFAS=2,58V ADC2=0V

mstrens, 2023-09-29:
With version 2.8.8, if you get VFAS=2,58V when it should be e.g. 22.80V (and you had SCALE1=1.0), then you have to enter a command to change SCALE1 to 22.80/2.58= 8.84. So command  should be SCALE1=8.84

closed 2023-11-21 by mstrens