mstrens/oXs_on_RP2040 #103: Version 2.8.3 compiles without errors, but code does not run.

KeesBlokland, 2023-09-24:
Hi, I'm hitting a snag: Updated my test system to 2.8.0, compile it on my own system, upload the code and all is happy and working as expected. Next I wanted to test 2.8.3, in preparation for connecting to the logger. Logger code compiles, and appears to work (on my separate rp2040+sdcard)

However, downloading and installing the 2.8.3 uf2 from here, does not work. After uploading I see the following behavior on my waveshare board: brief blue led flash, 6 times red led blink, ca 4 secs pause, repeats over and over. I can briefly select /dev/ttyACM0 port:  (normally I have an inverted LED)
The serial monitor says: 
---- Opened the serial port /dev/ttyACM0 ----
---- Closed serial port /dev/ttyACM0 due to disconnection from the machine ----

My feeling is that it is stuck in a boot loop, but as yet have no idea how to trouble shoot this. 

So, both my own compiled version and the included .uf2 behave exactly the same. 
When I compile the code on my system I get no errors, no warnings, and the resultant code size matches the one here. 

For the rest, I am very impressed with the system! 
Thanks,
Kees

mstrens, 2023-09-24:
Thanks for the feed back.
I will have to check.
I think a 2.8.3 version worked on my side but I made some minor changes without upgrading the version number.
So I would have to check again.
Can you also provide me your configuration.

Note:  I implemented a watchdog on oXs. So if it stay to long in a loop, it performs an automatic reset.
I will have to search where it blocks.

KeesBlokland, 2023-09-25:
Good morning, this is my config for this board: 

Version = 2.8.0 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    9  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   10  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   11  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255  255  255 (V1 / V4 = 26, 27, 28, 29)

Protocol is Jeti (Exbus)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using BMP280
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is not detected
Airspeed sensor is not detected
    No Vspeed compensation channel defined; oXs uses default settings
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox (configured by oXs) :GPS is detected but has not (yet) a fix
Led color is inverted
Failsafe type is HOLD

No sequencers are defined

Config parameters are OK
Press ? + Enter to get help about the commands

For the rest: VScode on Ubuntu, no errors, no warnings. 

PICO_SDK_PATH is /opt/pico-sdk
PICO platform is rp2040.
Build type is RelWithDebInfo
PICO target board is pico.
Using board configuration from /opt/pico-sdk/src/boards/include/boards/pico.h
TinyUSB available at /opt/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /opt/pico-sdk/lib/btstack
cyw43-driver available at /opt/pico-sdk/lib/cyw43-driver
Pico W Bluetooth build support available.
lwIP available at /opt/pico-sdk/lib/lwip
mbedtls available at /opt/pico-sdk/lib/mbedtls
-- Configuring done
-- Generating done
-- Build files have been written to: /home/x/Documents/PlatformIO/Projects/oXs-Orig/2.8.3/oXs_on_RP2040-test/build
PICO_SDK_PATH is /opt/pico-sdk
PICO platform is rp2040.
Build type is Release
PICO target board is pico.
Using board configuration from /opt/pico-sdk/src/boards/include/boards/pico.h
TinyUSB available at /opt/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /opt/pico-sdk/lib/btstack
cyw43-driver available at /opt/pico-sdk/lib/cyw43-driver
Pico W Bluetooth build support available.
lwIP available at /opt/pico-sdk/lib/lwip
mbedtls available at /opt/pico-sdk/lib/mbedtls
-- Configuring done
-- Generating done
-- Build files have been written to: /home/x/Documents/PlatformIO/Projects/oXs-Orig/2.8.3/oXs_on_RP2040-test/build

Is this what you meant?

regards, Kees

mstrens, 2023-09-25:
I put version 2.8.4 on github.
On my side, it does not block (I am not connected to a Rx and so red led is continuously blinking).
I made a small change : I check now if logger baudrate parameter is within valid range. If not, config is now not valid.
In previous version, it could be that when you define a logger pin, oXs restarted immediately with a wrong baudrate and block.
Now, you should at least have the opportunity to enter a valid baudrate after having enter a logger pin.

Let me know if it is now OK

KeesBlokland, 2023-09-25:
Thanks, yes, that works!  I Won't be able to get the logger connected till later, but so far so good!

Version = 2.8.4 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    9  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   10  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   11  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255  255  255 (V1 / V4 = 26, 27, 28, 29)
Logger  . . . . . . . . . =    0  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =    1  (ESC    = 0, 1, 2, ..., 29)

Protocol is Jeti (Exbus)
CRSF baudrate   = 420000
Logger baudrate = 115200

Thanks!

KeesBlokland, 2023-09-25:
Hi, you say: 

_On my side, it does not block (I am not connected to a Rx and so red led is continuously blinking).
I made a small change : I check now if logger baudrate parameter is within valid range. If not, config is now not valid.
In previous version, it could be that when you define a logger pin, oXs restarted immediately with a wrong baudrate and block.
Now, you should at least have the opportunity to enter a valid baudrate after having enter a logger pin._

I did not have anything connected either, but since I never got into the config setup, I could not assign anything, no logger pin, no baudrate. Anyway, we're good now!

mstrens, 2023-09-25:
I had expected that you had configured the logger gpio with the LOG command.

If not, I do no understand. I 
Perhaps the version was in the middle of some changes

KeesBlokland, 2023-09-25:
> I had expected that you had configured the logger gpio with the LOG command.
> If not, I do no understand. I Perhaps the version was in the middle of some changes

No, I never even let me talk to it, when pressing Start monitoring, VSCode opened the port ACM0, and immediately closed it, ( and said the system , ie rp2040 disconnected the terminal)

Same with Putty, Arduino terminal, and some other terminal sware. 

It’s one of those magic things, it works, better not dig too deep! Could have been a cosmic ray that messed up the chip, or any of a zillion other mysterious reasons.

closed 2023-09-26 by KeesBlokland