mstrens/oXs_on_RP2040 #104: Feature Request: Add ACCx/y/z data fields to the standard SPort/FPort protocol

rieden1, 2023-10-01:
As title - Could you possibly add the MPU acceleration fields (ACCx/y/z) to the SPort/FPort protocol?

Satcomix, 2023-10-03:
Hello Mstrens,
I did a few tests on the X20S with FBUS Protocol and the latest version 2.8.12.
All telemetry values ​​are shown on the display and at FV.
Doesn't the compensation factor have to change in FV if I change the PWM value of channel 7 from 0% to +100% or -100%?
Greetings,
Torsten

processing cmd

**Version = 2.8.12** 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    4  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255   10   11
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)
Logger  . . . . . . . . . =  255  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =  255  (ESC_PIN= 0, 1, 2, ..., 29)

Esc type is not defined

Protocol is Fbus(Frsky)
CRSF baudrate   = 420000
Logger baudrate = 0
Voltage parameters:
    Scales : 2.000000 , 1.000000 , 0.100000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 2.000000 , 0.000000 
    One temperature sensor is connected on V3
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 790 , -436 , -2916
     Gyro offsets         X, Y, Z = 14 , -162 , -193
Aispeed sensor is detected using MS4525
    **Vspeed compensation channel = 7**
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Foreseen GPS type is Ublox (configured by oXs) :GPS is detected and has a fix
Led color is normal (not inverted)
Failsafe uses predefined values
     Chan 1...4  =  1500  1500  1500  1500
     Chan 5...8  =  1500  1500  1500  1500
     Chan 9...12 =  1500  1500  1500  1500
     Chan 13...16=  1500  1500  1500  1500

No sequencers are defined

Config parameters are OK
Press ? + Enter to get help about the commands
processing cmd

Cmd to execute:   FV

GPS Latitude = 51.1225696 degree
GPS Longitude = 6.7173104 degree
GPS Groundspeed = 7 cm/s
GPS Heading = 131.970000 degree
GPS Altitude = 4797 cm
GPS Num sat. = 117
GPS Date J M A = 3 10 23 
GPS Time H M S = 7 53 15 
GPS Pdop = 152 
GPS Home bearing = 161 degree
GPS Home distance = 6 m
Volt 1 = 3708 mVolt
Current (Volt 2) = 1653 mA
Volt 4 = 341 mVolt
Capacity (using current) = 709 mAh
Temp 1 (Volt 3) = 23 degree
Vspeed = -3 cm/s
Baro Rel altitude = 37 cm
Pitch = -0.410000 degree
Roll = 1.720000 degree
RPM = 999 Hertz
Ads 1 1 = 119 mVolt
Ads 1 2 = 119 mVolt
Ads 1 3 = 119 mVolt
Ads 1 4 = 119 mVolt
Ads 2 1 = 118 mVolt
Ads 2 2 = 282 mVolt
Ads 2 3 = 836 mVolt
Ads 2 4 = 802 mVolt
Airspeed = -50 cm/s
Compensated Vspeed = -47 cm/s
Gps cumulative distance = 115
Acc X = 0.007000g
Acc Y = 0.029000g
Acc Z = 1.001000g
**Vspeed compensation = 1.10**

mstrens, 2023-10-03:
In the code, I said:
    // when channel is more than positive(in %), channel value is used to calculate the factor and compensated Vspeed is send
    // when channel is around at center, we use default value to calculate the factor and compensated Vspeed is send
    // when channel is more than negative (in %), compensated Vspeed is calculated with previous factor but not sent (we send normal Vspeed) 

Still I noticed that the program does not do exactly the above.
It used some fixed parameters:
#define DTE_MIN_CHANNEL_COMP_VALUE 0X0500
#define DTE_MAX_CHANNEL_COMP_VALUE 0X0700
#define DTE_NO_CHANNEL_COMP_VALUE  0X0200

A value 0X0400 is about mid range (it is a rc channel value as transmitted in Sbus so coded on 11 bits.)
So compensation factor should change with channel value when the rc channel is between min (0X0500) and max  (0X0700).
I noticed a bug in the way the factor is calculated: it can vary only between 0.9 and 1.0 (It should better be 0.9 and 1.3)

When the RC channel is between 0X0200 and 0X0500, it is considered as around center and so compensation should be the value defined in config.h (1.1 currently)

When the Rc channel is less than 0X0200, it is considered as negative and compensation factor keep previous value but oXs is supposed to transmit uncompensated Vspeed.
I noticed that 0X0200 is perhaps quite low and I could probably best replace it with 0X0300 (to be closier to mid range).

mstrens, 2023-10-03:
I made version 2.8.13 to let compensation factor varies from 0.9 up to 1.3 and I set DTE_NO_CHANNEL_COMP_VALUE = 0X0300 (to reduce the "mid range").

Satcomix, 2023-10-03:
Hello Mstrens,
I have ver. 2.8.13 tested with FBUS.
Channel 7 is a potentiometer.
If I adjust from the middle 0% to +100%, I see **no change** in the FV. **1.10 remains unchanged!**
When I adjust from 0% to -100% I see a change in comp. factor from 1.10 up to 8.90.
greetings,
Torsten

mstrens, 2023-10-03:
It is very strange.
I do not understand.
I added a debug message in 2.8.14
It should print (on USB) the value used for Rc channel compensation.
Note : it will print about 50 val per sec.
Value should change when you change the pot.
Value is in Hexa and is expected to be the Sbus value (= 11 bits) so between 0X000 and 0X7FF
Perhaps the bug is in the conversion from 11 bits to 16 bits
It should be useful to know the values in the debug messages when the pot varies

Satcomix, 2023-10-03:
Hello Mstrens,
For info:
I've made a mistake.
I swapped the values ​​-100% and +100%.
But nothing changes in the result.
+100% = Comp. 8.90
Now i will make some tests with version 2.1.14

Satcomix, 2023-10-03:
Hello Mstrens,
I can find only Ver.2.8.13
[Test-Comp-Factor.txt](https://github.com/mstrens/oXs_on_RP2040/files/12791605/Test-Comp-Factor.txt)

Greetings,
Torsten

mstrens, 2023-10-03:
OK.
I think I made a mistake changing a value 0.5 to 8.0. 
This explains why you got such high value (8.9).
In version 2.8.15 I put back 0.5
So now the value should varies from 0.9 up to 1.4

When you use a pot and set the channel to a huge negative value, you first goes to the mid range and so the value remains on 1.1.
If you use a switch to go directly from e.g. +75 to -75, the compensation factor will remain on his previous value (about 1.2 perhaps).
The idea was to use a pot to test the right value, then to fix it the TX and to use switch to get or the compensated Vpseed or the uncompensated Vspeed in the same telemetry field.

Satcomix, 2023-10-03:
Hello Mstrens,
With Version 2.8.15 erverything seems to be normal again!
1690us = 0.92
1700us = 0.93
1750us = 1.01
1800us = 1.09
1820us = 1.12
1850us = 1.17
1880us = 1.21
2000us = 1.40

Many thanks for the work and time.
best regards,
Torsten

closed 2023-11-21 by mstrens