mstrens/oXs_on_RP2040 #105: Unusual high values for Pitch and Roll

Satcomix, 2023-10-02:
Hello Mstrens,
I'll be back after a health break.
I'm currently doing a few tests with the version. 2.8.0-Main and my X20S with FBUS protocol.
I noticed unusually high values ​​for the pitch and roll values. I've already replaced the MP6050, but haven't noticed any improvement.
You can see it clearly on the Field View. Is there any news about Pitch and Roll that I don't know?
Best regards,
Torsten
Version = 2.8.0-Main
Cmd to execute:   FV

GPS Latitude = xx.xxxxxxxxxx degree
GPS Longitude = x.xxxxxxxxx degree
GPS Groundspeed = 6 cm/s
GPS Heading = 316.610000 degree
GPS Altitude = 4107 cm
GPS Num sat. = 117
GPS Date J M A = 2 10 23 
GPS Time H M S = 13 10 49 
GPS Pdop = 121 
GPS Home bearing = 130 degree
GPS Home distance = 2 m
Volt 1 = 3707 mVolt
Current (Volt 2) = 1653 mA
Volt 4 = 307 mVolt
Capacity (using current) = 89 mAh
Temp 1 (Volt 3) = 25 degree
Vspeed = -48 cm/s
Baro Rel altitude = 41 cm
Pitch = 3338 degree
Roll = 9991 degree
RPM = 1007 Hertz
Ads 1 1 = 119 mVolt
Ads 1 2 = 119 mVolt
Ads 1 3 = 119 mVolt
Ads 1 4 = 119 mVolt
Ads 2 1 = 119 mVolt
Ads 2 2 = 119 mVolt
Ads 2 3 = 119 mVolt
Ads 2 4 = 119 mVolt
Airspeed = 30 cm/s
Compensated Vspeed = -2 cm/s
Gps cumulative distance = 17
Vspeed compensation = 1.10

mstrens, 2023-10-02:
The issue is perhaps in the FV command.
The internal values are in 0.01 degree but in FV I forgot to divide the values by 100 to get degree.
if you divide the values by 100 do you get valid values?
In next version I will fix the FV format issue.

Satcomix, 2023-10-02:
Hello Mstrens,
My Display on my X20S with FBUS protocol show the same values from 0° to 10000° for Pitch and Roll
Greetings,
Torsten

mstrens, 2023-10-02:
can you try latest version (in test branch).
I fixed a bug some months ago about roll and pitch (values had to be divided by 100).
In 2.8.11, I just fixed the printout for FV (not tested).

Note : 2.8.11 contains several new features (partly tested):
- sequencer
- logger
- Acc X,Y,Z fields added

Satcomix, 2023-10-02:
Thanks Mstrens,
In the X20S Radio now everything is ok with version 2.8.0, with the new Firmware for Radio, there were more features. I must set it from -90° to +90° and a resulution from 0.00°.
I will test the 2.8.11 version with the new features.
Greetings,
Torsten

Satcomix, 2023-10-02:
Hello Mstrens,
with Ver.:2.8.11_test everything for pitch and roll in FV is ok.
I will close this issue and make some more test with log, esc, and sequences.
Greetings,
Torsten

closed 2023-10-02 by Satcomix