mstrens/oXs_on_RP2040 #106: Enhancement/Feature Request: MPU6050 calibration from Tx command

rieden1, 2023-10-03:
At present the MPU6050 can be calibrated using the USB command "MPUCAL". It would be extremely useful to be able to calibrate (zero the datums) at the start of a flying session with the unit installed in the aeroplane. Would it be possible to have calibration initiated from a Tx command - either from a PWB state, SBus data or something else?

Satcomix, 2023-10-03:
Hello Peter,
When you use a GY-521 board with MP6050, the MP6050 wil normally calibbrate itself after every PowerOn.
It is already calibrated when delivered.
When you use a GY-86 you must use MPUCAL.
Best regards,
Torsten

rieden1, 2023-10-03:
Thanks for confirming - I rather guessed that was happening but wasn't sure. However I think it would still be a useful feature because most of my fleet (100% electric these days) are powered up by connecting the main flight battery, and in many cases this is done with the model inverted  or on its side. It would be jolly nice if, once the battery was fitted and the model ready for flight, I could put it in a level datum position and do a "set zero now" command from the Tx

mstrens, 2023-10-03:
I expect that calibration should only be done one.
When done, the results are stored in flash and retrieved/reused after each power on.

I do not like the idea of using a Rc channel to force it because:
- it requires a rc channel. In many cases, oXs is used just for telemetry (e.g. with sport) and does not collect the Rc channels so it can't be used.
- each time one parameter is modified it has to be stored in flash. This process is slow and blocks the main loop. So some functionalities does not work during this process.  Further more, I perform a total reset once the saving is done. This reset can block some sensor (because this software reset does not reset the I2C at sensors side). So there are some risks to perform this saving when the device is in a model and used to generate e.g. PWM signals

closed 2023-10-03 by rieden1