mstrens/oXs_on_RP2040 #109: variable pwm no more servo output

secretman28, 2023-10-05:
i flash the test for acessing the pwm . i nuke the rp2040 to be shure to start new. all seam to work serial connection act as usual , set all the parameter as usual and all is set when a look in the serial return

i got nothing on the servo output anymore i can set any pwmhz and nothing come out test with a ocillioscope show nothing

flash back main all work as usual

secretman28, 2023-10-06:
i test again tonight same i can get pwm out to work, i will try to look in to the code a bit but.. im not that good hahah

secretman28, 2023-10-06:
other test, the pwm command show that the pico receive the rx signal i move the throttle stick in this example

Cmd to execute:   PWM
PWM values us (sbus) 1... 8   1500(  992)  1500(  992)   988(  172)  1500(  992)   988(  172)   988(  172)  1500(  992)  1500(  992)
PWM values us (sbus) 9...16    988(  172)  2011( 1810)  1500(  992)  1493(  980)  1500(  992)  1500(  992)  2012( 1811)  2012( 1811)
processing cmd

Cmd to execute:   PWM
PWM values us (sbus) 1... 8   1500(  992)  1500(  992)  1039(  254)  1500(  992)   988(  172)   988(  172)  1500(  992)  1500(  992)
PWM values us (sbus) 9...16    988(  172)  2011( 1810)  1500(  992)  1494(  982)  1500(  992)  1500(  992)  2012( 1811)  2012( 1811)
processing cmd

Cmd to execute:   PWM
PWM values us (sbus) 1... 8   1500(  992)  1500(  992)  2011( 1810)  1500(  992)   988(  172)   988(  172)  1500(  992)  1500(  992)
PWM values us (sbus) 9...16    988(  172)  2011( 1810)  1500(  992)  1494(  982)  1500(  992)  1500(  992)  2012( 1811)  2012( 1811)
processing cmd

Cmd to execute:   PWM
PWM values us (sbus) 1... 8   1500(  992)  1502(  994)   988(  172)  1500(  992)   988(  172)   988(  172)  1500(  992)  1500(  992)
PWM values us (sbus) 9...16    988(  172)  2011( 1810)  1500(  992)  1494(  982)  1500(  992)  1500(  992)  2012( 1811)  2012( 1811)

here the setup
Version = 2.8.22 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =   25  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =   24  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =  255  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =  255  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =    0    1    2    3 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =    4    5    6    7
PWM Channels 9,10,11,12   =    8    9  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255  255  255 (V1 / V4 = 26, 27, 28, 29)
Logger  . . . . . . . . . =  255  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =  255  (ESC_PIN= 0, 1, 2, ..., 29)

Esc type is not defined

Protocol is CRSF (=ELRS)
CRSF baudrate   = 420000
Logger baudrate = 0
PWM is generated at = 100 Hz

Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is not detected
Acc/Gyro is not detected
Airspeed sensor is not detected
    No Vspeed compensation channel defined; oXs uses default settings
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox (configured by oXs) :GPS is not (yet) detected
Led color is normal (not inverted)
Failsafe type is HOLD

No sequencers are defined

Config parameters are OK

Satcomix, 2023-10-06:
Hello secretman28,
I have just tested Ver.2.8.22 again with the command PWMHZ=50,100,333 on channels 1-10, with my X20S with TD-R10 and FBUS Protocol.
I get a value between 988us to 2012us on all PWM channels and the connected servos work perfectly.
Greetings,
Torsten

secretman28, 2023-10-06:
mmm im testing elrs can be the problem i need to chk if sbus work tanks for the help i will report back still, main work as usual all same parameter except the pwmhz

mstrens, 2023-10-09:
I think I found the bug for PWM with ELRS.
Can you now try version 2.8.26 (in test branch)

secretman28, 2023-10-09:
yes i will thus morning

secretman28, 2023-10-09:
cant get nothing even sbus to get out i dont have a tonne of time today but will still test more can be something i do but if i flash back main servo pwm work still need to mess more to tell

mstrens, 2023-10-10:
I checked the code once more (comparing main and test) and I do not see the bug.
When you have time to test, I can make a test version that should print more debug messages to see where the part of the code that are used and not used.

secretman28, 2023-10-10:
i willl test in an hour need to finish something before if you want reach me  faster discord me,  if not put it here i will be aware and check tonight

secretman28, 2023-10-11:
just download the test code and now all is good! on sbus on elrs etc etc can set pwmhz too ocilloscope show the hz change ans i wana say rp2040 is pretty good at getting stable signal! 

this version is working for pwmhz tanks for taking the time to implement it

closed 2023-10-11 by secretman28