mstrens/oXs_on_RP2040 #111: Logger for oXs_RP2040

Satcomix, 2023-10-12:
Hello Mstrens,
Can I open the issue here or on Logger GitHub?
In the meantime I tried to play the .uf2 from the logger project onto a board.
If I press ? and Enter, I don't see any help for entering commands.
Greeting,
Torsten

Rebooted by Watchdog!

Config parameters are OK
Press ? + Enter to get help about the commands
 
Type any character to begin +++++++++++

Satcomix, 2023-10-12:
**First test with Challenger RP2040 SDRTC**

Rebooted by Watchdog!

Config parameters are OK
Press ? + Enter to get help about the commands
 
Type any character to begin +++++++++++
error in sd.begin
 
max ms between 2 call to main loop= 0
max % used of Serial2 fifo= 0
max % of bytes in csvbuf= 0
max us to fill csv rec= 0
max % used of fromNbr queue= 0
max % used of lastWritten queue= 0
max us to write on sd card= 0
max us to synchro on sd card= 0
 total bytes written in csv file= 0
 
Commands can be entered to change the config parameters
- To select data to be part of log file, enter ADD = xx / yy
- To unselect data, enter DEL = xx / yy
      where xx and yy are the index of the fields provided by oXs
- To change the minimum interval between 2 log entries, enter INTV=XXXX
       XXXX is in msec and can be 0 (to get all log entries)
- To change the logging mode, enter MODE=C or MODE=T or MODE=F
      C (continuous = log always), T (Triggered = start when MIN & MAX match) , F (Filtered = log when MIN & MAX match)
- To change the MIN, enter MINF=XX (for the field index) and enter MINV=YYYYY (for the value)
- To change the MAX, enter MAXF=XX (for the field index) and enter MINV=YYYYY (for the value)
- To change (invert) led color, enter LED=N or LED=I
- To change the gpio that get the data from oXs, enter DATA= XX  (with XX = 5, 9, 21, 25)
- To change the baudrate, enter BAUD= XXXXXX  (with XXXXXX = the baudrate e.g. 15200)
- To change the SD MOSI gpio, enter MOSI= XX  (with XX = 3, 7, 19, 23)
- To change the SD MISO gpio, enter MISO= XX  (with XX = 0, 4, 16, 20)
- To change the SD SCLK gpio, enter SCLK= XX  (with XX = 2, 6, 18, 22)
- To change the SD CS   gpio, enter CS= XX    (with XX in range 0...29)
 
   Note: some changes require a manual reset to be applied
 
- To get the content of current csv file, enter PF (PF means Print File)
 
-To get the current config, just press Enter
 
max ms between 2 call to main loop= 0
max % used of Serial2 fifo= 0
max % of bytes in csvbuf= 0
max us to fill csv rec= 0
max % used of fromNbr queue= 0
max % used of lastWritten queue= 0
max us to write on sd card= 0
max us to synchro on sd card= 0
 total bytes written in csv file= 0

Satcomix, 2023-10-12:
For the ILabs Challenger RP2040 SD/RTC
You must use these GPio:
RP2040 SD Card connector
Connection between MCU and the SD card.
The processor connects to the SD card via the second spi (SPI1) communication channel.
SPI1 Pin usage:

    GPIO10 is the serial clock (SCK)
    GPIO12 is the serial data input (SDI)
    GPIO11 is the serial data output (SDO)
    GPIO9 is the SD Card chip select signal (#CS)

In addition to the described signals a card detect signal is also connected to the connector. When this signal is logically low a card has been inserted and when it is high the card has been removed.
    D14/GPIO13 card detect (Has a weak external pull (220K) up for low power consumption).

https://ilabs.se/challenger-rp2040-sd-rtc-datasheet/

mstrens, 2023-10-14:
I expect that you could not make any test because the logger did not allow to use SPI1.

I put on github a version 2.0.1 that should support SPI1.
To use SPI1 you have to change some parameters with usb command (set SPI to 1 and change gpio). Default is still SPI0

Satcomix, 2023-10-14:
Hello Mstrens,
I can't stop testing at all. :-)))
Thank you for the new version, I have a problem with my Arduino IDE and with VSCode. I cannot modify and create builds for the logger myself.
I will test your version 2.0.1 directly.
greetings,
Torsten

Satcomix, 2023-10-14:
Hello Mstrens,
As I wrote, I'm trying in vain to create a build with changes for the Challenger RP2040 SD/RTC.
MISO, MOSI, SCLK, and CS must be changed as specified (see post above) and the LED is on GPio pin 18.
Maybe it would be possible for you to create the build for me.
Thank you very much,
Torsten

oXs Logger Ver.0.0.1
- To change the SD MOSI gpio, enter MOSI= XX  (with XX = 3, 7, 19, 23)
- To change the SD MISO gpio, enter MISO= XX  (with XX = 0, 4, 16, 20)
- To change the SD SCLK gpio, enter SCLK= XX  (with XX = 2, 6, 18, 22)
- To change the SD CS   gpio, enter CS= XX    (with XX in range 0...29)

mstrens, 2023-10-14:
In principe, you should be able to use the uf2 provided on github.
As I had to write the program with arduino, the uf2 is in build directory.
You should (in theory) be able to change the parameter via usb command (so no need to recompile).

As this is a debug version, the program blocks at start on. You have to send a char to let it really begin.
Once it begins to really run, you can send the ENTER command to get the config and the ? to get the list commands.
I forgot to adapt some msg about the ? command. I think it is now done in 0.0.3

Anyway, perhaps I can help you about arduino/vscode compilation.

Satcomix, 2023-10-14:
First Test with Ver.0.0.3
SD=1
CS=9
MOSI=11
MISO=12
SCLK=10
DATA=5
for the Challenger RP2040 SD/RTC

Version = 0.0.3
Fields in CSV log (+ = included; - = not included)
   + 0 Gps
   + 1  
   + 2 Gps groundspeed (m/s)
   + 3 Gps Heading
   + 4 Gps Altitude(m)
   + 5 Gps Num Sat
   + 6 Gps Date
   + 7 Gps Time
   + 8 Gps PDOP
   + 9 Gps home bearing
   + 10 Gps home dist
   + 11 Volt 1 (V)
   + 12 Current(A)
   + 13 Volt 3(V)
   + 14 Volt 4(V)
   + 15 Capacity (Ah)
   + 16 Temp 1 
   + 17 Temp 2
   + 18 VSpeed (m/s)
   + 19 Baro Altitude (m)
   + 20 Pitch
   + 21 Roll
   + 22 Yaw
   + 23 RPM
   + 24 ADS_1_1
   + 25 ADS_1_2
   + 26 ADS_1_3
   - 27 ADS_1_4
   - 28 ADS_2_1
   - 29 ADS_2_2
   - 30 ADS_2_3
   + 31 ADS_2_4
   + 32 Airspeed (m/s)
   + 33 Comp VSpeed (m/s)
   + 34 Sbus Hold count
   + 35 Sbus Failsafe Count
   + 36 GPS cumul dist (m)
   + 37 Unused 37
   + 38 Unused 38
   + 39 Unused 39
   + 40 Unused 40
   + 41 Rc1
   + 42 Rc2
   + 43 Rc3
   + 44 Rc4
   + 45 Rc5
   + 46 Rc6
   - 47 Rc7
   - 48 Rc8
   - 49 Rc9
   - 50 Rc10
   - 51 Rc11
   - 52 Rc12
   - 53 Rc13
   - 54 Rc14
   - 55 Rc15
   - 56 Rc16
   - 57 Unused 57
   - 58 Unused 58
   - 59 Unused 59
   - 60 Unused 60
   - 61 Unused 61
   - 62 Unused 62
Minimum interval betwwen 2 log entries = 0 ms
Log mode is Continous (log always, MIN/MAX are discarded)
MIN : to log, value of field index 56 must be more than 100
MAX : to log, value of field index 56 must be less than 2500
SD card use SPI1    (0=SPI,  1=SPI1)
GPIO's used by sd card are:
    CS   = 9
    MOSI = 11
    MISO = 12
    SCLK  = 10
UART with oXs uses:
    gpio to receive data from oXs = 5
    Baudrate = 115200
Protocol is O (oXs + csv)
Led color is normal (not inverted)

Config parameters are OK
Press ? + Enter to get help about the commands
config has been saved

 >> 
 
max ms between 2 call to main loop= 0
max % used of Serial2 fifo= 0
max % of bytes in csvbuf= 0
max us to fill csv rec= 0
max % used of fromNbr queue= 0
max % used of lastWritten queue= 0
max us to write on sd card= 0
max us to synchro on sd card= 0
 total bytes written in csv file= 0
 
max ms between 2 call to main loop= 0
max % used of Serial2 fifo= 0
max % of bytes in csvbuf= 0
max us to fill csv rec= 0
max % used of fromNbr queue= 0
max % used of lastWritten queue= 0
max us to write on sd card= 0
max us to synchro on sd card= 0
 total bytes written in csv file= 0
 
max ms between 2 call to main loop= 0
max % used of Serial2 fifo= 0
max % of bytes in csvbuf= 0
max us to fill csv rec= 0
max % used of fromNbr queue= 0
max % used of lastWritten queue= 0
max us to write on sd card= 0
max us to synchro on sd card= 0
 total bytes written in csv file= 0

Satcomix, 2023-10-14:
After first commands and test, the board doesn`t response after reboot.
No ? and ENTER

Clean boot

Config parameters are OK
Press ? + Enter to get help about the commands
 
Type any character to begin +++++++++++

Satcomix, 2023-10-14:
Only Nuke and a new flash with Version 0.0.3 bring back the RP2040 SD/RTC

Satcomix, 2023-10-14:
Version = 0.0.3
Fields in CSV log (+ = included; - = not included)
   + 0 Gps
   + 1  
   + 2 Gps groundspeed (m/s)
   + 3 Gps Heading
   + 4 Gps Altitude(m)
   + 5 Gps Num Sat
   + 6 Gps Date
   + 7 Gps Time
   + 8 Gps PDOP
   + 9 Gps home bearing
   + 10 Gps home dist
   + 11 Volt 1 (V)
   + 12 Current(A)
   + 13 Volt 3(V)
   + 14 Volt 4(V)
   + 15 Capacity (Ah)
   + 16 Temp 1 
   + 17 Temp 2
   + 18 VSpeed (m/s)
   + 19 Baro Altitude (m)
   + 20 Pitch
   + 21 Roll
   + 22 Yaw
   + 23 RPM
   + 24 ADS_1_1
   + 25 ADS_1_2
   + 26 ADS_1_3
   - 27 ADS_1_4
   - 28 ADS_2_1
   - 29 ADS_2_2
   - 30 ADS_2_3
   + 31 ADS_2_4
   + 32 Airspeed (m/s)
   + 33 Comp VSpeed (m/s)
   + 34 Sbus Hold count
   + 35 Sbus Failsafe Count
   + 36 GPS cumul dist (m)
   + 37 Unused 37
   + 38 Unused 38
   + 39 Unused 39
   + 40 Unused 40
   + 41 Rc1
   + 42 Rc2
   + 43 Rc3
   + 44 Rc4
   + 45 Rc5
   + 46 Rc6
   - 47 Rc7
   - 48 Rc8
   - 49 Rc9
   - 50 Rc10
   - 51 Rc11
   - 52 Rc12
   - 53 Rc13
   - 54 Rc14
   - 55 Rc15
   - 56 Rc16
   - 57 Unused 57
   - 58 Unused 58
   - 59 Unused 59
   - 60 Unused 60
   - 61 Unused 61
   - 62 Unused 62
Minimum interval betwwen 2 log entries = 0 ms
Log mode is Continous (log always, MIN/MAX are discarded)
MIN : to log, value of field index 56 must be more than 100
MAX : to log, value of field index 56 must be less than 2500
SD card use SPI1    (0=SPI,  1=SPI1)
GPIO's used by sd card are:
    CS   = 9
    MOSI = 11
    MISO = 12
    SCLK  = 10
UART with oXs uses:
    gpio to receive data from oXs = 5
    Baudrate = 115200
Protocol is O (oXs + csv)
Led color is normal (not inverted)

Config parameters are OK
Press ? + Enter to get help about the commands
config has been saved

 >>

Satcomix, 2023-10-14:
oXs RP2040 GPio0 to GPio5 oXs Logger with 115200baud, nothing happens.

Version = 0.0.3
Fields in CSV log (+ = included; - = not included)
   + 0 Gps
   + 1  
   + 2 Gps groundspeed (m/s)
   + 3 Gps Heading
   + 4 Gps Altitude(m)
   + 5 Gps Num Sat
   + 6 Gps Date
   + 7 Gps Time
   + 8 Gps PDOP
   + 9 Gps home bearing
   + 10 Gps home dist
   + 11 Volt 1 (V)
   + 12 Current(A)
   + 13 Volt 3(V)
   + 14 Volt 4(V)
   + 15 Capacity (Ah)
   + 16 Temp 1 
   + 17 Temp 2
   + 18 VSpeed (m/s)
   + 19 Baro Altitude (m)
   + 20 Pitch
   + 21 Roll
   + 22 Yaw
   + 23 RPM
   + 24 ADS_1_1
   + 25 ADS_1_2
   + 26 ADS_1_3
   - 27 ADS_1_4
   - 28 ADS_2_1
   - 29 ADS_2_2
   - 30 ADS_2_3
   + 31 ADS_2_4
   + 32 Airspeed (m/s)
   + 33 Comp VSpeed (m/s)
   + 34 Sbus Hold count
   + 35 Sbus Failsafe Count
   + 36 GPS cumul dist (m)
   + 37 Unused 37
   + 38 Unused 38
   + 39 Unused 39
   + 40 Unused 40
   + 41 Rc1
   + 42 Rc2
   + 43 Rc3
   + 44 Rc4
   + 45 Rc5
   + 46 Rc6
   - 47 Rc7
   - 48 Rc8
   - 49 Rc9
   - 50 Rc10
   - 51 Rc11
   - 52 Rc12
   - 53 Rc13
   - 54 Rc14
   - 55 Rc15
   - 56 Rc16
   - 57 Unused 57
   - 58 Unused 58
   - 59 Unused 59
   - 60 Unused 60
   - 61 Unused 61
   - 62 Unused 62
Minimum interval betwwen 2 log entries = 0 ms
Log mode is Continous (log always, MIN/MAX are discarded)
MIN : to log, value of field index 56 must be more than 100
[... 140 lines cut ...]
max % used of fromNbr queue= 0
max % used of lastWritten queue= 0
max us to write on sd card= 0
max us to synchro on sd card= 0
 total bytes written in csv file= 0
 
max ms between 2 call to main loop= 0
max % used of Serial2 fifo= 0
max % of bytes in csvbuf= 0
max us to fill csv rec= 0
max % used of fromNbr queue= 0
max % used of lastWritten queue= 0
max us to write on sd card= 0
max us to synchro on sd card= 0
 total bytes written in csv file= 0
 
max ms between 2 call to main loop= 0
max % used of Serial2 fifo= 0
max % of bytes in csvbuf= 0
max us to fill csv rec= 0
max % used of fromNbr queue= 0
max % used of lastWritten queue= 0
max us to write on sd card= 0
max us to synchro on sd card= 0
 total bytes written in csv file= 0

mstrens, 2023-10-14:
In this github issue,  it is not clear if the fields to be logged have a + or a - in front of the name.
Only those with a + should be logged.

In oXs on rp2040, did you define the gpio to be used to generate the data to the logger?

In oXs logger, do you get some file (even empty) on the Sd card?

Satcomix, 2023-10-14:
It's the default config from logger, most parameters are +, even ads is -.
In the Post above i write that i connect oXs GPio0 to oXs Logger GPio5.
When i aktivste on normal oXs BoardLog=0 , i get massive Sensor lost.
I will make the next tests tomorow morning.
Greetings,
Torsten

mstrens, 2023-10-14:
If you have many sensors defined in normal oXs, it could be that oXs has to many values to transmit and that 115200 baud is to slow to transmit all of them.
Could you start (to morrow) with a test where oXs does not get any RC channels (e.g. no Sbus) and where there is only one voltage to measure (so disabling gps, I2C, ...). In this case, oXs should log much less data and (I hope) avoid sensor lost.
This is the only test I made a few weeks ago and it was OK. Still I had no connection with a receiver and so I had also no way to detect sensor lost.

If this works, then we can try to let oXs generates step per step more data e.g. the RC channels (this supposes to connect again oXs to a receiver connected to an handset).

Satcomix, 2023-10-15:
Good morning Mstrens,
First Test with Version 2.8.29
TLM=1
LOG=0
V1=26
All other GPio=255
Protocol: S.PORT
With FBUS protocol there are errors with the FBUS Frame Size.
The Challenger RP2040 SD/RTC Board doesn`t work anymore with version 0.0.3 after configuration the MISO, MOSI, CS, SLCK GPio`s. No way to get any information over terminal.

Clean boot

Config parameters are OK
Press ? + Enter to get help about the commands
 
Type any character to begin +++++++++++

mstrens, 2023-10-15:
About the challenger that does not react to commands:
It could be that there is some issue while changing SPI parameters because it is not possible to change the SPI number and the SPI pins simultaneously.

I made a small change in 0.0.4 in order to avoid configuring SPI as long the config is not OK.
Hoping it help.

In your post you say that protocol is Sport but you say also that with  FBUS protocol there are errors with the FBUS Frame Size. 
I presume that it concerns 2 different tests?

mstrens, 2023-10-15:
Note: I just made a small test with my board (using SPI and not SPI1) and even I do not connect an oXs device, the logger creates a new file on the SD card when I power on the logger.

I get e.g. those messages:
log file will be in oXsLog001.csv
No data within 7867
No data within 5001
No data within 5001

The No data within... is because I have no oXs connected

Satcomix, 2023-10-15:
When I set the protocol to FBUS, I keep getting the error message about FBUS frame size. Then I switched to S.PORT and the error message went away. So it's somewhere related to the PWM or SBUS.
I will test the 0.0.4 now.

Satcomix, 2023-10-15:
First test with 0.0.4
SD=1
MISO=12
MOSI=11
SCLK=10
CS=9

Version = 0.0.4
Fields in CSV log (+ = included; - = not included)
   + 0 Gps
   + 1  
   + 2 Gps groundspeed (m/s)
   + 3 Gps Heading
   + 4 Gps Altitude(m)
   + 5 Gps Num Sat
   + 6 Gps Date
   + 7 Gps Time
   + 8 Gps PDOP
   + 9 Gps home bearing
   + 10 Gps home dist
   + 11 Volt 1 (V)
   + 12 Current(A)
   + 13 Volt 3(V)
   + 14 Volt 4(V)
   + 15 Capacity (Ah)
   + 16 Temp 1 
   + 17 Temp 2
   + 18 VSpeed (m/s)
   + 19 Baro Altitude (m)
   + 20 Pitch
   + 21 Roll
   + 22 Yaw
   + 23 RPM
   + 24 ADS_1_1
   + 25 ADS_1_2
   + 26 ADS_1_3
   - 27 ADS_1_4
   - 28 ADS_2_1
   - 29 ADS_2_2
   - 30 ADS_2_3
   + 31 ADS_2_4
   + 32 Airspeed (m/s)
   + 33 Comp VSpeed (m/s)
   + 34 Sbus Hold count
   + 35 Sbus Failsafe Count
   + 36 GPS cumul dist (m)
   + 37 Unused 37
   + 38 Unused 38
   + 39 Unused 39
   + 40 Unused 40
   + 41 Rc1
   + 42 Rc2
   + 43 Rc3
   + 44 Rc4
   + 45 Rc5
   + 46 Rc6
   - 47 Rc7
   - 48 Rc8
   - 49 Rc9
   - 50 Rc10
   - 51 Rc11
   - 52 Rc12
   - 53 Rc13
   - 54 Rc14
   - 55 Rc15
   - 56 Rc16
   - 57 Unused 57
   - 58 Unused 58
   - 59 Unused 59
   - 60 Unused 60
   - 61 Unused 61
   - 62 Unused 62
Minimum interval betwwen 2 log entries = 0 ms
Log mode is Continous (log always, MIN/MAX are discarded)
MIN : to log, value of field index 56 must be more than 100
MAX : to log, value of field index 56 must be less than 2500
SD card use SPI1    (0=SPI,  1=SPI1)
GPIO's used by sd card are:
    CS   = 9
    MOSI = 11
    MISO = 12
    SCLK  = 10
UART with oXs uses:
    gpio to receive data from oXs = 5
    Baudrate = 115200
Protocol is O (oXs + csv)
Led color is normal (not inverted)

Config parameters are OK
Press ? + Enter to get help about the commands
config has been saved

 >> 
 
max ms between 2 call to main loop= 0
max % used of Serial2 fifo= 0
max % of bytes in csvbuf= 0
max us to fill csv rec= 0
max % used of fromNbr queue= 0
max % used of lastWritten queue= 0
max us to write on sd card= 0
max us to synchro on sd card= 0
 total bytes written in csv file= 0

Satcomix, 2023-10-15:
After repower, no chance to set a ? and ENTER in IDE Terminal.
Clean boot

Config parameters are OK
Press ? + Enter to get help about the commands
 
Type any character to begin +++++++++++

mstrens, 2023-10-15:
It is strange.
I will try to change the wiring of my board in order to use the same SPI and SPI pins as you.

mstrens, 2023-10-15:
I found the bug. it was about initialisation of SPI1.
On my side a file is created on Sd card.
So version is 0.0.5

Note : I changed the default parameters in logger config.h in order to match your setting (and so use SPI1 from a clean installation.

About the issue with fbus in oXs, can you give me the exact message.
Do you still have this message when the logging is not activated.

Anyway, I suggest to test the logger  step by step; so first with sport and only one sensor.
Then perhaps more sensors
Then with Sbus and no receiver connected
Then with Sbus and a receiver connected to the handset.

Satcomix, 2023-10-15:
Version 0.0.5 works, but there is an error with the SD card.
The strange thing is:
After repowering, the terminal display only works after I? and Enter, the board does not respond to a normal ENTER.

Clean boot

Config parameters are OK
Press ? + Enter to get help about the commands
 
Type any character to begin +++++++++++
**error in sd.begin**
 
max ms between 2 call to main loop= 0
max % used of Serial2 fifo= 0
max % of bytes in csvbuf= 0
max us to fill csv rec= 0
max % used of fromNbr queue= 0
max % used of lastWritten queue= 0
max us to write on sd card= 0
max us to synchro on sd card= 0
 total bytes written in csv file= 0

mstrens, 2023-10-15:
This is a debug version. Currently I ask the user to enter a character at power on (like "a"+"enter") to continue the start up.
So you have more opportunity to get all messages on the serial terminal.

Is your sd card formatted in fat32. 
Currently the logger needs a card formatted in fat32.

Satcomix, 2023-10-15:
I changed the SD-CARD from 64GB to 32GB
Now i can see the messages

    gpio to receive data from oXs = 5
    Baudrate = 115200
Protocol is O (oXs + csv)
Led color is normal (not inverted)

Config parameters are OK
Press ? + Enter to get help about the commands
No data within 5001
 
max ms between 2 call to main loop= 9
max % used of Serial2 fifo= 0
max % of bytes in csvbuf= 0
max us to fill csv rec= 0
max % used of fromNbr queue= 0
max % used of lastWritten queue= 0
max us to write on sd card= 0
max us to synchro on sd card= 0
 total bytes written in csv file= 0
No data within 5001
No data within 5001
No data within 5001
No data within 5001
 
max ms between 2 call to main loop= 1
max % used of Serial2 fifo= 0
max % of bytes in csvbuf= 0
max us to fill csv rec= 0
max % used of fromNbr queue= 0
max % used of lastWritten queue= 0
max us to write on sd card= 0
max us to synchro on sd card= 0
 total bytes written in csv file= 0
No data within 5001
No data within 5001

mstrens, 2023-10-15:
Fine.
Now, if you connect oXs with a voltage sensor, I presume, it would record some data

Satcomix, 2023-10-15:
Now it works with oXs Version 2.8.29 
Protocol S.PORT 
TLM=1
LOG=0
V1=26
any GPio= 255

Version = 0.0.5
Fields in CSV log (+ = included; - = not included)
   + 0 Gps
   + 1  
   + 2 Gps groundspeed (m/s)
   + 3 Gps Heading
   + 4 Gps Altitude(m)
   + 5 Gps Num Sat
   + 6 Gps Date
   + 7 Gps Time
   + 8 Gps PDOP
   + 9 Gps home bearing
   + 10 Gps home dist
   + 11 Volt 1 (V)
   + 12 Current(A)
   + 13 Volt 3(V)
   + 14 Volt 4(V)
   + 15 Capacity (Ah)
   + 16 Temp 1 
   + 17 Temp 2
   + 18 VSpeed (m/s)
   + 19 Baro Altitude (m)
   + 20 Pitch
   + 21 Roll
   + 22 Yaw
   + 23 RPM
   + 24 ADS_1_1
   + 25 ADS_1_2
   + 26 ADS_1_3
   - 27 ADS_1_4
   - 28 ADS_2_1
   - 29 ADS_2_2
   - 30 ADS_2_3
   + 31 ADS_2_4
   + 32 Airspeed (m/s)
   + 33 Comp VSpeed (m/s)
   + 34 Sbus Hold count
   + 35 Sbus Failsafe Count
   + 36 GPS cumul dist (m)
   + 37 Unused 37
   + 38 Unused 38
   + 39 Unused 39
   + 40 Unused 40
   + 41 Rc1
   + 42 Rc2
   + 43 Rc3
   + 44 Rc4
   + 45 Rc5
   + 46 Rc6
   - 47 Rc7
   - 48 Rc8
   - 49 Rc9
   - 50 Rc10
   - 51 Rc11
   - 52 Rc12
   - 53 Rc13
   - 54 Rc14
   - 55 Rc15
   - 56 Rc16
   - 57 Unused 57
   - 58 Unused 58
   - 59 Unused 59
   - 60 Unused 60
   - 61 Unused 61
   - 62 Unused 62
Minimum interval betwwen 2 log entries = 0 ms
Log mode is Continous (log always, MIN/MAX are discarded)
MIN : to log, value of field index 56 must be more than 100
MAX : to log, value of field index 56 must be less than 2500
SD card use SPI1    (0=SPI,  1=SPI1)
GPIO's used by sd card are:
    CS   = 9
    MOSI = 11
    MISO = 12
    SCLK  = 10
UART with oXs uses:
    gpio to receive data from oXs = 5
    Baudrate = 115200
Protocol is O (oXs + csv)
Led color is normal (not inverted)

Config parameters are OK
Press ? + Enter to get help about the commands
 
max ms between 2 call to main loop= 11
max % used of Serial2 fifo= 0
max % of bytes in csvbuf= 1
max us to fill csv rec= 138
max % used of fromNbr queue= 1
max % used of lastWritten queue= 1
max us to write on sd card= 4973
max us to synchro on sd card= 21823
 total bytes written in csv file= 75495

Satcomix, 2023-10-15:
After switching to PROTOCOL=F
TLM=255
PRI=5
LOG=0
V1=26

fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame

Satcomix, 2023-10-15:
**Next Test:**

**oXs_RP2040:**

processing cmd

Version = 2.8.29 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    1  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    4  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)
Logger  . . . . . . . . . =    0  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =  255  (ESC_PIN= 0, 1, 2, ..., 29)

Esc type is not defined

Protocol is Sport (Frsky)
CRSF baudrate   = 420000
Logger baudrate = 115200
PWM is generated at = 50 Hz

Voltage parameters:
    Scales : 1.000000 , 1.000000 , 0.100000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 2.000000 , 0.000000 
    One temperature sensor is connected on V3
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 449 , 85 , -2802
     Gyro offsets         X, Y, Z = 3 , -163 , -192
Aispeed sensor is detected using MS4525
    Vspeed compensation channel = 7
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Foreseen GPS type is Ublox (configured by oXs) :GPS is detected but has not (yet) a fix
Led color is normal (not inverted)
Failsafe uses predefined values
     Chan 1...4  =  1500  1500  1500  1500
     Chan 5...8  =  1500  1500  1500  1500
     Chan 9...12 =  1500  1500  1500  1500
     Chan 13...16=  1500  1500  1500  1500

Number of: sequencers=6   sequences=12   steps= 12
Sequencer = [  Gpio  Type(0=servo,1=analog) Clock(msec) ChannelNr Default Min Max ]
Sequence  = (  RC_value(-100...100)  to_Repeat  Uninterrupted   Only_priority_interrupted   is_a_Priority_seq  )
Step      = {  Smooth(clocks) Pwm(-100...100) Keep(clocks)  }
SEQ=
[ 6 0 20 3 -100 -100 100 ] 
  ( 50 P )
    {0 -100 255} 
  ( 100 O )
    {0 100 255} 
[ 7 0 20 3 -100 -100 100 ] 
  ( 40 P )
    {0 -100 255} 
  ( 90 O )
    {0 100 255} 
[ 8 1 500 3 0 0 100 ] 
  ( 30 P )
    {0 0 255} 
  ( 80 O )
    {0 100 255} 
[ 9 0 20 3 -100 -100 100 ] 
  ( 20 P )
    {0 -100 255} 
  ( 70 O )
    {0 100 255} 
[ 10 0 20 3 -100 -100 100 ] 
  ( 10 P )
    {0 -100 255} 
  ( 60 O )
    {0 100 255} 
[ 11 0 20 3 -100 -100 100 ] 
  ( -100 O )
    {0 100 255} 
  ( -50 P )
    {0 -100 255} 

Config parameters are OK
Press ? + Enter to get help about the commands

**Challenger RP2040 SD/RTC**

Clean boot

Config parameters are OK
Press ? + Enter to get help about the commands
 
Type any character to begin +++++++++++
log file will be in oXsLog001.csv
No data within 7273
Reset writeIdx
Reset writeIdx
Reset writeIdx
Reset writeIdx
Reset writeIdx
 
max ms between 2 call to main loop= 7273
max % used of Serial2 fifo= 0
max % of bytes in csvbuf= 13
max us to fill csv rec= 274
max % used of fromNbr queue= 8
max % used of lastWritten queue= 6
max us to write on sd card= 6131
max us to synchro on sd card= 25461
 total bytes written in csv file= 289270
Reset writeIdx
Reset writeIdx
Reset writeIdx
Reset writeIdx
Reset writeIdx
Reset writeIdx
Reset writeIdx
Reset writeIdx
Reset writeIdx
Reset writeIdx
 
max ms between 2 call to main loop= 1
max % used of Serial2 fifo= 0
max % of bytes in csvbuf= 10
max us to fill csv rec= 270
max % used of fromNbr queue= 6
max % used of lastWritten queue= 4
max us to write on sd card= 5361
max us to synchro on sd card= 21845
 total bytes written in csv file= 775135
Reset writeIdx
Reset writeIdx
Reset writeIdx
Reset writeIdx
Reset writeIdx
Reset writeIdx
Reset writeIdx
Reset writeIdx
Reset writeIdx
Reset writeIdx
 
max ms between 2 call to main loop= 1
max % used of Serial2 fifo= 0
max % of bytes in csvbuf= 12
max us to fill csv rec= 278
max % used of fromNbr queue= 6
max % used of lastWritten queue= 5
max us to write on sd card= 6232
max us to synchro on sd card= 25837
 total bytes written in csv file= 1258795
Reset writeIdx
Reset writeIdx
Reset writeIdx

Satcomix, 2023-10-15:
With the last config (see above post) the sequencer don`t work anymore.
You can see the push on the button1-3 in PWM, but nothing happened at GPio6-8

Number of: sequencers=6   sequences=12   steps= 12
Sequencer = [  Gpio  Type(0=servo,1=analog) Clock(msec) ChannelNr Default Min Max ]
Sequence  = (  RC_value(-100...100)  to_Repeat  Uninterrupted   Only_priority_interrupted   is_a_Priority_seq  )
Step      = {  Smooth(clocks) Pwm(-100...100) Keep(clocks)  }
SEQ=
[ 6 0 20 3 -100 -100 100 ] 
  ( 50 P )
    {0 -100 255} 
  ( 100 O )
    {0 100 255} 
[ 7 0 20 3 -100 -100 100 ] 
  ( 40 P )
    {0 -100 255} 
  ( 90 O )
    {0 100 255} 
[ 8 1 500 3 0 0 100 ] 
  ( 30 P )
    {0 0 255} 
  ( 80 O )
    {0 100 255} 
[ 9 0 20 3 -100 -100 100 ] 
  ( 20 P )
    {0 -100 255} 
  ( 70 O )
    {0 100 255} 
[ 10 0 20 3 -100 -100 100 ] 
  ( 10 P )
    {0 -100 255} 
  ( 60 O )
    {0 100 255} 
[ 11 0 20 3 -100 -100 100 ] 
  ( -100 O )
    {0 100 255} 
  ( -50 P )
    {0 -100 255} 

Config parameters are OK
Press ? + Enter to get help about the commands
processing cmd

Cmd to execute:   PWM
PWM values us (sbus) 1... 8   1500(  992)  1500(  992)  2012( 1811)  1500(  992)  1500(  992)  1500(  992)  1500(  992)  1500(  992)
PWM values us (sbus) 9...16   1500(  992)  1500(  992)  1500(  992)  1500(  992)  1500(  992)  1500(  992)  1500(  992)  1500(  992)

mstrens, 2023-10-15:
I do not understand why sequencer do not work.
Previously it worked and I did not change the code.
I see that you use a keep delay of 255.
In the new logic of sequencer, 255 does not mean anymore that the PWM must be hold.
it now means that it has to wait 255*20ms = about 5 sec  before changing the PWM value.
Did you wait the 5 sec?

Satcomix, 2023-10-15:
Hello Mstrens,
With FBUS and LOG=255 it works perfect!, and you dont`t have to wait 5sec. When i push the button, the reaction of the servo is without delay.

Satcomix, 2023-10-15:
No function of the sequencer with SBUS and S.PORT. Telemetry is ok.

processing cmd

Version = 2.8.29 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    1  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    4  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)
Logger  . . . . . . . . . =  255  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =  255  (ESC_PIN= 0, 1, 2, ..., 29)

Esc type is not defined

Protocol is Sport (Frsky)
CRSF baudrate   = 420000
Logger baudrate = 115200
PWM is generated at = 50 Hz

Voltage parameters:
    Scales : 1.000000 , 1.000000 , 0.100000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 2.000000 , 0.000000 
    One temperature sensor is connected on V3
RPM multiplier = 1.000000
Baro sensor is not detected
Acc/Gyro is not detected
Airspeed sensor is not detected
    Vspeed compensation channel = 7
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox (configured by oXs) :GPS is not (yet) detected
Led color is normal (not inverted)
Failsafe uses predefined values
     Chan 1...4  =  1500  1500  1500  1500
     Chan 5...8  =  1500  1500  1500  1500
     Chan 9...12 =  1500  1500  1500  1500
     Chan 13...16=  1500  1500  1500  1500

Number of: sequencers=6   sequences=12   steps= 12
Sequencer = [  Gpio  Type(0=servo,1=analog) Clock(msec) ChannelNr Default Min Max ]
Sequence  = (  RC_value(-100...100)  to_Repeat  Uninterrupted   Only_priority_interrupted   is_a_Priority_seq  )
Step      = {  Smooth(clocks) Pwm(-100...100) Keep(clocks)  }
SEQ=
[ 6 0 20 3 -100 -100 100 ] 
  ( 50 P )
    {0 -100 255} 
  ( 100 O )
    {0 100 255} 
[ 7 0 20 3 -100 -100 100 ] 
  ( 40 P )
    {0 -100 255} 
  ( 90 O )
    {0 100 255} 
[ 8 1 500 3 0 0 100 ] 
  ( 30 P )
    {0 0 255} 
  ( 80 O )
    {0 100 255} 
[ 9 0 20 3 -100 -100 100 ] 
  ( 20 P )
    {0 -100 255} 
  ( 70 O )
    {0 100 255} 
[ 10 0 20 3 -100 -100 100 ] 
  ( 10 P )
    {0 -100 255} 
  ( 60 O )
    {0 100 255} 
[ 11 0 20 3 -100 -100 100 ] 
  ( -100 O )
    {0 100 255} 
  ( -50 P )
    {0 -100 255} 

Config parameters are OK
Press ? + Enter to get help about the commands

Satcomix, 2023-10-15:
processing cmd

Version = 2.8.29 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    1  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    4  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)
Logger  . . . . . . . . . =  255  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =  255  (ESC_PIN= 0, 1, 2, ..., 29)

Esc type is not defined

Protocol is Sport (Frsky)
CRSF baudrate   = 420000
Logger baudrate = 115200
PWM is generated at = 50 Hz

Voltage parameters:
    Scales : 1.000000 , 1.000000 , 0.100000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 2.000000 , 0.000000 
    One temperature sensor is connected on V3
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 449 , 85 , -2802
     Gyro offsets         X, Y, Z = 3 , -163 , -192
Aispeed sensor is detected using MS4525
    Vspeed compensation channel = 7
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Foreseen GPS type is Ublox (configured by oXs) :GPS is detected and has a fix
Led color is normal (not inverted)
Failsafe uses predefined values
     Chan 1...4  =  1500  1500  1500  1500
     Chan 5...8  =  1500  1500  1500  1500
     Chan 9...12 =  1500  1500  1500  1500
     Chan 13...16=  1500  1500  1500  1500

Number of: sequencers=6   sequences=12   steps= 12
Sequencer = [  Gpio  Type(0=servo,1=analog) Clock(msec) ChannelNr Default Min Max ]
Sequence  = (  RC_value(-100...100)  to_Repeat  Uninterrupted   Only_priority_interrupted   is_a_Priority_seq  )
Step      = {  Smooth(clocks) Pwm(-100...100) Keep(clocks)  }
SEQ=
[ 6 0 20 3 -100 -100 100 ] 
  ( 50 P )
    {0 -100 255} 
  ( 100 O )
    {0 100 255} 
[ 7 0 20 3 -100 -100 100 ] 
  ( 40 P )
    {0 -100 255} 
  ( 90 O )
    {0 100 255} 
[ 8 1 500 3 0 0 100 ] 
  ( 30 P )
    {0 0 255} 
  ( 80 O )
    {0 100 255} 
[ 9 0 20 3 -100 -100 100 ] 
  ( 20 P )
    {0 -100 255} 
  ( 70 O )
    {0 100 255} 
[ 10 0 20 3 -100 -100 100 ] 
  ( 10 P )
    {0 -100 255} 
  ( 60 O )
    {0 100 255} 
[ 11 0 20 3 -100 -100 100 ] 
  ( -100 O )
    {0 100 255} 
  ( -50 P )
    {0 -100 255} 

Config parameters are OK
Press ? + Enter to get help about the commands
processing cmd

Cmd to execute:   PWM
PWM values us (sbus) 1... 8    881(    0)   881(    0)   881(    0)   881(    0)   881(    0)   881(    0)   881(    0)   881(    0)
PWM values us (sbus) 9...16    881(    0)   881(    0)   881(    0)   881(    0)   881(    0)   881(    0)   881(    0)   881(    0)

mstrens, 2023-10-15:
You said:
With FBUS and LOG=255 it works perfect!, and you dont`t have to wait 5sec. When i push the button, the reaction of the servo is without delay.

Indeed it is the expected behavior.

mstrens, 2023-10-15:
I am a little lost with the different tests.
1) Is it so that sequencer with FBus works when logger is not activated
2) Is it so that sequencer with FBus does not work when logger is activated
3) Is it so that sequencer with Sport + Sbus never works

Satcomix, 2023-10-15:
I'm a bit lost right now too, because I don't know what I've already tested because of all the tests.

mstrens, 2023-10-15:
Just a little comment:
You have e.g. a sequencer with 
[ 6 0 20 3 -100 -100 100 ]
( 50 P ) {0 -100 255}
( 100 O ) {0 100 255}

There is nothing wrong with it; but you would have exactly the same result without the O and the P if you want that a new function is applied immediately.
[ 6 0 20 3 -100 -100 100 ]
( 50 ) {0 -100 255}
( 100 ) {0 100 255}

mstrens, 2023-10-15:
Perhaps we can try to let first focus on sequencer without logger activated for SBUS and FBUS.

Satcomix, 2023-10-15:
I use the P in Sequence as a priority button

mstrens, 2023-10-15:
The P is not required when the other sequence has no U and no O

Satcomix, 2023-10-15:
I know

Satcomix, 2023-10-15:
**"Perhaps we can try to let first focus on sequencer without logger activated for SBUS and FBUS".**
I will do the next tests only for sequencer with S.PORT, SBUS and one test with FBUS

Satcomix, 2023-10-15:
**Test with S.PORT and SBUS: no Function on Sequencer,  all telemetry data ok
I can see the PWM values of CH1-16 and SBUS_out work on all channels**
On SBUS_out i can see at CH3 the button press.

processing cmd

Version = 2.8.29 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    1  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    4  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)
Logger  . . . . . . . . . =  255  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =  255  (ESC_PIN= 0, 1, 2, ..., 29)

Esc type is not defined

Protocol is Sport (Frsky)
CRSF baudrate   = 420000
Logger baudrate = 115200
PWM is generated at = 50 Hz

Voltage parameters:
    Scales : 1.000000 , 1.000000 , 0.100000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 2.000000 , 0.000000 
    One temperature sensor is connected on V3
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 449 , 85 , -2802
     Gyro offsets         X, Y, Z = 3 , -163 , -192
Aispeed sensor is detected using MS4525
    Vspeed compensation channel = 7
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Foreseen GPS type is Ublox (configured by oXs) :GPS is detected and has a fix
Led color is normal (not inverted)
Failsafe uses predefined values
     Chan 1...4  =  1500  1500  1500  1500
     Chan 5...8  =  1500  1500  1500  1500
     Chan 9...12 =  1500  1500  1500  1500
     Chan 13...16=  1500  1500  1500  1500

Number of: sequencers=6   sequences=12   steps= 12
Sequencer = [  Gpio  Type(0=servo,1=analog) Clock(msec) ChannelNr Default Min Max ]
Sequence  = (  RC_value(-100...100)  to_Repeat  Uninterrupted   Only_priority_interrupted   is_a_Priority_seq  )
Step      = {  Smooth(clocks) Pwm(-100...100) Keep(clocks)  }
SEQ=
[ 6 0 20 3 -100 -100 100 ] 
  ( 50 P )
    {0 -100 1} 
  ( 100 O )
    {0 100 1} 
[ 7 0 20 3 -100 -100 100 ] 
  ( 40 P )
    {0 -100 1} 
  ( 90 O )
    {0 100 1} 
[ 8 1 500 3 0 0 100 ] 
  ( 30 P )
    {0 0 1} 
  ( 80 O )
    {0 100 1} 
[ 9 0 20 3 -100 -100 100 ] 
  ( 20 P )
    {0 -100 1} 
  ( 70 O )
    {0 100 1} 
[ 10 0 20 3 -100 -100 100 ] 
  ( 10 P )
    {0 -100 1} 
  ( 60 O )
    {0 100 1} 
[ 11 0 20 3 -100 -100 100 ] 
  ( -100 O )
    {0 100 1} 
  ( -50 P )
    {0 -100 1} 

Config parameters are OK
Press ? + Enter to get help about the commands

Satcomix, 2023-10-15:
I deactivate the ACC, but no change.
No Vspeed compensation channel defined; oXs uses default settings
ACC=255

mstrens, 2023-10-15:
I loaded 2.8.30.
I hope it solves the bug about sequencer not working with sbus

Satcomix, 2023-10-15:
**In version 2.8.30 the squencer works perfect with SBUS and S:PORT Protocol. FBUS protocol is also ok.**

mstrens, 2023-10-15:
I also uploaded 2.8.31.
It will perhaps solve the issue with messages with Fbus messages (about first pos) when logger is activated.

Satcomix, 2023-10-15:
**In Terminal it shows Version 2.8.30
Sensor lost every time.**
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame
fbus: first pos not a valid length frame

mstrens, 2023-10-15:
I expect that there was an issue with the upload on github.
I made a reload under version 2.8.32.
If the message is still generated, the Hex value of the first pos should be part of the message

Satcomix, 2023-10-15:
**Version 2.8.32
FBUS Protocol
Log=255**

fbus: first pos not a valid length frame; value is F0
fbus: first pos not a valid length frame; value is F0
fbus: first pos not a valid length frame; value is F0
fbus: first pos not a valid length frame; value is F0
fbus: first pos not a valid length frame; value is F0
fbus: first pos not a valid length frame; value is F0
fbus: first pos not a valid length frame; value is F0
fbus: first pos not a valid length frame; value is 0
fbus: first pos not a valid length frame; value is 0
fbus: first pos not a valid length frame; value is F0
fbus: first pos not a valid length frame; value is F0
fbus: first pos not a valid length frame; value is F0
fbus: first pos not a valid length frame; value is 0
fbus: first pos not a valid length frame; value is F0
fbus: first pos not a valid length frame; value is F8
fbus: first pos not a valid length frame; value is F0
fbus: first pos not a valid length frame; value is F8
fbus: first pos not a valid length frame; value is F0
fbus: first pos not a valid length frame; value is 0
fbus: first pos not a valid length frame; value is 0

mstrens, 2023-10-15:
So, this seems Ok
It seems that it writes a lot of data on the sd card.
Did you already look at the content of the csv file on the sd card (with e.g. excel)?

Satcomix, 2023-10-15:
I have a total of 3 files on the SD card.
I looked at the last one with openoffice calc and everything seems plausible to me

Satcomix, 2023-10-15:
Hello Mstrens,
I think that version 2.8.32 and Logger 0.0.5 run well.
I would hope that others who have expressed their desire for these features would also help with the tests in the future.
Thank you for your work.
I'll close the issue now.
Greetings,
Torsten

closed 2023-10-15 by Satcomix

Satcomix, 2023-10-18:
Hello Mstrens,
I looked at the .csv file from 0.0.11 Logger with OpenOffice.
The values ​​for GPS position seem implausible to me and the values ​​for GPS date/time are not displayed.

[oXsLog000.zip](https://github.com/mstrens/oXs_on_RP2040/files/13021864/oXsLog000.zip)

Greetings,
Torsten

Satcomix, 2023-10-18:
Hello Mstrens,
If I understand you correctly, you want a logger .csv file only with GPS and one log file from my handset.
Since I have never saved logs on the handset, I first have to see how I can do it.

mstrens, 2023-10-18:
It does not matter if the oXs logger file (and the handset log file) contains only GPS data.
It can contains other data too.
What matter is that the GPS got a fix and so that the gps position fields are present (in order to look at the format)

In openTX, I think you have use the menu special function in order to activate the log.
Then you have also (I do not remember where but probably in telemetry menu) to specify which fields must be included.
I presume it is the same for Edge or Ethos.

Satcomix, 2023-10-18:
Hello Mstrens,
I just noticed something else:
If I deactivate V1-V4 with 255, and there was a temp sensor on V3 and it is not deleted with 255, the GPS does not initialize via TX/RX.

mstrens, 2023-10-18:
When you say that you deactivate V1-V4 do you mean that you deactivate only V1 and V4 (or that you deactivate V1,V2,V3,V4).

Anyway, I am not sure that the issue is related to activating or not some voltage.
I suggest that when you change some parameters, you always make a power off.
After saving the parameters (after a change), oXs perform a reset but the GPS (and other I2C devices) are not automatically reset. So it could be that they do not react as expected (e.g. the GPS is not anymore expecting commands at 9600 bauds in case of M8)

Satcomix, 2023-10-18:
I deactivate V1 V2 V3 V4 with 255,
but i forget to deactivate TEMP=1, and so V3 was a Tempsensor.
I make power off and on but nothing happens. GPS cant initialize with 10HZ and no telemetry data was on my handset.

[52 more replies]