mstrens/oXs_on_RP2040 #112: ZTW Mantis G2 ESC telemetry support

onki69, 2023-10-13:
ZTW is one of the major ESC manufactureres and many other brands offer these ESCs as OEM solution.
https://www.ztwoem.com/product/mantis-g2-85a-105a-125a/
The current Mantis G2 series come with a RX/TX port and using the BT Adapter telemetry values can be read out.
Maybe OxS can take advantage of this in order to implament a telemetry feature for this series similar to the Hobbywing Telemetry support.
Unfortunately I have no information about the data protocol and my reverse engineering skills are very bad to non existing.

mstrens, 2023-10-13:
Without doc about the protocol, it is not possible to implement it.

onki69, 2023-10-24:
I got the protocol information from the ESC manufacturer.
How can I send it to you?

mstrens, 2023-10-24:
I propose that you keep it currently.
I try to focus on gyro now.

onki69, 2023-10-25:
OK -no problem.
Just let me know when you like to take a look at it.

mstrens, 2023-10-31:
I hope to be able to look at the doc quite soon.
Thanks to provide me the doc for ZTW ESC.
I presume you can put it here as a zip file or provide a link to something like google drive

onki69, 2023-11-03:
This the info I got:

This data package is used by the ESC for reporting data in real time, and it reports every 50ms. UART Baud 115200 Bps. 
If the reported voltage is 58.2V, the data will be 0x02, 0x46.
   Byte0: Packet Header - ---  0xDD
   Byte1: Protocol No   ----  0x01
   Byte2: Data length   -----  0x20
   Byte3: Voltage-H      
   Byte4: Voltage-L      (10-1V, data range: 0  0x3e8  100V)
   Byte5: Current-H
   Byte6: Current-L       (10-1A, data range: 0  0x1388 500A) 
   Byte7: Throttle-PCT    (0x01-1%, data range: 0 – 0x64  100% input)
   Byte8: RPM-H
   Byte9: RPM-L          (0x01-10RPM, data range: 0 – 0xffff )
    Byte10: Mos-temp       (0x01 – 1℃, data range: 0 –0x96 150℃) 
Byte11: Motor-temp    (0x01 – 1℃, data range: 0 –0x96 150℃)
    Byte12: Throttle-PWM    (0x01-1% , data range: 0  0x64  100% output)
    Byte13: State-H
Byte14: State-L
Byte15: Mah-used H  high value of the used/consumed power
Byte16: Mah-used L  low value of the used/consumed power

Byte17: UART-TH  serial throttle input
Byte18: CAN-TH  can throttle
Byte19: BEC voltage  (0-25V)
    Byte20- Byte29: reserved
    Byte30 &31: byte0  byte29 Sum. accumulate & verify
State-L  explanations for statuses
          0x01 Short-circuit protection              0x10 Low-voltage protection
          0x02  motor wire break               0x20 Temperature protection
          0x04  PPM TH loss protection          0x40 Start locked-rotor
          0x08   Power on TH is not zero       0x80 Current protection
          State-H  explanations for statuses
0x01 PPM throttle is not within the regulated range, the PPM throttle is in an abnormal state, and the throttle is not within 700us~2500us.         
0x10: the battery voltage is not within the regulated range.
0x02 UART Throttle is not within the regulated range, UART throttle is in an abnormal state, the throttle value exceeds 1000. 
0x04 UART throttle loss, UART TH loss  
0x08 CAN throttle loss, CAN TH loss

mstrens, 2023-11-03:
Thanks for the info.
I put 2.9.16 on github (test branch).
I hope it support ZTW Mantis ESC.
Please note that even if other ESC where already coded in oXs, I think they where not tested.
So there could be some more bugs than expected.
Let me know if there are bugs.
There is a new value for usb command : ESC_TYPE=ZTW1

onki69, 2023-11-03:
Wow - that was quick. Testing may take some time as I aam not at home at the moment.
I will repond as soon as get some results.

mstrens, 2023-11-03:
OK.
Always use the latest version in test branch when you start testing. I am also fixing bugs for gyro.

onki69, 2023-11-12:
I have connected everthing to a FrSky Archer M+ receiver with FBus. PWM is generated correctly since esc is running the motor on ch3. But the LED is blinking red and no EXC telemetry.
her is my config:
`
processing cmd

Version = 2.9.26 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    9  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =  255  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =  255  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =    0  255    2  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255  255  255 (V1 / V4 = 26, 27, 28, 29)
RGB led . . . . . . . . . =   16  (RGB    = 0, 1, 2, ..., 29)
Logger  . . . . . . . . . =  255  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =   10  (ESC_PIN= 0, 1, 2, ..., 29)Esc type is ZTW1 (ZTW mantis)

Protocol is Fbus(Frsky)
CRSF baudrate   = 420000
Logger baudrate = 115200
PWM is generated at = 50 Hz
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is not detected
Acc/Gyro is not detected
Airspeed sensor is not detected
    No Vspeed compensation channel defined; oXs uses default settings
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox (configured by oXs) :GPS is not (yet) detected
Led color is normal (not inverted)
Failsafe type is HOLD

Gyro is not configured

No sequencers are defined

Config parameters are OK
Press ? + Enter to get help about the commands
`

onki69, 2023-11-15:
I changed to a Jeti receiver but still the LED on the 2040 is blinking red although Telemetry is received in the transmitter and servos outputs are working. I also added a temp sensor on V3.
This is the current log output:
`Version = 2.9.26 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    9  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =  255  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =  255  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =    0    1    2    3 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =    4    5    6  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255   29  255 (V1 / V4 = 26, 27, 28, 29)
RGB led . . . . . . . . . =   16  (RGB    = 0, 1, 2, ..., 29)
Logger  . . . . . . . . . =  255  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =   10  (ESC_PIN= 0, 1, 2, ..., 29)Esc type is ZTW1 (ZTW mantis)

Protocol is Jeti (Exbus)
CRSF baudrate   = 420000
Logger baudrate = 115200
PWM is generated at = 50 Hz
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 3.300000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is not detected
Acc/Gyro is not detected
Airspeed sensor is not detected
    No Vspeed compensation channel defined; oXs uses default settings
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox (configured by oXs) :GPS is not (yet) detected
Led color is normal (not inverted)
Failsafe type is HOLD

Gyro is not configured

No sequencers are defined

Config parameters are OK
Press ? + Enter to get help about the commands
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
8091
  A8
  F0
8028
  64
  F0
80E4
  F0
8051
  F0
80F0
8032
  F0
80F0
80F0
80F0
80F0
80F0
80A4
   8
  F0
80F0
80F0
80A4
  F0
80F0
80C4
  8A
  F0
8053
  F0
80C9
  F0
80D3
  12
  F0
80F8
80B9
  F8
8028
  F8
80A8
  F8
8009
  F6
  F0
8069
  F5
  F0
80F0
8053
  F0
8053
  F0
80F0
80F0
80F0
80D5
  F0
80F0
80F0
8052
  29
  F5
  F0
80E5
  F0
8052
   9
  F0
80F0
8011
  F0
80C9
  F5
  F0
80A9
  F0
80A2
  12
  ED
  F0
800D
  F0
80F0
80ED
  F0
80F0
8035
  F0
8085
  F0
80F0
80F0
80E8
  F0
80E8
  5A
  F0
80C8
  6A
  F0
8020
  F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
80F0
`

the 80F0 line onl appears if the motor is not running. The other output is during a motor run.
No ESC telemetry visible on the transmitter (only V3 as Temp sensor).
By the way, is there a possibiilty to change the name or at least the unit when a temp sensor is connected on the A/D input.
I can't change name/unit in the transmitter.

Hope that info is helpful to get the telemetry to work.

mstrens, 2023-11-15:
Thanks for the feedback.
None of the data match the expected one.
When the line begin with 80, it means that it is the first character of a frame (in fact the first char after some delay without char).
The doc says that each frame should contains 32 char and so the first one should be "DD" (so here it should be "80DD").
None of the char you got could be a valid frame.
That explain why oXs does not provide telemetry.

I would have to make a check on my side in order to generate a valid frame with another computer and see if oXs can get it.

About the temp, I can change the text being displayed and the symbol for the unit.
What do you get currently?

mstrens, 2023-11-17:
I made some tests.
I simulate a frame from ZTW and first oXs did not read the expected values.
I made some changes (version 2.10.3) and now oXs get the expected values.
Please try this version and post here the data sent by oXs to the USB/serial.

onki69, 2023-11-17:
I flshed the 2.10.3 and tested it with the same configuration.
The Telemetry fields from the ESC are transmitted to my Jeti transmitter but I get no data from the ESC.
Here is a short data dump from the Terminal:
`Version = 2.10.3 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    9  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =  255  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =  255  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =    0    1    2    3 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =    4    5    6  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255   29  255 (V1 / V4 = 26, 27, 28, 29)
RGB led . . . . . . . . . =   16  (RGB    = 0, 1, 2, ..., 29)
Logger  . . . . . . . . . =  255  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =   10  (ESC_PIN= 0, 1, 2, ..., 29)Esc type is ZTW1 (ZTW mantis)

Protocol is Jeti (Exbus)
CRSF baudrate   = 420000
Logger baudrate = 115200
PWM is generated at = 50 Hz
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 3.300000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is not detected
Airspeed sensor is not detected
    No Vspeed compensation channel defined; oXs uses default settings
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox (configured by oXs) :GPS is not (yet) detected
Led color is normal (not inverted)
Failsafe type is HOLD
Acc/Gyro is not detected

Gyro is not configured

No sequencers are defined
Error in parameters: gyro vertical orientation (0) is not valid

Config parameters are OK
Press ? + Enter to get help about the commands
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
8048
  50
  AC
  E0
8008
  E0
80A4
  88
  E0
80A4
  28
  E0
80E0
80E0
8046
  AB
  E0
8023
  D1
  E0
80E0
80E0
8024
  D1
  E0
8049
  E0
8012
  E0
8052
  E0
8092
  E0
8092
  E0
804A
  A3
  E0
8089
  F4
  E0
8053
  E0
8093
  E0
80A7
  D4
  E0
804A
  E0
80E0
80E0
80E0
8050
  E0
80E0
8092
  E0
8092
  A8
  E0
8012
  AC
  E0
80E0
80E0
8050
  E0
8050
  E0
8050
  E0
8027
  E0
80E0
80E0
8025
  E0
80B2
  E0
80A4
  12
  E0
8054
  E0
80E0
8002
  E0
8022
  E0
8044
  24
  E0
80C2
  E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
80E0
`

Same sequence at the beginning motor is off and I do get the 80E0
Then the motor is started and I do get different data.
Motor is stopped and data is same as after start (80E0).
Hope this helps.

Regarding the temperature it might be most flexible (if this can be done) to have a kind of "Vx_type" command.
By default it is set to "volt" so "Voltage x" is the description and "V" is the unit.
When using for instance "v3_type=temp" the description changes to Temp 3 and unit is °C 

Best regards and many thanks for your work
Onki

mstrens, 2023-11-17:
The characters read by oXs are in no way a valid frame. So it is normal that you do not get data on the handset.
I do not understand why oXs does not get valid frame while it works on my side when I simulate sending the frame.

Perhaps the issue is linked to the way the ESC output is build at hardware level.
Could you try to add a pullup resistor of e.g. 1K between the ESC pin and the 3.3V of the RP2040 and see if you get other characters when the motor runs .

onki69, 2023-11-17:
With a3k9 Pullup at the ESC telemetry input it looks like this:

8040
  E0
8040
  E0
8040
  E0
8040
  E0
8040
  E0
8040
  E0
8040
  E0
8040
  E0
80C4
  E0
80E0
80A6
  40
  E0
80C4
  40
  E0
80D3
  64
  40
  E0
8013
  40
  E0
8053
  E4
  E0
80D3
  E0
80E9
  F4
  40
  E0
8040
  E0
8040
  E0
8040
  E0
8040
  E0
8040
  E0
8040
  E0
8040
  E0

mstrens, 2023-11-17:
This does not match the expected frame at all.

I have no more idea.
Could it be that ESC requires some setup to generate telemetry data?
Best would be to have a logic analyser to look at the signal but I know you do not have one.

mstrens, 2023-11-17:
It seems that there are several projects that use a RP2040 as logic analyser.
Here a link to one of them:
https://hackaday.com/2022/03/02/need-a-logic-analyzer-use-your-pico/
I never used it because I have a salea clone.

onki69, 2023-11-18:
I have done a hex dump using my picoScope software.
Probe was connected to the Port 10 with 3k9 pullup.
At the beginning the motor was off, then shortly on.

Maybe this helps
[Dump_motor.csv](https://github.com/mstrens/oXs_on_RP2040/files/13400687/Dump_motor.csv)

mstrens, 2023-11-18:
Thanks for the capture.
What is the baudrate that you used? Is it 115200?
I noticed that the first byte of a frame is DD and the second is 01 as expected in the doc. Still the third one is 00 where the doc says it should be 20.
Anyway, even if there is a difference in the third byte, I do yet understand why oXs does not provide the same content for the frames.

mstrens, 2023-11-18:
For the capture, did you used a set up : 8bits Even parity 1 stop bit?

onki69, 2023-11-18:
For the capture I have used 115200 8N1. Do you need other parameters (8E1)?

mstrens, 2023-11-18:
In the doc (some days ago) , it was said that signal was 8E1. So I used this setup in oXs.
Still I think that it is not 8E1 but 8N1 because if I calculate the difference between 2 values in the 3rd column, I get (often but not always) a value of 87 (us) with is the delay for 8N1 at 115200.
Still, I do not understand why the capture gives some negative values in the 3rd column and the difference is not always 87us.
Do you have an explanation?

onki69, 2023-11-18:
I can't find the 8E1 setting in my docs.
So I have sent another mail to ZTW asking for the confimation about the comm details.

mstrens, 2023-11-18:
Sorry, it was not specified if it was 8N1 or 8N1.
As kontronik use 8N1, I selected the same option for ZTW.
It was a mistake.
I put on github a version 2.10.5 where I use 8N1.
Can you try this version.
I continue to let it display the received char.
I hope you will now get the same as with the picolog.

steffku, 2024-04-08:
I have a ZTW Skyhawk ESC. Does the current Implementation support this ESC?

mstrens, 2024-04-08:
I implemented the ZTW telemetry protocol that is documented in a post here above.
Still I am not sure that it has been tested.
So there could be some bug.
I do not know if the skyhawk model supports this ZTW protocol 
Best is to make a test.

I can try to fix eventual bugs if I get some feedback.

steffku, 2024-04-22:
Hmm, until now i cannot see a sensor on my transmitter when i select discover new sensors. Do i need this resistor between TLM and RX?

steffku, 2024-04-24:
These are the frames that i get if i give a little bit power on the motor:
80FE
  FE
  FE
  FE
  FE
  FE
80FE
  FE
  FE
  FE
  FE
  FE
80F8
  E6
80F8
  80
80F8
  86
  E0
80F8
  FE
80F8
  86
  F8
80F8
  86
80F8
  E6
  9E
80F8
  F8
80F8
  E6
80F8
80F8
  FE
80F8
  E6
80FE
  FE
  FE
  FE
  FE
  FE
80FE

I have a Saleae logic analyzer, but don't know how to use it. If you can give me some support here i can have a try.

r3wdi3, 2024-09-11:
@mstrens  Are you still interested in getting this fixed? Tried it today with a Mantis G2 65A but no luck.

I used a RPi2040 which is working flawlessly with a BLHeli_32 ESC and changed the ESC type but no luck. The ESC sends out telemetry to the program card so there is data available.

I do have another RPi2040 setup with the uLA firmware and can get some PWM signals from a receiver in Pulseview. I also get UART binary code when I listen to the ESC, but then my limited knowledge on logic analyzers reaches it's end. ;)

mstrens, 2024-09-11:
Yes I am interested in getting this fixed.
Still I am currently busy with some other topics.
Could you recall me this in 2 weeks.
Thanks

RuudRuudRuud, 2024-09-24:
@mstrens friendly reminder; just ping me when you have time for this and let me know what you need.

r3wdi3, 2024-09-24:
@mstrens friendly reminder; just ping me when you have time for this and let me know what you need.

fl0zan0, 2025-03-28:
Hin any news about that ?