mstrens/oXs_on_RP2040 #113: RTC function RP2040

Satcomix, 2023-10-18:
Good morning Mstrens,
First test without connection to normal oXs.
**First boot:**
Rebooted by Watchdog!

Config parameters are OK
Press ? + Enter to get help about the commands
 
Type any character to begin +++++++++++

Autoprobing for a RTC...
Autoprobe found MCP7941x at 0x6F
Calibration: 0
log file will be in oXsLog004.csv
No data within 19859
No data within 5001
No data within 5001
No data within 5001

Challenger RP2040 SD/RTC board uses I2C0 for the RTC communication:
    gpio to receive data from oXs = 5
    Baudrate = 115200
RTC uses:
    SDA = 
0
    SCL = 
1
    Rtc is installed; date&time = 2001-01-01T00:12:27
Protocol is O (oXs + csv)
Led color is normal (not inverted)

**Second Boot: with actual Date and Time**

Version = 0.0.7
Fields in CSV log (+ = included; - = not included)
   + 0 Gps
   + 1  
   + 2 Gps groundspeed (m/s)
   + 3 Gps Heading
   + 4 Gps Altitude(m)
   + 5 Gps Num Sat
   + 6 Gps Date
   + 7 Gps Time
   + 8 Gps PDOP
   + 9 Gps home bearing
   + 10 Gps home dist
   + 11 Volt 1 (V)
   + 12 Current(A)
   + 13 Volt 3(V)
   + 14 Volt 4(V)
   + 15 Capacity (Ah)
   + 16 Temp 1 
   + 17 Temp 2
   + 18 VSpeed (m/s)
   + 19 Baro Altitude (m)
   + 20 Pitch
   + 21 Roll
   + 22 Yaw
   + 23 RPM
   + 24 ADS_1_1
   + 25 ADS_1_2
   + 26 ADS_1_3
   - 27 ADS_1_4
   - 28 ADS_2_1
   - 29 ADS_2_2
   - 30 ADS_2_3
   + 31 ADS_2_4
   + 32 Airspeed (m/s)
   + 33 Comp VSpeed (m/s)
   + 34 Sbus Hold count
   + 35 Sbus Failsafe Count
   + 36 GPS cumul dist (m)
   + 37 Unused 37
   + 38 Unused 38
   + 39 Unused 39
   + 40 Unused 40
   + 41 Rc1
   + 42 Rc2
   + 43 Rc3
   + 44 Rc4
   + 45 Rc5
   + 46 Rc6
   - 47 Rc7
   - 48 Rc8
   - 49 Rc9
   - 50 Rc10
   - 51 Rc11
   - 52 Rc12
   - 53 Rc13
   - 54 Rc14
   - 55 Rc15
   - 56 Rc16
   - 57 Unused 57
   - 58 Unused 58
   - 59 Unused 59
   - 60 Unused 60
   - 61 Unused 61
   - 62 Unused 62
Minimum interval betwwen 2 log entries = 0 ms
Log mode is Continous (log always, MIN/MAX are discarded)
MIN : to log, value of field index 56 must be more than 100
MAX : to log, value of field index 56 must be less than 2500
SD card use SPI1    (0=SPI,  1=SPI1)
GPIO's used by sd card are:
    CS   = 9
    MOSI = 11
    MISO = 12
    SCLK  = 10
UART with oXs uses:
    gpio to receive data from oXs = 5
    Baudrate = 115200
RTC uses:
    SDA = 
0
    SCL = 
1
    Rtc is installed; date&time = 2033-10-18T08:27:09
Protocol is O (oXs + csv)
Led color is normal (not inverted)

Config parameters are OK
Press ? + Enter to get help about the commands
No data within 5001
No data within 5001
No data within 5001
No data within 5001

Third Boot with LOG=0
SD card use SPI1    (0=SPI,  1=SPI1)
GPIO's used by sd card are:
    CS   = 9
    MOSI = 11
    MISO = 12
    SCLK  = 10
UART with oXs uses:
    gpio to receive data from oXs = 5
    Baudrate = 115200
RTC uses:
    SDA = 
0
    SCL = 
1
    Rtc is installed; date&time = 2033-10-18T08:30:29
Protocol is O (oXs + csv)
Led color is normal (not inverted)

Config parameters are OK
Press ? + Enter to get help about the commands
Reset writeIdx
Reset writeIdx
Reset writeIdx
Reset writeIdx
Reset writeIdx
Reset writeIdx
Reset writeIdx
Reset writeIdx
Reset writeIdx
Reset writeIdx
Reset writeIdx
Reset writeIdx
Reset writeIdx

**I must deactivate some Telemetrie sensors, too much traffic.**

Greetings,
Torsten

Satcomix, 2023-10-18:
First Bug in Ver.:0.0.7
When **you write** DT=**23**-10-18T08:42,
**you get**:
Rtc is installed; date&time = 20**33**-10-18T08:42:04

mstrens, 2023-10-18:
I put version 0.0.8 that is supposed to:
- fix the bug about the year+10
- remove some debug messages.
Note: ResetWriteIdx does not mean that there are to many data. It was just a warning for me.
Do you have another reason why you think that there are to many data.

Note : in version 0.0.7 and 0.0.8 I also removed the messages about the statistics. I can add them again if you want (it is just a flag at compilation like the one that ask to press a key at power on)

Satcomix, 2023-10-18:
"Do you have another reason why you think that there are to many data."
Yes, with LOG=0, i get too many **Sensor lost** messages on the handheld radio.

Satcomix, 2023-10-18:
It works with ver.0.0.8 but there is a little bug in the command:
Cmd to execute:  DT = 23-10-18T09:25 
Wrong command: command is discarded
After another ENTER everything is ok

mstrens, 2023-10-18:
This issue has to be solved in oXs_on_RP2040.
Perhaps can you first try to increase the baudrate between oXs and the logger. You can try e.g. a baudrate of 400000 instead of 115200.
If this does not solve the  issue, then best would be to get capture with a digital logic analyser of 3 or 2 signals (simultanously):
- sbus + Sport + log
or 
- Fbus+log
I do not know the protocol that you used for the test.

The cmd that was rejected is probably because I forgot to remove the space after the "=".
I will change this.

Satcomix, 2023-10-18:
I use FBUS with my FrSky X20S.
I will test S.PORT and SBUS.
Than i will try baudrates up to 400000baud.

Satcomix, 2023-10-18:
When i deactivate RPM=255, i get nomore SENSOR LOST in handheld.
Baudrate=115200baud

processing cmd

Version = 2.8.33 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)
Logger  . . . . . . . . . =    0  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =  255  (ESC_PIN= 0, 1, 2, ..., 29)

Esc type is not defined

Protocol is Fbus(Frsky)
CRSF baudrate   = 420000
Logger baudrate = 115200
PWM is generated at = 50 Hz

Voltage parameters:
    Scales : 1.000000 , 1.000000 , 0.100000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 2.000000 , 0.000000 
    One temperature sensor is connected on V3
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 449 , 85 , -2802
     Gyro offsets         X, Y, Z = 3 , -163 , -192
Aispeed sensor is detected using MS4525
    Vspeed compensation channel = 7
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Foreseen GPS type is Ublox (configured by oXs) :GPS is detected and has a fix
Led color is normal (not inverted)
Failsafe uses predefined values
     Chan 1...4  =  1500  1500  1500  1500
     Chan 5...8  =  1500  1500  1500  1500
     Chan 9...12 =  1500  1500  1500  1500
     Chan 13...16=  1500  1500  1500  1500

Number of: sequencers=2   sequences=4   steps= 7
Sequencer = [  Gpio  Type(0=servo,1=analog) Clock(msec) ChannelNr Default Min Max ]
Sequence  = (  RC_value(-100...100)  to_Repeat  Uninterrupted   Only_priority_interrupted   is_a_Priority_seq  )
Step      = {  Smooth(clocks) Pwm(-100...100) Keep(clocks)  }
SEQ=
[ 6 0 20 3 -100 -100 100 ] 
  ( 50 P )
    {0 -100 0} 
  ( 100 O )
    {0 100 0} 
[ 7 0 50 3 -100 -100 100 ] 
  ( 30 P )
    {0 0 20} 
  ( 80 R O )
    {10 100 10} {10 0 10} {10 100 10} {10 0 10} 

Config parameters are OK
Press ? + Enter to get help about the commands

Version = 0.0.8
Fields in CSV log (+ = included; - = not included)
   + 0 Gps
   + 1  
   + 2 Gps groundspeed (m/s)
   + 3 Gps Heading
   + 4 Gps Altitude(m)
   + 5 Gps Num Sat
   + 6 Gps Date
   + 7 Gps Time
   + 8 Gps PDOP
   + 9 Gps home bearing
   + 10 Gps home dist
   + 11 Volt 1 (V)
   + 12 Current(A)
   + 13 Volt 3(V)
   + 14 Volt 4(V)
   + 15 Capacity (Ah)
   + 16 Temp 1 
   + 17 Temp 2
   + 18 VSpeed (m/s)
   + 19 Baro Altitude (m)
   + 20 Pitch
   + 21 Roll
   + 22 Yaw
   + 23 RPM
   + 24 ADS_1_1
   + 25 ADS_1_2
   + 26 ADS_1_3
   - 27 ADS_1_4
   - 28 ADS_2_1
   - 29 ADS_2_2
   - 30 ADS_2_3
   + 31 ADS_2_4
   + 32 Airspeed (m/s)
   + 33 Comp VSpeed (m/s)
   + 34 Sbus Hold count
   + 35 Sbus Failsafe Count
   + 36 GPS cumul dist (m)
   + 37 Unused 37
   + 38 Unused 38
   + 39 Unused 39
   + 40 Unused 40
   + 41 Rc1
   + 42 Rc2
   + 43 Rc3
   + 44 Rc4
   + 45 Rc5
   + 46 Rc6
   - 47 Rc7
   - 48 Rc8
   - 49 Rc9
   - 50 Rc10
   - 51 Rc11
   - 52 Rc12
   - 53 Rc13
   - 54 Rc14
   - 55 Rc15
   - 56 Rc16
   - 57 Unused 57
   - 58 Unused 58
   - 59 Unused 59
   - 60 Unused 60
   - 61 Unused 61
   - 62 Unused 62
Minimum interval betwwen 2 log entries = 0 ms
Log mode is Continous (log always, MIN/MAX are discarded)
MIN : to log, value of field index 56 must be more than 100
MAX : to log, value of field index 56 must be less than 2500
SD card use SPI1    (0=SPI,  1=SPI1)
GPIO's used by sd card are:
    CS   = 9
    MOSI = 11
    MISO = 12
    SCLK  = 10
UART with oXs uses:
    gpio to receive data from oXs = 5
    Baudrate = 115200
RTC uses:
    SDA = 
0
    SCL = 
1
    Rtc is installed; date&time = 2023-10-18T10:10:01
Protocol is O (oXs + csv)
Led color is normal (not inverted)

Config parameters are OK
Press ? + Enter to get help about the commands

Satcomix, 2023-10-18:
How can i STOP the command PF???

Satcomix, 2023-10-18:
Next results:
FBUS PRI=5 on GPio5 everything is OK, Telemetry Data and SBUS_out OK
S.PORT TLM=1 and PRI=5 everything OK, Telemetry Data and SBUS_out OK
I get since an half hour no **SENSOR LOST** message with all features of oXs RP2040 with Logger port, when i have deactivated **RPM=255**

Satcomix, 2023-10-18:
Ver.0.0.9
Cmd to execute:  DT = 23-10-18T10:55 
Wrong command: command is discarded

Satcomix, 2023-10-18:
Ver.0.0.10

Rebooted by Watchdog!

Config parameters are OK
Press ? + Enter to get help about the commands
 
Type any character to begin +++++++++++

Autoprobing for a RTC...
Autoprobe found MCP7941x at 0x6F
Calibration: 0
log file will be in oXsLog016.csv
   ---- RTC year = 123
   ---- RTC month = 9
   ---- RTC day = 18
No data within 8434

Version = 0.0.10
Fields in CSV log (+ = included; - = not included)
   + 0 Gps
   + 1  
   + 2 Gps groundspeed (m/s)
   + 3 Gps Heading
   + 4 Gps Altitude(m)
   + 5 Gps Num Sat
   + 6 Gps Date
   + 7 Gps Time
   + 8 Gps PDOP
   + 9 Gps home bearing
   + 10 Gps home dist
   + 11 Volt 1 (V)
   + 12 Current(A)
   + 13 Volt 3(V)
   + 14 Volt 4(V)
   + 15 Capacity (Ah)
   + 16 Temp 1 
   + 17 Temp 2
   + 18 VSpeed (m/s)
   + 19 Baro Altitude (m)
   + 20 Pitch
   + 21 Roll
   + 22 Yaw
   + 23 RPM
   + 24 ADS_1_1
   + 25 ADS_1_2
   + 26 ADS_1_3
   - 27 ADS_1_4
   - 28 ADS_2_1
   - 29 ADS_2_2
   - 30 ADS_2_3
   + 31 ADS_2_4
   + 32 Airspeed (m/s)
   + 33 Comp VSpeed (m/s)
   + 34 Sbus Hold count
   + 35 Sbus Failsafe Count
   + 36 GPS cumul dist (m)
   + 37 Unused 37
   + 38 Unused 38
   + 39 Unused 39
   + 40 Unused 40
   + 41 Rc1
   + 42 Rc2
   + 43 Rc3
   + 44 Rc4
   + 45 Rc5
   + 46 Rc6
   - 47 Rc7
   - 48 Rc8
   - 49 Rc9
   - 50 Rc10
   - 51 Rc11
   - 52 Rc12
   - 53 Rc13
   - 54 Rc14
   - 55 Rc15
   - 56 Rc16
   - 57 Unused 57
   - 58 Unused 58
   - 59 Unused 59
   - 60 Unused 60
   - 61 Unused 61
   - 62 Unused 62
Minimum interval betwwen 2 log entries = 0 ms
Log mode is Continous (log always, MIN/MAX are discarded)
MIN : to log, value of field index 56 must be more than 100
MAX : to log, value of field index 56 must be less than 2500
SD card use SPI1    (0=SPI,  1=SPI1)
GPIO's used by sd card are:
    CS   = 9
    MOSI = 11
    MISO = 12
    SCLK  = 10
UART with oXs uses:
    gpio to receive data from oXs = 5
    Baudrate = 115200
RTC uses:
    SDA = 
0
    SCL = 
1
    Rtc is installed; date&time = 2023-10-18T11:03:09
Protocol is O (oXs + csv)
Led color is normal (not inverted)

Config parameters are OK
Press ? + Enter to get help about the commands
 
Cmd to execute:  DT = 23-10-18T11:04 
Wrong command: command is discarded

 >> 

Version = 0.0.10
Fields in CSV log (+ = included; - = not included)
   + 0 Gps
   + 1  
   + 2 Gps groundspeed (m/s)
   + 3 Gps Heading
   + 4 Gps Altitude(m)
   + 5 Gps Num Sat
   + 6 Gps Date
   + 7 Gps Time
   + 8 Gps PDOP
   + 9 Gps home bearing
   + 10 Gps home dist
   + 11 Volt 1 (V)
   + 12 Current(A)
   + 13 Volt 3(V)
   + 14 Volt 4(V)
   + 15 Capacity (Ah)
   + 16 Temp 1 
   + 17 Temp 2
   + 18 VSpeed (m/s)
   + 19 Baro Altitude (m)
   + 20 Pitch
   + 21 Roll
   + 22 Yaw
   + 23 RPM
   + 24 ADS_1_1
   + 25 ADS_1_2
   + 26 ADS_1_3
   - 27 ADS_1_4
   - 28 ADS_2_1
   - 29 ADS_2_2
   - 30 ADS_2_3
   + 31 ADS_2_4
   + 32 Airspeed (m/s)
   + 33 Comp VSpeed (m/s)
   + 34 Sbus Hold count
   + 35 Sbus Failsafe Count
   + 36 GPS cumul dist (m)
   + 37 Unused 37
   + 38 Unused 38
   + 39 Unused 39
   + 40 Unused 40
   + 41 Rc1
   + 42 Rc2
   + 43 Rc3
   + 44 Rc4
   + 45 Rc5
   + 46 Rc6
   - 47 Rc7
   - 48 Rc8
   - 49 Rc9
   - 50 Rc10
   - 51 Rc11
   - 52 Rc12
   - 53 Rc13
   - 54 Rc14
   - 55 Rc15
   - 56 Rc16
   - 57 Unused 57
   - 58 Unused 58
   - 59 Unused 59
   - 60 Unused 60
   - 61 Unused 61
   - 62 Unused 62
Minimum interval betwwen 2 log entries = 0 ms
Log mode is Continous (log always, MIN/MAX are discarded)
MIN : to log, value of field index 56 must be more than 100
MAX : to log, value of field index 56 must be less than 2500
SD card use SPI1    (0=SPI,  1=SPI1)
GPIO's used by sd card are:
    CS   = 9
    MOSI = 11
    MISO = 12
    SCLK  = 10
UART with oXs uses:
    gpio to receive data from oXs = 5
    Baudrate = 115200
RTC uses:
    SDA = 
0
    SCL = 
1
    Rtc is installed; date&time = 2023-10-18T11:04:03
Protocol is O (oXs + csv)
Led color is normal (not inverted)

Config parameters are OK
Press ? + Enter to get help about the commands

mstrens, 2023-10-18:
I uploaded 0.0.11
I think that DT = YYYY... was wel processed but the error msg was still given (by mistake).
I also fixed the YYYY and MM (add 1900 and 1).

So I hope logger is ok.
I still could add a check in order to discard RTC for updating creation timestamp of log file when RTC YYYY is less than 2023.
This would still give the opportunity to update the timestamp when there is a GPS and no RTC backup battery. 

For the issue about oXs RPM, I still have to analyse. But I will be out for some hours this PM

Satcomix, 2023-10-18:
Ver.0.0.11 works !!!

Rebooted by Watchdog!

Config parameters are OK
Press ? + Enter to get help about the commands
 
Type any character to begin +++++++++++

Autoprobing for a RTC...
Autoprobe found MCP7941x at 0x6F
Calibration: 0
log file will be in oXsLog017.csv
   ---- RTC year = 2023
   ---- RTC month = 10
   ---- RTC day = 18
end of setup
No data within 37124

Version = 0.0.11
Fields in CSV log (+ = included; - = not included)
   + 0 Gps
   + 1  
   + 2 Gps groundspeed (m/s)
   + 3 Gps Heading
   + 4 Gps Altitude(m)
   + 5 Gps Num Sat
   + 6 Gps Date
   + 7 Gps Time
   + 8 Gps PDOP
   + 9 Gps home bearing
   + 10 Gps home dist
   + 11 Volt 1 (V)
   + 12 Current(A)
   + 13 Volt 3(V)
   + 14 Volt 4(V)
   + 15 Capacity (Ah)
   + 16 Temp 1 
   + 17 Temp 2
   + 18 VSpeed (m/s)
   + 19 Baro Altitude (m)
   + 20 Pitch
   + 21 Roll
   + 22 Yaw
   + 23 RPM
   + 24 ADS_1_1
   + 25 ADS_1_2
   + 26 ADS_1_3
   - 27 ADS_1_4
   - 28 ADS_2_1
   - 29 ADS_2_2
   - 30 ADS_2_3
   + 31 ADS_2_4
   + 32 Airspeed (m/s)
   + 33 Comp VSpeed (m/s)
   + 34 Sbus Hold count
   + 35 Sbus Failsafe Count
   + 36 GPS cumul dist (m)
   + 37 Unused 37
   + 38 Unused 38
   + 39 Unused 39
   + 40 Unused 40
   + 41 Rc1
   + 42 Rc2
   + 43 Rc3
   + 44 Rc4
   + 45 Rc5
   + 46 Rc6
   - 47 Rc7
   - 48 Rc8
   - 49 Rc9
   - 50 Rc10
   - 51 Rc11
   - 52 Rc12
   - 53 Rc13
   - 54 Rc14
   - 55 Rc15
   - 56 Rc16
   - 57 Unused 57
   - 58 Unused 58
   - 59 Unused 59
   - 60 Unused 60
   - 61 Unused 61
   - 62 Unused 62
Minimum interval betwwen 2 log entries = 0 ms
Log mode is Continous (log always, MIN/MAX are discarded)
MIN : to log, value of field index 56 must be more than 100
MAX : to log, value of field index 56 must be less than 2500
SD card use SPI1    (0=SPI,  1=SPI1)
GPIO's used by sd card are:
    CS   = 9
    MOSI = 11
    MISO = 12
    SCLK  = 10
UART with oXs uses:
    gpio to receive data from oXs = 5
    Baudrate = 115200
RTC uses:
    SDA = 
0
    SCL = 
1
    Rtc is installed; date&time = 2023-10-18T12:28:47
Protocol is O (oXs + csv)
Led color is normal (not inverted)

Config parameters are OK
Press ? + Enter to get help about the commands
 
Cmd to execute:  DT = 23-10-18T12:30 
first char = 32

 >>

closed 2023-10-18 by Satcomix