mstrens/oXs_on_RP2040 #114: RPM and Log GPio = Sensor Lost

Satcomix, 2023-10-19:
Hello Mstrens,
I continued with the tests this morning,
I discovered the following:
Ver.2.8.33 FBUS or S.PORT with SBUS Protocol,
Sensor lost messages set to 5 second delay.
LOG=255 everything else activated,
All 41 telemetry data on the handset are plausible.
RPM is displayed as 988-1002U/min at handset and in terminal, with 1KHz input, ok.
Everything works as usual. 
No Sensor Lost since 30min
Then I enter LOG=0, the board reboots, then only shows data without the I2C bus.
Power down and up the board.
I2C and everything else active again, but sensor lost every 5 seconds from the RPM sensor, in the meantime it only shows 0rpm on the handset and only -1Hz in the terminal with FV Command.
-1Hz is of course recognized as Sensor Lost because there can only be positive values.
I will make some more tests.
Greetings,
Torsten

mstrens, 2023-10-19:
I think I understood the issue.
RP2040 has what they call "PIO" and "SM" that can be use to add e.g. UART interfaces.
There are max 8 pio/sm available.
RPM uses one pio/sm (to count the number of pulses).
I forgot to update in my internal doc about the pio/sm used for RPM and so I reused the same pio/sm for the logger.
The problem is that all pio/sm are already used.
So I have to see if I can calculate RPM without a pio/sm or if I can free one other pio/sm

Satcomix, 2023-10-19:
Hello Mstrens,
That sounds good now that we know what the problem is.
You'll find a solution. Good luck.
greetings,
Torsten

mstrens, 2023-10-19:
I put oXs version 2.8.34.
I used 2 pio/sm for gps (one to send the gps config commands and one to get GPS data)
I now changed the code in order to reuse the same pio/sm to get GPS data after it has been configured.
So I got one more pio/sm free to use for RPM.
I hope it works.
Has to be tested with GPS+RPM+logger activated.

Satcomix, 2023-10-19:
Ver.2.8.34 
The RP2040 board reboots all the time, no proper communication via terminal is possible.

mstrens, 2023-10-19:
I will try to reproduce this issue.

Satcomix, 2023-10-19:
Do I have to NUKE Ver.2.8.34 beforehand and then install the new version because of the state machines?

Satcomix, 2023-10-19:
I have NUKE the board and reconfig it, but nothing change with Ver.2.8.34 :-((

mstrens, 2023-10-19:
It is strange.
I had some automatic reboot too but not always.
I activate in 2.8.35 the option DEBUG in order to delay the power on and so get the first debug message and now I do not have the reboot anymore.
I do not understand where is the bug because I did not change something else.

Satcomix, 2023-10-19:
What I noticed particularly in the last versions is that every time I enter a new value, the board logs in and out and I have to restart the Arduino IDE terminal each time.

mstrens, 2023-10-19:
I put 2.8.36.
I hope it works.
On my side I do not have automatic reboot.

ps: I will be out for a few hours.

Satcomix, 2023-10-19:
Hello Mstrens,
Version 2.8.36 works with FBUS or S.PORT and SBUS with all 33 telemetry values, sequencer, ​​and logger connected. 
Thank you for the effort.
If I need to test something special, I wouldn't be able to do it again until tomorrow
greetings,
Torsten

closed 2023-10-19 by Satcomix

Satcomix, 2023-10-30:
I'll continue writing here.
I ran the next tests.
Version 2.8.0-main works, even as a separate build with changes such as camera stabilizer.
After that I checked version 2.8.19. REBOOT error occurs even after NUKE before with your own build with SDK1.5.1.
Then checked version 2.8.22. Same problem.
I will now check the versions before 2.8.19.
I assume it has to do with the ESC function since I didn't test it during this development period because I was sick for a long time.

mstrens, 2023-10-30:
If I understand well you said that when you use the uf2 file that I provided with version 2.8.19, then you get reboot errors.

If it is so, I should also be able to reproduce the error.
Up to know, I could not reproduce it but as I say I am using a RP2040 that is not connected to a receiver nor to sensors.

That is the reason why I asked if you could try disconnecting the RP2040 from the receiver and the GPS (but keeping the GPS pins defined as if a GPS was connected).

Satcomix, 2023-10-30:
Next test next build.
Test with own 2.8.0 and camera stabilizer, all sensors ok, no reboot.
Next test with version 2.8.8 copied to the board with all settings active, reboot again and again.
Then NUKE,
PROTOCOL=F
PRI=5
NO REBOOT
GPS_RX=12
REBOOT everytime.
Since I don't have the versions before 2.8.8 saved on my PC, I first have to check GH to see how I can download them.

Satcomix, 2023-10-30:
"That is the reason why I asked if you could try disconnecting the RP2040 from the receiver and the GPS (but keeping the GPS pins defined as if a GPS was connected)."
**WITH or WITHOUT connected GPS cable everytime reboot.**

Satcomix, 2023-10-30:
What is the difference between 2.8.0 and 2.8.8?
In Terminal you see LOG=255 and ESC=255.
We know that activating LOG and RPM in this version still causes errors due to the SM being used twice.
But RPM=255, LOG=255 and ESC=255 should not interfere with the GPS UART. The problems only came when it was activated.
Has anything changed in the CMake.list?
Because it works 100% with my own builds without LOG=255 and ESC=255 up to version 2.8.0.

Satcomix, 2023-10-30:
If I use the original .uf2 the error doesn't appear. I have already mentioned it several times.
Only when activating LOG and RPM, but that is normal in this version 2.8.8
It's about creating my own build with Camera Stabilizer enabled, or other changes.
But only from versions after 2.8.0, up to version 2.8.0, do I have any problems creating my own working builds with Pico SDK1.5.1

Satcomix, 2023-10-30:
Here is my Version of 2.8.8, i renamed it in 2.8.8.1 with acktivateted stabilizer.
Everytime reobot after 10sec.
You can have a look at my build in .zip

[oXs_on_RP2040-test.zip](https://github.com/mstrens/oXs_on_RP2040/files/13203540/oXs_on_RP2040-test.zip)

processing cmd

Version = 2.8.8.1 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =    8    9  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255   10   11
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)
Logger  . . . . . . . . . =  255  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =  255  (ESC_PIN= 0, 1, 2, ..., 29)

Esc type is not defined

Protocol is Fbus(Frsky)
CRSF baudrate   = 420000
Logger baudrate = 115200
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 0.100000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 2.000000 , 0.000000 
    One temperature sensor is connected on V3
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 954 , 21 , -2823
     Gyro offsets         X, Y, Z = 25 , -163 , -193
Aispeed sensor is detected using MS4525
    Vspeed compensation channel = 5
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Foreseen GPS type is Ublox (configured by oXs) :GPS is not (yet) detected
Led color is normal (not inverted)
Failsafe uses predefined values
     Chan 1...4  =  1500  1500  1500  1500
     Chan 5...8  =  2012  2012  1500  1500
     Chan 9...12 =  1500  1500  1500  1500
     Chan 13...16=  1500  1500  1500  1500

Number of sequencers= 1   Number of steps= 2
     { Gpio  Type(0=servo,1=analog) Clock(msec) Channel Default Min Max}...{ }
SEQ= {7 0 20 3 -100 -100 100} 
    

      { Range(-100/-75/-50/-25/0/25/50/75/100) Smooth(clocks) Pwm(-100/100) Keep(clocks) } ...{}
STEP= -  {50 0 100 255} 
         {100 0 -100 255} 

Config parameters are OK
Press ? + Enter to get help about the commands
assertion "hw_is_claimed(bits, bit_index)" failed: file "C:/Program Files/Raspberry Pi/Pico SDK v1.5.1/pico-sdk/src/rp2_common/hardware_claim/claim.c", line 51, function: hw_claim_clear

Satcomix, 2023-10-30:
Here is my version 2.8.0 , i renamed it into 2.8.0.1 with activated stabilizer, no reboot, no problems.

[oXs_on_RP2040-main.zip](https://github.com/mstrens/oXs_on_RP2040/files/13203571/oXs_on_RP2040-main.zip)

mstrens, 2023-10-30:
I uploaded your uf2 for 2.8.8.1 and I get the reboot.
I will now try to compile your files and compile myself.
If I still have the reboot, I will look further with debug messages in your version 2.8.8.1.

Satcomix, 2023-10-30:
Thank you very much.
Now we are both on the same page.

mstrens, 2023-10-30:
I compiled and upload my compilation and it continue to reboot.
So I can now further continue to look at the issue.

Satcomix, 2023-10-30:
The strange thing is,
Nothing happens with the original version, the errors only appear when a version is changed and a new build is created, but only after version 2.8.0

mstrens, 2023-10-30:
I found the instruction that block the rp2040.
It is when I ask to remove the pio/sm used for sending the config to the GPS.
I use a function provided by the SDK and it stops the processor.
I will read again the sdk doc about this function.

Still I do not understand why this did not occurred when I used my previous compilations.

Satcomix, 2023-10-30:
Maybe it has something to do with the fact that I've been using several Pimoroni Servo2040 for all the tests and functions of oXs_on_RP2040 for a year now, even though the GPio wiring should be the same. I think you use a Waveshare RP2040Zero.
I will make some tests with the Zero. I still have some here.

mstrens, 2023-10-30:
In gps.cpp file you have one line with
pio_sm_unclaim(gpsPio, gpsSmTx);                           // free the SM used for GPS TX

Can you try to replace it with
pio_sm_set_enabled(gpsPio, gpsSmTx, false);

It seems I do not have the reboot anymore.
Probably I can't use this sdk function because I did not use previously the claim function.

Satcomix, 2023-10-30:
Hello Mstrens,
By changing the gps.cpp and activating the camera stabilization everything works without a REBOOT. Great, thank you.
But in the other scripts of the gps.cpp I don't find "pio_sm_unclaim(gpsPio, gpsSmTx); // free the SM used for GPS TX ".
s it a different line, since a lot has been changed in the newer versions (2.9.8).
Greetings,
Torsten

mstrens, 2023-10-30:
I do not remember when I added this (it is related to the reuse of the PIO/SM from sending  for receiving.
Fine that it is now OK.
I still do not understand why I could not reproduce the error on my side prior installing your zip file.
Anyway, what matter is that it works now. 
I will include in the next version (I am currently adding the usb command to manage the gyro config)

Satcomix, 2023-10-30:
I make the next tests till 2.8.40-test and 2.9.0-main
When i change gps.cpp and activate camera stabilistor, everything is OK without any reboot.
Now i will test the next releases up to 2.9.8

Satcomix, 2023-10-30:
Next tests up to 2.9.8 .
With changes in gps.cpp everything works and is ok, without the camera stabilizer, thats of course from the new gyro project.
Thanks again. :-))
Greetings,
Torsten