mstrens/oXs_on_RP2040 #116: OXS with ESC

Klippeneck, 2023-11-08:
Hello Mstrens,
Firstly, many thanks for your great work.
I have already set up 3 sensors with Vario and GPS, they all work perfectly. Now I am trying to integrate a Hobbywing Fly Fun V5 ESC into a vario.
Unfortunately I can't get any telemetry data from the ESC on the transmitter.
The sensor only registers with vario and altitude.
Is the attached configuration correct? I use Jeti. 
Version = 2.9.18 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =    2  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =    3  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255  255  255 (V1 / V4 = 26, 27, 28, 29)
RGB led . . . . . . . . . =   16  (RGB    = 0, 1, 2, ..., 29)
Logger  . . . . . . . . . =  255  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =    1  (ESC_PIN= 0, 1, 2, ..., 29)  

Esc type is HW4 (Hobbywing V4)

Protocol is Jeti (Exbus)
CRSF baudrate   = 420000
Logger baudrate = 115200
PWM is generated at = 50 Hz
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is not detected
Airspeed sensor is not detected
    No Vspeed compensation channel defined; oXs uses default settings
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox (configured by oXs) :GPS is not (yet) detected
Led color is normal (not inverted)
Failsafe type is HOLD

Gyro is not configured

No sequencers are defined

Config parameters are OK
Press ? + Enter to get help about the commands

Satcomix, 2023-11-08:
Hello Klippeneck,
According to the description, Mstrens has implemented the Hobbywing V4 telemetry protocol in the oXs_RP2040, not the V5.
_- an ESC from Hobbywing (using V4 telemetry protocol), ZTW Mantis or from Kontronik. Those ESC provide one voltage, one current (+ current consumption) + RPM + 2 temperatures._
There are also some tips on how to use the V5 on the MSRC page:
(1) HW V5 = HW FlyFun. If ESC doesn't arm, enable ESC Init Delay or connect MSRC after the ESC is armed
https://github.com/dgatf/msrc
Greetings,
Torsten

Klippeneck, 2023-11-08:
Hello Thorsten,
if I interpret the documents from dgat correctly,
V4 and V5 should use the same protocol.
My controller also offers voltage, current and 2 temperatures and capacity.
What makes me suspicious is that the controller does not appear at all.
Greetings Jürgen

Satcomix, 2023-11-08:
Hello Mstrens,
m0p3d confirmed at 29.09.23 in "Support Hobbywing ESC telemetry"
Version 2.8.8 Confirmed works for -RPM, TEMP1,2 ,Current
Strange VFAS=2,58V ADC2=0V
Greetings,
Torsten

mstrens, 2023-11-08:
Indeed.
I forgot that it had been tested and that it works.
Anyway with latest version, it should be possible to check if oXs get frames from ESC and if the content of HW5 frames  is indeed similar to frames from HW4

Klippeneck, 2023-11-08:
Hello Mstrens,
Test with SW 2.9.22 gives the following result:
Greetings Jürgen
processing cmd

Cmd to execute: 

Version = 2.9.22 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =    2  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =    3  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255  255  255 (V1 / V4 = 26, 27, 28, 29)
RGB led . . . . . . . . . =   16  (RGB    = 0, 1, 2, ..., 29)
Logger  . . . . . . . . . =  255  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =    1  (ESC_PIN= 0, 1, 2, ..., 29)    Esc type is HW4 (Hobbywing V4)

Protocol is Jeti (Exbus)
CRSF baudrate   = 420000
Logger baudrate = 115200
PWM is generated at = 50 Hz
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is not detected
Airspeed sensor is not detected
    No Vspeed compensation channel defined; oXs uses default settings
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox (configured by oXs) :GPS is not (yet) detected
Led color is normal (not inverted)
Failsafe type is HOLD

Gyro is not configured

No sequencers are defined

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
 0
I think the V4 protocol and V5 are not the same but according to dgat it should work.

mstrens, 2023-11-08:
Thanks.
Having the lines with 0 shows that the ESC sent some characters.
I expect I made a small mistake in the code I added for the message that explains why it is always 0.
I put version 2.9.23 on github.
Please make the same test with this new version (I hope to see other codes than 0).

Klippeneck, 2023-11-08:
Hello mstrens,
Test with 2.9.23, result slightly different.
Greetings Jürgen
processing cmd

Cmd to execute: 

Version = 2.9.23 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =    2  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =    3  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255  255  255 (V1 / V4 = 26, 27, 28, 29)
RGB led . . . . . . . . . =   16  (RGB    = 0, 1, 2, ..., 29)
Logger  . . . . . . . . . =  255  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =    1  (ESC_PIN= 0, 1, 2, ..., 29)    Esc type is HW4 (Hobbywing V4)

Protocol is Jeti (Exbus)
CRSF baudrate   = 420000
Logger baudrate = 115200
PWM is generated at = 50 Hz
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is not detected
Airspeed sensor is not detected
    No Vspeed compensation channel defined; oXs uses default settings
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox (configured by oXs) :GPS is not (yet) detected
Led color is normal (not inverted)
Failsafe type is HOLD

Gyro is not configured

No sequencers are defined

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
8000
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0

mstrens, 2023-11-08:
It is very strange.
The first 8000 says that the first char received after some delay without signal is 0.
And then all others char received are 0.

For sure this is not a valid frame.

Do you have a logical analyser to capture the signal provided by the ESC (and so be sure there is no bug in the way oXs read the frame)?

Klippeneck, 2023-11-08:
Hello mstrens,
thank you for your efforts, unfortunately I don't have a logic analyser available,
before I retired this would have been possible.

mstrens, 2023-11-08:
I will try to make a test where I use another RP2040 to generate dummy data (similar to an ESC HW4) and I will apply this signal to oXs. So I can check if the "0" are the result of an oXs bug or are really sent by the ESC.

I will try to do it to morrow.

Satcomix, 2023-11-08:
Hello Mstrens, 
For my future oXs tests, an ESC RP2040 dummy would be a very good solution, since I don't have a HobbyWing V4/V5 ESC either. If it is possible, could I get a v4_v5_dummy_esc.uf2 from you?
Thank you,
Greetings, 
Torsten

mstrens, 2023-11-09:
I had a look at this manual
https://www.towerhobbies.com/on/demandware.static/-/Sites-horizon-master/default/dwaee8da49/Manuals/HWI30214002-Manual-EN.pdf

I expect it is the manual of your ESC.
I do not find any reference saying that it can provide telemetry data.
To work with oXs, it has to provide on one wire telemetry data via an UART signal at 19200 bauds 8N1.

It is not clear for me which HW models can provide such a signal.

mstrens, 2023-11-09:
I also found this doc
https://cdn.shopify.com/s/files/1/0109/9702/files/HW-FOC_Communication_Protocol-20200326.pdf?v=1616086642
It says that when motor is running the data are provide every 50ms.
Perhaps there is no data when the motor does not run(?)

Satcomix, 2023-11-09:
Hello everyone,
Normally you can say that the ESCs from HobbyWing, which are programmed with the old LED box, also have no possibility of transmitting telemetry data.
Those programmed with the new display box with USB connection could also send telemetry.
Unfortunately I only have ESCs (also new models) for the LED box, that`s HobbyWing.
Greetings,
Torsten

Klippeneck, 2023-11-09:
Hello everyone,
the Hobbywing V5/80A definitely has telemetry, I made a test with the Arduino based converter from dgat.
However, the controller only sends data after a successful initialisation, recognisable by the acoustic signals.
For this, the ESC needs a delay of 2-3sec in the telemetry converter after switching on, otherwise only zeros are sent. 
Greetings Jürgen

Satcomix, 2023-11-09:
Hence dgatf's reference to the MSRC:
(1) HW V5 = HW FlyFun. If ESC doesn't arm, **enable ESC Init Delay or connect MSRC after the ESC is armed**
https://github.com/dgatf/msrc

mstrens, 2023-11-09:
I put a version 2.9.25 on github.
I had no time to test it at all.
It is supposed to print the frames received from the ESC after that the ESC starts to send something else than "0".
so you should not have many 0 at the power on.

With this version, all parameters already defined with USB commands are lost because I also added some parameters for the gyro..
Once edited with usb commands, the parameters will be kept (up to a new "major" change in config).

Mike-HRC, 2023-11-09:
Mstrens - I used a HW40 with som eof the early work that Daniel did on MSRC, from memory that HW80 was a trciky ESC (I can't remember why - I'll try and find the info) if you look here https://github.com/dgatf/msrc Daniel explains the HW Telemetry structure for the ESCs towards the bottom of the page.

Also some of the HW FunFly ESCs have the telemetry as an undocumented feature (presumably so that you buy the more expensive Platinum range - I have a FunFly 30a that provides Teletry even though all of the information on the HW web-site says that it does not.

Satcomix, 2023-11-09:
Hello Mike,
_"Also some of the HW FunFly ESCs have the telemetry as an undocumented feature"._
Thank you for this information. Now I'll check my HobbyWing ESCs for telemetry data.
Greetings,
Torsten

mstrens, 2023-11-09:
I wrote the code for ESC HW V4 reusing the work of Daniel (on MSRC).
One user reports that it works with a HW V4 ESC.
The issue is that now another user says it does not work with a HW V5.
I made some changes to try to debug the issue (printing the characters received from the ESC).
The latest oXs version should provide those characters after that the ESC sent all "0"

Satcomix, 2023-11-09:
Hello Mstrens,
I did a lot of research about the MSRC project today.
Most of the time you hear that the old MSRC works on Arduino and the new one has never worked properly.
I searched for HobbyWing realtime data, but Daniel's explanations match HobbyWing's, which can be found at the bottom of his page, such as every 20ms....
I will have a HW Platinum Pro V4 60A with LCD box for testing next week.
Greetings,
Torsten

mstrens, 2023-11-09:
I think this link provide also a good explanation

https://cdn.shopify.com/s/files/1/0109/9702/files/HW-FOC_Communication_Protocol-20200326.pdf?v=1616086642

I wait to see if the HW5 send frames and if the format is the same as HW4.
The current version of oXs should allow this check (because I skip the 0 at power on.

Satcomix, 2023-11-09:
I had already saved this .pdf this morning. At HobbyWing you can also find some things for V3 well explained in the instructions: "For developers only"
V3 differs in the number of bytes =10 for the values ​​and V4/V5 have 19 bytes.
Also the explanations for:
Before throttle is raised from 0, signature packets are sent between telemetry packets. This is used to identify the hardware and firmware of the ESC,
is very helpful.
Next week i can make tests and logs with my LogicAnalyzer

Satcomix, 2023-11-11:
First Test with HobbyWing Platinum V4 60A and oXs Ver.2.9.26
At Handset i can see VFAS, FASS, Capacity, TEMP1, RPM
FASS=0 and Capacity=0 because the V4 60A has no Current telemetry.

processing cmd

Version = 2.9.26 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255  255   29 (V1 / V4 = 26, 27, 28, 29)
RGB led . . . . . . . . . =   18  (RGB    = 0, 1, 2, ..., 29)
Logger  . . . . . . . . . =  255  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =    1  (ESC_PIN= 0, 1, 2, ..., 29)    Esc type is HW4 (Hobbywing V4)

Protocol is Fbus(Frsky)
CRSF baudrate   = 420000
Logger baudrate = 115200
PWM is generated at = 50 Hz
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 0.100000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 2.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 1319 , 263 , -2756
     Gyro offsets         X, Y, Z = 17 , -162 , -192
     Orientation                  = 0
Aispeed sensor is detected using MS4525
    Vspeed compensation channel = 7
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Foreseen GPS type is Ublox (configured by oXs) :GPS is detected and has a fix
Led color is normal (not inverted)
Failsafe uses predefined values
     Chan 1...4  =  1500  1500  1500  1500
     Chan 5...8  =  2012  2012  1500  1500
     Chan 9...12 =  1500  1500  1500  1500
     Chan 13...16=  1500  1500  1500  1500

Gyro configuration is:
Channels for :  mode/gain=5  ,  Ail stick=1  ,  Elv stick=2  ,  Rud stick=4
Gain per axis (-128/127):  Roll=127     Pitch=127    Yaw=127
Gain on throw is 1 (1=on full throw, 2=on half, 3=on quater)
Max rotate is 3 (1=Very low , 2=low , 3=medium , 4=high)
Stick rotate enabled in rate mode is 1 (1=disabled , 2=enabled)
Stabilize mode is OFF (Hold mode is enabled)
PID              Roll(aileron)           Pitch(elevator)         Yaw(rudder)
   Mode         Kp      Ki      Kd          Kp      Ki      Kd         Kp      Ki      Kd
  Normal  PIDN= 1000    0       500         1000    0       500        1000    0       500  
  Hold    PIDH= 500     500     500         500     500     500        500     500     500  
  Stab.   PIDS= 500     0       500         500     0       500        500     0       500  

Gyro mixers are calibrated:
Sticks centered at:    Ail=0%   Elv=0%   Rud=0%
Gyro corrections (from center pos in %) on:

Number of: sequencers=2   sequences=4   steps= 6
Sequencer = [  Gpio  Type(0=servo,1=analog) Clock(msec) ChannelNr Default Min Max ]
Sequence  = (  RC_value(-100...100)  to_Repeat  Uninterrupted   Only_priority_interrupted   is_a_Priority_seq  )
Step      = {  Smooth(clocks) Pwm(-100...100) Keep(clocks)  }
SEQ=
[ 6 0 20 9 -100 -100 100 ] 
  ( 50 P )
    {0 -100 20} 
  ( 100 O )
    {0 100 20} {0 -100 20} 
[ 7 0 20 9 -100 -100 100 ] 
  ( 50 P )
    {0 -100 20} 
  ( 100 O )
    {0 100 20} {0 -100 20} 

Config parameters are OK
Press ? + Enter to get help about the commands
80FF
80FF
8000
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0
809C
   0
  3F
   0
  45
   0
  3F
   0
   0
   5
  78
8030
  31
  30
  34
  30
  38
  34
  35
809B
  9B
   3
  E8
   1
   8
  5B
   0
   1
   0
  21
  21
  B9
809B
   0
   0
   1
   0
   0
   0
  14
   0
   0
   0
   5
  73
   0
   0
   D
  68
   D
  96
Esc Volt=1395   current=0  consumed=0  temp1=3432  temp2=3478
809B
  9B
   3
  E8
   1
   8
  5B
   0
   1
   0
  21
  21
  B9
809B
   0
   0
   2
   0
   0
   0
  14
   0
   0
   0
   5
  6D
   0
   0
   D
  64
   D
  95
Esc Volt=1389   current=0  consumed=0  temp1=3428  temp2=3477
809B
  9B
   3
  E8
   1
   8
  5B
   0
   1
   0
  21
  21
  B9
809B
   0
   0
   3
   0
   0
   0
  14
   0
   0
   0
   5
  70
   0
   0
   D
  65
   D
  9A
Esc Volt=1392   current=0  consumed=0  temp1=3429  temp2=3482
809B
  9B
   3
  E8
   1
   8
  5B

Satcomix, 2023-11-11:
After ten seconds, the ESC connection hangs and the telemetry values ​​for the ESC are no longer transmitted and the ESC debug output on Terminal stops. All other values ​​ok and transmitted.
Maybe it has something to do with the fact that I don't have a brushless motor connected to the ESC, it won't be delivered until next week.

Satcomix, 2023-11-16:
Hello Mstrens,
Second Test with X20S TD-R10 and FBUS Protocol.
oXs_RP2040 Version: 2.9.26
**ESC:**
Hobbywing Platinum V4 60A
**Motor:** 
SPMXAM1100
**Telemetry Handset:**
VFAS: ok
FASS: ok
Capacity: ok
TEMP1: ok
TEMP2: ok
RPM: ok
**With the Hobbywing V4 protocol everything seems to work for me.**

processing cmd

Version = 2.9.26 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255    8  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255    9  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255   28   29 (V1 / V4 = 26, 27, 28, 29)
RGB led . . . . . . . . . =   18  (RGB    = 0, 1, 2, ..., 29)
Logger  . . . . . . . . . =    0  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =    1  (ESC_PIN= 0, 1, 2, ..., 29)    Esc type is HW4 (Hobbywing V4)

Protocol is Fbus(Frsky)
CRSF baudrate   = 420000
Logger baudrate = 115200
PWM is generated at = 50 Hz
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 0.100000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 2.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 1319 , 263 , -2756
     Gyro offsets         X, Y, Z = 17 , -162 , -192
     Orientation                  = 0
Aispeed sensor is detected using MS4525
    Vspeed compensation channel = 7
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Foreseen GPS type is Ublox (configured by oXs) :GPS is detected and has a fix
Led color is normal (not inverted)
Failsafe uses predefined values
     Chan 1...4  =  1500  1500  1500  1500
     Chan 5...8  =  2012  2012  1500  1500
     Chan 9...12 =  1500  1500  1500  1500
     Chan 13...16=  1500  1500  1500  1500

Gyro is not configured

Number of: sequencers=2   sequences=4   steps= 6
Sequencer = [  Gpio  Type(0=servo,1=analog) Clock(msec) ChannelNr Default Min Max ]
Sequence  = (  RC_value(-100...100)  to_Repeat  Uninterrupted   Only_priority_interrupted   is_a_Priority_seq  )
Step      = {  Smooth(clocks) Pwm(-100...100) Keep(clocks)  }
SEQ=
[ 6 0 20 9 -100 -100 100 ] 
  ( 50 P )
    {0 -100 20} 
  ( 100 O )
    {0 100 20} {0 -100 20} 
[ 7 0 20 9 -100 -100 100 ] 
  ( 50 P )
    {0 -100 20} 
  ( 100 O )
    {0 100 20} {0 -100 20} 

Config parameters are OK
Press ? + Enter to get help about the commands
8000
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0
   0
809C
   0
  41
   0
  41
   0
  42
   0
   0
   5
  83
8030
  31
  30
  34
  30
  38
  34
  35
809B
  9B
   3
  E8
   1
   8
  5B
   0
   1
   0
  21
  21
  B9
809B
   0
   0
   1
   0
   0
   0
  14
   0
   0
   0
   5
  80
   0
   0
   C
  44
   C
  78
Esc Volt=1408   current=0  consumed=0  temp1=3140  temp2=3192
809B
  9B
   3
  E8
   1
   8
  5B
   0
   1
   0
  21
  21
  B9
809B
   0
   0
   2
   0
   0
   0
  14
   0
   0
   0
   5
  83
   0
   0
   C
  46
   C
  77
Esc Volt=1411   current=0  consumed=0  temp1=3142  temp2=3191
809B

mstrens, 2023-11-16:
Thanks.
FYI, it seems that the ESC sent several types of frame (begin with 0X9B)
I use the frame that contains 19 bytes.

Satcomix, 2023-11-16:
Different frames could be related to the function of the port.
Hello Mstrens,
Different frames could be related to the function of the port.
The telemetry connection of the Platinum V4 60A is also the configuration connection of the USB LCD programming box.
If Jürgen (Klippeneck) would now confirm the functions of the Hobbywing V5 protocol with all telemetry values, the ESC debug messages could be removed from the next oXs_RP2040-test version.
Greetings,
Torsten

Satcomix, 2023-11-16:
Hello Mstrens:
FYI:
I just looked at the transferred frames again.
Everything is exactly as Daniel described on his MSRC page.
1.Frame with 19bytes
2.Frame with 13bytes
Before throttle is raised from 0, signature packets are sent between telemetry packets. This is used to identify the hardware and firmware of the ESC.
**Telemetry packets:**
 Byte  | 1                              | 2                             | 3                           | 4                            | 5                  | 6                    |
Value | Package Head (0x9B) | Package Number 1 | Package Number 2 | Package Number 3 | Rx Throttle 1 | Rx Throttle  2 |
           7             | 8                       | 9         | 10       | 11       | 12            | 13           | 14            | 15            | 16             | 
 Output PWM 1 | Output PWM 2 | RPM 1 | RPM 2 | RPM 3 | Voltage 1 | Voltage 2 | Current 1 | Current 2 | TempFET 1 | 
         17      | 18        |    19
 TempFET 2 | Temp 1 | Temp 2
**Signature packets:**
                  Byte | 1       | 2      | 3      | 4       | 5      | 6       | 7      | 8       | 9      | 10     | 11    | 12     | 13
V4LV25/60/80A | 0x9B | 0x9B | 0x03 | 0xE8 | 0x01 | 0x08 | 0x5B | 0x00 | 0x01 | 0x00 | 0x21 | 0x21 | 0xB9

**HobbyWing Platinum V4 60A:**
Byte Frame1  Frame2
1       809B    809B
2       0          9B
3       0           3
4       2         E8
5       0           1
6       0           8
7       0         5B
8     14           0
9       0           1
10     0           0
11     0         21
12     5         21
13   83         B9
14     0
15     0
16     C
17   46
18    C
19   77
Esc Volt=1411   current=0  consumed=0  temp1=3142  temp2=3191

Klippeneck, 2023-11-16:
Hello Mstrens, Satcomics,
I have tested with ESC Hobbywing V5/80A.
The result is as follows
:processing cmd

Cmd to execute: 

Version = 2.9.26 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =    2  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =    3  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255  255  255 (V1 / V4 = 26, 27, 28, 29)
RGB led . . . . . . . . . =   16  (RGB    = 0, 1, 2, ..., 29)
Logger  . . . . . . . . . =  255  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =    1  (ESC_PIN= 0, 1, 2, ..., 29)    Esc type is HW4 (Hobbywing V4)

Protocol is Jeti (Exbus)
CRSF baudrate   = 420000
Logger baudrate = 115200
PWM is generated at = 50 Hz
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is not detected
Airspeed sensor is not detected
    No Vspeed compensation channel defined; oXs uses default settings
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox (configured by oXs) :GPS is not (yet) detected
Led color is normal (not inverted)
Failsafe type is HOLD

Gyro is not configured

No sequencers are defined

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
809B
   0
   0
   2
   0
   0
   0
  32
   0
   0
   0
   3
  9A
   0
   0
   D
  A7
   D
  AE
Esc Volt=922   current=0  consumed=0  temp1=3495  temp2=3502
809B
  9B
   3
  E8
   1
  21
  FF
  21
  67
  78
  21
  21
  B9
809B
   0
   0
   3
   0
   0
   0
  32
   0
   0
   0
   3
  9A
   0
   0
   D
  A4
   D
  AD
Esc Volt=922   current=0  consumed=0  temp1=3492  temp2=3501
On the transmitter I only see vario and altitude, no ESC data, also no 
fields for ESC data. In addition, the ESC only initialises when I switch the sensor to the RP 2040 after initialising the ESC.
This behaviour is also described by dgat and recommends a delay of 2-3 seconds for initialisation.
Jürgen

Satcomix, 2023-11-16:
Hello Jürgen,
your values ​​from the terminal ESC debug output look the same as mine, with the HW Platinum V4 60A
Have you only taken the telemetry wire, GND and PWM to ESC from the ESC? 
The red wire with the BEC voltage from the ESC should not be used.
greetings,
Torsten

Satcomix, 2023-11-16:
Hello Jürgen,
I also had to put my ESC into a certain mode with my USB LCD programming box to even receive telemetry values. Helicopter (store governor), infixed Wing i get no data.
I see all telemetry values at my handset display VFAS, FASS, Capacity, TEMP1, TEMP2, RPM
greetings,
Torsten

mstrens, 2023-11-16:
Based on the values you provided, it is clear that oXs can decode the frames sent by ESC.
For Exbus, oXs has to build at power up a list of all fields that can be transmitted (based on the config) even if some sensors (like GPS) does not provide immediately their data.
This list is mainly build based on the pin being used.
I forgot to adapt it when a pin was defined for ESC.
This explain why you did not got the ESC data.
I put on github a version 2.10.0 that should fix the issue.

Satcomix, 2023-11-16:
Ver. 2.10.0 works with FBUS protocol and HW V4

Esc Volt=1334   current=0  consumed=0  temp1=3319  temp2=3428
809B
   0
   1
  1B
   1
  CD
   1
  C8
   0
  D8
  36
   5
  36
   0
   0
   C
  F6
   D
  6A
Esc Volt=1334   current=0  consumed=0  temp1=3318  temp2=3434
809B
   0
   1
  1C
   1
  CD
   1
  C9
   0
  D6
  D3
   5
  35
   0
   0
   C
  F6
   D
  70
Esc Volt=1333   current=0  consumed=0  temp1=3318  temp2=3440
809B
   0
   1
  1D
   1
  CD
   1
  C4
   0
  D7
  38
   5
  37
   0
   0
   C
  F6
   D
  69
Esc Volt=1335   current=0  consumed=0  temp1=3318  temp2=3433
809B
   0
   1
  1E
   1
  CD
   1
  C3
   0
  D7
  6B
   5
  3C
   0
   0
   C
  F3
   D
  6D
Esc Volt=1340   current=0  consumed=0  temp1=3315  temp2=3437

Klippeneck, 2023-11-16:
Hello Mstrens,
Test with 2.10.0 has the problem with the missing telemetry windows
has been fixed. Function is fine so far. Only the problem with the missing initialisation of the ESC when the ESC telemetry pins are permanently connected to the PR 2040 still needs to be solved. 
Temperatures are already correct, for current and voltage some factors should still be adjustable, possibly in the configuration via USB.
Jürgen

mstrens, 2023-11-16:
About the missing initialisation of the ESC:
Sorry but at this stage, it is not clear for me what has to be done.
Is it so that you have to start the RP2040 without connecting the ESC telemetry wire and only after some delay you can connect the ESC telemetry wire?

About some factors :  for voltage, you can use scaleVolt1 and for current scaleVolt2.

Klippeneck, 2023-11-16:
Hello Mstrens,
in my test setup I use the BEC of the controller as power supply 
the receiver and the RP2040.
When I connect the battery to the ESC with the telemetry input of the RP 2040 connected, the ESC the receiver and the RP 2040 receive power simultaneously. Then the ESC does not initialise, only if I disconnect the telemetry input of the RP for 2-3 seconds when connecting the power supply and then connect it, the initialisation works.
Dgat solves the problem with an "Init Delay", presumably the telemetry input of the RP is switched to high impedance for 2-3 seconds.
Thanks for the tip with the factors.
Jürgen

Satcomix, 2023-11-16:
Hello Jürgen,
to adjust V1=Voltage and V2=Current, you must set SCALE1=, OFFSET1= and SCALE2=, OFFSET2=.
Greetings,
Torsten

mstrens, 2023-11-16:
The ESC was configured with a pullup resistor (by oXs).
I removed this in 2.10.1.
You can test this version and see if it works.
I hope that HobbyWing drives the line High.
If this is not the case, it could be that oXs does not get anymore a valid signal. Then I would have to implement a delay in oXs but it is more complex.

Satcomix, 2023-11-17:
Good morning Mstrens,
I just tested version 2.10.1 with X20S and FBUS protocol.
Some telemetry sensors are missing!
**V.Speed ​​02 0110 (MP6050) is missing.**
The comp V.Speed ​​02 011F from the MS4525 is available.
Furthermore, **ESC telemetry values ​​are no longer displayed** due to the change of the resistor (pullup)

processing cmd

Version = 2.10.1 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255    8  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255    9  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255   28   29 (V1 / V4 = 26, 27, 28, 29)
RGB led . . . . . . . . . =   18  (RGB    = 0, 1, 2, ..., 29)
Logger  . . . . . . . . . =    0  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =    1  (ESC_PIN= 0, 1, 2, ..., 29)    Esc type is HW4 (Hobbywing V4)

Protocol is Fbus(Frsky)
CRSF baudrate   = 420000
Logger baudrate = 115200
PWM is generated at = 50 Hz
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 0.100000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 2.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Aispeed sensor is detected using MS4525
    Vspeed compensation channel = 7
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Foreseen GPS type is Ublox (configured by oXs) :GPS is detected and has a fix
Led color is normal (not inverted)
Failsafe uses predefined values
     Chan 1...4  =  1500  1500  1500  1500
     Chan 5...8  =  2012  2012  1500  1500
     Chan 9...12 =  1500  1500  1500  1500
     Chan 13...16=  1500  1500  1500  1500
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 1319 , 263 , -2756
     Gyro offsets         X, Y, Z = 1 , -9 , -12
     Orientation          Horizontal is FRONT(X+)     Vertical is WRONG

Gyro is not configured

Number of: sequencers=2   sequences=4   steps= 6
Sequencer = [  Gpio  Type(0=servo,1=analog) Clock(msec) ChannelNr Default Min Max ]
Sequence  = (  RC_value(-100...100)  to_Repeat  Uninterrupted   Only_priority_interrupted   is_a_Priority_seq  )
Step      = {  Smooth(clocks) Pwm(-100...100) Keep(clocks)  }
SEQ=
[ 6 0 20 9 -100 -100 100 ] 
  ( 50 P )
    {0 -100 20} 
  ( 100 O )
    {0 100 20} {0 -100 20} 
[ 7 0 20 9 -100 -100 100 ] 
  ( 50 P )
    {0 -100 20} 
  ( 100 O )
    {0 100 20} {0 -100 20} 
Error in parameters: gyro vertical orientation (0) is not valid

Config parameters are OK
Press ? + Enter to get help about the commands
8041
   0
  41
   0
  42
   0
   0
   5
  82
808A
  82
  A2
  82
  C2
  A2
  AA
80B3
  B3
   8
  5B
   0
   1
   0
  21
  21
  B9
8000
   1
   0
   0
   0
  14
   0
   0
   0
   5
  83
   0
   0
   D
  61
   D
  97
  B9
80B3
  B3
   8
  5B
   0
   1
   0
  21
  21
  B9
8000
   2
   0
   0
   0
  14
   0
   0
   0
   5
  84
   0
   0
   D
  63
   D
  9A
  B9
80B3
  B3
   8
  5B
   0
   1
   0
  21
  21
  B9
8000
   3
   0
   0
   0
  14
   0
   0
   0
   5
  86
   0
   0
   D
  60
   D
  95
  B9
80B3
  B3
   8
  5B
   0
   1
   0
  21
  21
  B9
8000
   4
   0
   0
   0
  14
   0
   0
   0
   5
  84
   0
   0
   D
  5A
   D
  95
  B9
80B3
  B3
   8
  5B
   0
   1
   0
  21
  21
  B9
8000
   5
   0
   0
   0
  14
   0
   0
   0
   5
  85
   0
   0
   D
  5A
   D
  96
  B9
80B3
  B3
   8
  5B
   0
   1
   0
  21
  21
  B9
8000
   6
   0
  6B
   0
  14
   0
   0
   0
   5
  84
   0
   0
   D
  54
   D
  94
  B9
80B3
  B3
   8
  5B
   0
   1
   0
  21
  21
  B9
8000
   7
   1
  1A
   0
  14
   0
   0
   0
   5
  84
   0
   0
   D
  50
   D
  94
  B9
80B3
  B3
   8
  5B
   0
   1
   0
  21
  21
  B9
8000
   8
   1
  B7
   0
  14
   0
   0
   0
   5
  84
   0
   0
   D
  50
   D
  93
8000
   9
   1
  DB
   0
  14
   0
   4
  EC
   5
  84
   0
   0
   D
  50
   D
  93
8000
   A
   1
  DC
   0

Satcomix, 2023-11-17:
The new 2 step calibration prozess ???
Force MPU6050 calibration   MPUCAL=Y            Y = H (Horizontal and still) or V (Vertical=nose up)

Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = -64 , 373 , -2756
     Gyro offsets         X, Y, Z = 0 , -9 , -12
     **Orientation          Horizontal is UP(Z+)     Vertical is UP(Z+)**

mstrens, 2023-11-17:
Can you test 2.10.2.
I tried to activate pullup only after a delay of 3 sec.
So the first bytes read by oXs would probably be wrong but after 3 sec, it should be OK.

Satcomix, 2023-11-17:
**Telemetry can be seen again on the handset**
Test with Ver.2.10.2 and X20S TD-R10 with FBUS protocol.
HobbyWing Platinum **V4** 60A

Config parameters are OK
Press ? + Enter to get help about the commands
809C
   0
  41
   0
  41
   0
  41
   0
   0
   5
  82
8030
  31
  30
  34
  30
  38
  34
  35
809B
  9B
   3
  E8
   1
   8
  5B
   0
   1
   0
  21
  21
  B9
809B
   0
   0
   1
   0
   0
   0
  14
   0
   0
   0
   5
  87
   0
   0
   D
  63
   D
  89
Esc Volt=1415   current=0  consumed=0  temp1=3427  temp2=3465
809B
  9B
   3
  E8
   1
   8
  5B
   0
   1
   0
  21
  21
  B9
809B
   0
   0
   2
   0
   0
   0
  14
   0
   0
   0
   5
  83
   0
   0
   D
  58
   D
  87
Esc Volt=1411   current=0  consumed=0  temp1=3416  temp2=3463
809B
  9B
   3
  E8
   1
   8
  5B
   0
   1
   0
  21
  21
  B9
809B
   0
   0
   3
   0
   0
   0
  14
   0
   0
   0
   5
  87
   0
   0
   D
  5B
   D
  86
Esc Volt=1415   current=0  consumed=0  temp1=3419  temp2=3462
809B
  9B
   3
  E8
   1
   8
  5B
   0
   1
   0
  21
  21
  B9
809B
   0
   0
   4
   0
   0
   0
  14
   0
   0
   0
   5
  84
   0
   0
   D
  52
   D
  8C
Esc Volt=1412   current=0  consumed=0  temp1=3410  temp2=3468
809B
  9B
   3
  E8
   1
   8
  5B
   0
   1
   0
  21
  21
  B9
809B
   0
   0
   5
   0
   0
   0
  14
   0
   0
   0
   5
  84
   0
   0
   D
  4F
   D
  87
Esc Volt=1412   current=0  consumed=0  temp1=3407  temp2=3463
809B
  9B
   3
  E8
   1
   8
  5B
   0
   1
   0
  21
  21
  B9
809B
   0
   0
   6
   0
   0
   0
  14
   0
   0
   0
   5
  86
   0
   0
   D
  50
   D
  84
Esc Volt=1414   current=0  consumed=0  temp1=3408  temp2=3460
809B
  9B
   3
  E8
   1
   8
  5B
   0
   1
   0
  21
  21
  B9
809B
   0
   0
   7
   0
   0
   0
  14
   0
   0
   0
   5
  84
   0
   0
   D
  47
   D
  88
Esc Volt=1412   current=0  consumed=0  temp1=3399  temp2=3464
809B
  9B
   3
  E8
   1
   8
  5B
   0
   1
   0
  21
  21
  B9
809B
   0
   0
   8
   0
   0
   0
  14
   0
   0
   0
   5
  85
   0
   0
   D
  48
   D
  84
Esc Volt=1413   current=0  consumed=0  temp1=3400  temp2=3460
809B
  9B
   3
  E8
   1
   8
  5B
   0
   1
   0
  21
  21
  B9
809B
   0
   0
   9
   0
  EB
   0
  14
   0
   0
   0
   5
  84
   0
   0
   D
  41
   D
  84
Esc Volt=1412   current=0  consumed=0  temp1=3393  temp2=3460
809B
  9B
   3
  E8
   1
   8
  5B
   0
   1
   0
  21
  21
  B9
809B
   0
   0
   A
   1
  22
   0
  14
   0
   0
   0
   5
  87
   0
   0
   D
  44
   D
  88
Esc Volt=1415   current=0  consumed=0  temp1=3396  temp2=3464
809B
  9B
   3
  E8
   1
   8
  5B
   0
   1
   0
  21
  21
  B9
809B
   0
   0
   B
   1
  90
   0
  14
   0
   0
   0
   5
  85
   0
   0
   D
  40
   D
  88
Esc Volt=1413   current=0  consumed=0  temp1=3392  temp2=3464
809B
   0
   0
   C
   1
  94
   0
  14
   0
   0
   0
   5
  85
   0
   0
   D
  40
   D
  88
Esc Volt=1413   current=0  consumed=0  temp1=3392  temp2=3464
809B
   0
   0
   D
   1
  95
   0
  14
   0
   0
   0
   5
  85
   0
   0
   D
  40
   D
  88
Esc Volt=1413   current=0  consumed=0  temp1=3392  temp2=3464
809B
   0
   0
   E
   1
  97
   0
  14
   0
   5
  35
   5
  85
   0
   0
   D
  40
   D
  88
Esc Volt=1413   current=0  consumed=0  temp1=3392  temp2=3464
809B
   0
   0
   F
   1
  99
   0
  14
   0
   3
  E9
   5
  85
   0
   0
   D
  40
   D
  88
Esc Volt=1413   current=0  consumed=0  temp1=3392  temp2=3464
809B
   0
   0
  10
   1
  99
   0
  14
   0
   4
  8F
   5
  85

mstrens, 2023-11-17:
> The new 2 step calibration prozess ??? Force MPU6050 calibration MPUCAL=Y Y = H (Horizontal and still) **or** V (Vertical=nose up)
> 
> Acc/Gyro is detected using MP6050 Acceleration offsets X, Y, Z = -64 , 373 , -2756 Gyro offsets X, Y, Z = 0 , -9 , -12 **Orientation Horizontal is UP(Z+) Vertical is UP(Z+)**
> 
> Orientation(Hoz./Vert.)GOH = X or GOV = X 0(front X+), 1(back X-), 2(left Y+), 3(right Y-), 4(up Z+), 5(down Z-)
> 
> What do I have to adjust when installing the MP6050 horizontally with X axis (to the nose) and Y axis (to the wing) orientation.

Note: best is to discuss gyro in a specific issue.
The easier (also can be part of testing) is to use the 2 usb commands:
While the mp6050 is horizontal (X and Y axis horizontal) and still, enter the command MPUCAL=H and wait a few seconds (result will be displayed.
Then hold the MP6050 in such a way that X axis (the arrow) point to the sky (so nearly vertically) and while in this position enter the command MPUCAL=V.
After this, if you ask the config, you should get something like : horizontal = UP   vertical = front

mstrens, 2023-11-17:
About vspeed:
Vspeed is probably missing because the mpu6050 did not had a valid orientation.
In this case, oXs can't calculate the vertical acceleration.
I suggest that you first calibrate the MP6050.

Satcomix, 2023-11-17:
Thanks, the V.Speed ​​0110 is back in telemetry.
MPUCAL=H at bottom and MPUCAL=V while X points to the horizon brought the solution.

mstrens, 2023-11-17:
> Thanks, the V.Speed ​​0110 is back in telemetry. MPUCAL=H at bottom and MPUCAL=V while X points to the horizon brought the solution.

The configuration you had previously (both horizontal and vertical in the same direction - or opposite) is not possible (physically).
In principe this is detected by oXs and gives a message when the config is printed. Still it does not set the config as invalid because  this should block all oXs functions and so it would not be possible to make the MPUCAL to fix the orientation.

Satcomix, 2023-11-17:
You wrote: Note: best is to discuss gyro in a specific issue. ;-))

Klippeneck, 2023-11-17:
Hello,
SW 2.10.2 has solved the problem with the missing initialisation.
Vario, altitude, voltage, temperatures and consumption seem plausible.
Current is constantly at 0 A although consumption is counting up.
The LED on the RP flashes red when the sensor is connected to the receiver, green on the terminal programme (Arduino).
Jürgen

Satcomix, 2023-11-17:
Hello Jürgen,
So voltage, current and capacity now appear on your handset display?
Current is constant 0A but the capacity counts up? 
Normaly it stays at 0mAh by 0A
I don't understand the next statement.
The LED on the RP flashes red when the sensor is connected to the receiver, green on the terminal program (Arduino).
The ESC V5 must be connected to the oXs board at ESC_PIN= and the oXs board is connected to the receiver via TLM= or PRI=(FBUS).
The board is connected to the Arduino terminal via USB.
greetings,
Torsten

Klippeneck, 2023-11-17:
Hello Torsten,
Yes, voltage, current, 2 temperatures (BEC temperature and FET temperature) and the capacity are displayed on the transmitter. The current is constantly at zero even when the motor is running with the LS, but the capacity counts up. The other values are OK.
When configuring the terminal programme via USB, the LED flashes green; if I connect the sensor to the receiver (EX bus input), the LED flashes red. The ESC to sensor connection is as you have described. 
PRI goes to EX bus input receiver.
Jürgen
11:18:13.582 -> processing cmd
11:18:13.582 -> 
11:18:13.582 -> Version = 2.10.2 
11:18:13.582 ->     Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
11:18:13.582 -> Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
11:18:13.582 -> Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
11:18:13.582 -> Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
11:18:13.582 -> GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
11:18:13.582 -> GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
11:18:13.582 -> Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
11:18:13.582 -> RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
11:18:13.582 -> SDA (I2C sensors) . . . . =    2  (SDA     = 2, 6, 10, 14, 18, 22, 26)
11:18:13.582 -> SCL (I2C sensors) . . . . =    3  (SCL     = 3, 7, 11, 15, 19, 23, 27)
11:18:13.582 -> PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
11:18:13.582 -> PWM Channels 5, 6, 7 ,8   =  255  255  255  255
11:18:13.582 -> PWM Channels 9,10,11,12   =  255  255  255  255
11:18:13.582 -> PWM Channels 13,14,15,16  =  255  255  255  255
11:18:13.582 -> Voltage 1, 2, 3, 4        =  255  255  255  255 (V1 / V4 = 26, 27, 28, 29)
11:18:13.582 -> RGB led . . . . . . . . . =   16  (RGB    = 0, 1, 2, ..., 29)
11:18:13.582 -> Logger  . . . . . . . . . =  255  (LOG    = 0, 1, 2, ..., 29)
11:18:13.582 -> ESC . . . . . . . . . . . =    1  (ESC_PIN= 0, 1, 2, ..., 29)    Esc type is HW4 (Hobbywing V4)
11:18:13.582 -> 
11:18:13.582 -> Protocol is Jeti (Exbus)
11:18:13.582 -> CRSF baudrate   = 420000
11:18:13.582 -> Logger baudrate = 115200
11:18:13.582 -> PWM is generated at = 50 Hz
11:18:13.582 -> Voltage parameters:
11:18:13.582 ->     Scales : 13.100000 , 12.400000 , 1.000000 , 1.000000 
11:18:13.582 ->     Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
11:18:13.582 ->     No temperature sensors are connected on V3 and V4
11:18:13.582 -> RPM multiplier = 1.000000
11:18:13.582 -> Baro sensor is detected using MS5611
11:18:13.582 ->     Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
11:18:13.582 ->     Hysteresis = 5 
11:18:13.582 -> Airspeed sensor is not detected
11:18:13.582 ->     No Vspeed compensation channel defined; oXs uses default settings
11:18:13.582 -> First analog to digital sensor is not detected
11:18:13.582 -> Second analog to digital sensor is not detected
11:18:13.582 -> Foreseen GPS type is Ublox (configured by oXs) :GPS is not (yet) detected
11:18:13.582 -> Led color is normal (not inverted)
11:18:13.582 -> Failsafe type is HOLD
11:18:13.582 -> Acc/Gyro is not detected
11:18:13.582 -> 
11:18:13.582 -> Gyro is not configured
11:18:13.582 -> 
11:18:13.582 -> No sequencers are defined
11:18:13.582 -> Error in parameters: gyro vertical orientation (0) is not valid
11:18:13.582 -> 
11:18:13.582 -> Config parameters are OK
11:18:13.582 -> Press ? + Enter to get help about the commands

Satcomix, 2023-11-17:
Hello Jürgen,
I think you have to invert RGB **LED=I**, because at USB without Receiver connected normally it flashes red, and with Receiver connected it flashes green.
Greetings,
Torsten

mstrens, 2023-11-17:
For Hobbywing, I used the code find on the web.
In order to reject dummy values received from time to time, this code has a check to avoid sending the current when it exceed some max value. This max value can be defined in the config.h file.
It was set on 250.0. I expect it was 250A.
In fact oXs should normally set this value in mA (using scaleVolt2).
I put on github a version 2.10.3 that set the max value to 250000.
I hope this can fix the issue.

Satcomix, 2023-11-17:
Version 2.10.3 works with FBUS and HW V4
All telemetry data are sent and displayed.

Klippeneck, 2023-11-17:
Hello Mstrens,
Power is now displayed with 2.10.3, change was successful.
Next I will determine the scale factors for Hobbywing FLY Fun 80A V5 and then go into flight testing.
Many thanks for your great work.
Juergen

mstrens, 2023-11-21:
Seems to be OK

closed 2023-11-21 by mstrens

Klippeneck, 2023-11-21:
Hello Mstrens,
In my test setup, I switch the motor via a three-step switch on the transmitter, OFF, approx. 50% and full throttle. the displayed current value remains at approx. 4.5A, regardless of the throttle position. Measured with a shunt, it is approx. 10A at 50% and 22A at full throttle. This means that I cannot determine a current coefficient. Next I will carry out tests with a motor with a higher current consumption (approx. 50A).
I have to wait for the result of this test, then I will report again.
Greetings Jürgen

mstrens, 2023-11-21:
It is not normal that current does not change between 50% and 100% throttle.
Perhaps that the value is not transmitted anymore at 100%. You can probably see it the value is not transmitted anymore on the handset (e.g. in openTX the screen to discover the sensors shows blinking "*" when a new value is received).

There are some checks in the code for ESC HW4 that avoid (in theory) sending wrong data.
if (throttle > 1024 || pwm > 1024 || rpm > 200000 || escRxBuffer[11] & 0xF0 ||\
                escRxBuffer[13] & 0xF0 || escRxBuffer[15] & 0xF0 || escRxBuffer[17] & 0xF0 || escRxBuffer[1] == 0x9B)
        
Perhaps I have to put some debug msg again to see to display the raw values provided by the ESC.

Klippeneck, 2023-11-22:
Hello Mstrens,
was able to make further tests with the ESC.
Old setup with FLY FUN 80A, controlled with three-step switch on the transmitter.
First attempt with Scale2 factor 90, then the current is correct at full throttle, 40A.
At 50% 33A are displayed, real at the shunt 1O A.
Second test with the same V2 factor and smaller motor.
At 100% 38A, target is 20A, at 50% 33A, target 8A.
Offset2 is always 0.
The remaining telemetry values of the ESC are correct.
There must be a bug in the current display.
Jürgen

mstrens, 2023-11-22:
Can you try version 2.10.11.
In this version, I display (on the pc) the raw values received from the ESC.
Can you capture those values for different levels of current.
Please provide those captures and say what is the "real" current for each capture.

[16 more replies]