mstrens/oXs_on_RP2040 #117: Request : Add number of GPS SAT

J-charles-C, 2023-11-19:
Good morning,

Everything works very well (RPM, GPS, Vario). Configuration via the serial monitor is a big advantage.

Can you, please, add the number of satellites to the GPS data?

THANKS

Jc

mstrens, 2023-11-19:
Number of satellites is normally transmitted but some protocols does not have a field to transmit the data.
Which protocol do you use?

J-charles-C, 2023-11-19:
I,

fr-sky S-port

mstrens, 2023-11-19:
Number of sat is transmitted in field ID 5100.
When a 3d fix is available, oXs add 100 to the real number of sat.
See in doc folder the file fields per protocol.txt

J-charles-C, 2023-11-20:
Good morning,

Unfortunately, when searching for new sensors, this data does not appear. I have GPOS speed, GPS altitude, angle, time and coordinates.

THANKS

mstrens, 2023-11-20:
did you first deleted all existing telemetry fields on the handset.
I had cases where I had to delete all telemetry fields before asking for new discovery.

Otherwise, if you use the command FVP and then ask for new discovery do you get it (with dummy fix values then). This is a way to test if all possible fields are transmitted.
To see if the data is really available (internally to oXs) you can also use the command FV that displays all internal values on the pc

Satcomix, 2023-11-20:
Hello J-charles-C,
I have just carried out tests again with version 2.10.6-test and FrSky X20S and X20Pro with TD-R10 Receiver once each with **S.PORT and SBUS as well as FBUS Protocol**.
I receive all telemetry values ​​defined in fields_per_protocol.txt displayed on my handset display.
I have just carried out tests again with version 2.10.6-test and FrSky X20S and X20Pro with TD-R10 Receiver once each with S.PORT and SBUS as well as FBUS Protocol. Of course also DIY 5100 GPS 3D+Number.

**Protocol is Sport (Frsky)**
Cmd to execute:   FV

GPS Latitude = xx.xxxxxxxxxxx degree
GPS Longitude = x.xxxxxxxxxx degree
GPS Groundspeed = 36 cm/s
GPS Heading = 74.880000 degree
GPS Altitude = 2963 cm
**GPS Num sat. = 106**
GPS Date J M A = 20 11 23 
GPS Time H M S = 7 58 33 
GPS Pdop = 346 
GPS Home bearing = 183 degree
GPS Home distance = 103 m
Volt 1 = 1555 mVolt
Current (Volt 2) = 1812 mA
Volt 4 = 1601 mVolt
Capacity (using current) = 250 mAh
Temp 1 (Volt 3) = 21 degree
Vspeed = -6 cm/s
Baro Rel altitude = 184 cm
Pitch = 2.000000 degree
Roll = 0.000000 degree
RPM = 1008 Hertz
Ads 1 1 = 119 mVolt
Ads 1 2 = 119 mVolt
Ads 1 3 = 119 mVolt
Ads 1 4 = 119 mVolt
Ads 2 1 = 119 mVolt
Ads 2 2 = 119 mVolt
Ads 2 3 = 118 mVolt
Ads 2 4 = 119 mVolt
Airspeed = 72 cm/s
Compensated Vspeed = 6 cm/s
Gps cumulative distance = 335
Acc X = 0.041000g
Acc Y = -0.001000g
Acc Z = 0.992000g
Vspeed compensation = 1.10
pwmTop= 20000

greetings,
Torsten

J-charles-C, 2023-11-20:
Good morning,

Thank you both. I was using 2.8.0, taking the latest from the main branch, and using DIY sensors it works.
THANKS

closed 2023-11-20 by J-charles-C