mstrens/oXs_on_RP2040 #124: Hobbywing Capacity Problem

Klippeneck, 2024-02-23:
Hello Mstrens,
I think I have found another problem with the OXS RP2040 sensor with SW 2.10.11.
The sensor is configured as Vario, GPS and Hobbywing telemetry for Jeti,
The telemetry values from the Vario, GPS and ESC voltage and current and temperatures are ok, only the calculation of the removed capacity still seems to be faulty.
For example, on the first flight with the electric glider, the 120mAh capacity taken for the start
after 45 minutes of sailing (without motor) jumped to 1600mAh the next time the motor was switched on.
Attached is a snapshot of the log file.
Juergen

[image]

mstrens, 2024-02-23:
When telemetry data are provided by the ESC, oXs does not perform capacity calculation. It transmits the values provided by the ESC.
So I presume that the issue is in the ESC software.
Perhaps, that the capacity is calculated only on the current used for the motor (and not for powering the servos/receiver)?

mstrens, 2024-02-23:
I made some changes in 2.11.41 (test branch).
I hope it helps.

Current (and so consumption) was discarded when throttle was less than 1/4.
I removed this check (was in some code I reused).

Klippeneck, 2024-02-23:
Hello Mstrens,
Thank you very much for your quick response.
If the weather permits tomorrow, I'll test it and get back to you.
Jürgen

Klippeneck, 2024-02-24:
Good morning Mstrens,
Tests with SW2.11.41 show the same result, including jumps in the capacity curve after pauses in engine operation.
 See screenshot.
Jürgen 
[image]

mstrens, 2024-02-24:
That is strange.
I do not see directly the bug.
Could you provide me the full log (e.g. in csv file)
I would e.g. see if volt1 and current changes when the motor is off.

mstrens, 2024-02-24:
It looks like oXs does not get valid data from the esc when throttle is off.
oXs remembers the last time it got a valid frame.
When it get a new valid frame, oXs calculates the elapse time and multiply it with the new current.
This could explain the graph that you show.

In the code I initially reused, there was a test to discard the ESC current when throttle % was less than 25%.

It seems that the hobbywing esc sent different frames and there are checks to discard some of them.

I could make a version that print a message on the usb/serial with the raw content of all frames received from esc.
This would help to see what happens.
Could you then make a test with the oXs connected to the PC and to the ESC?
I so, I will make such a version.

mstrens, 2024-02-24:
I put version 2.11.42 on github.
There are more debug messages sent to usb/serial. This could be useful if you can make a test when connected to the PC and copy/paste the messages.

I also reset the elapse time after an invalid frame. This will perhaps help.

Klippeneck, 2024-02-24:
[Test_RP2040_ESC.csv](https://github.com/mstrens/oXs_on_RP2040/files/14392506/Test_RP2040_ESC.csv)
With Version 2.11.41

mstrens, 2024-02-24:
Your test with 2.11.41 confirms that:
- oXs does not get valid data from ESC during about 7min (data does not changed)
- the first valid current after 7min was 1.14 Amp and this has been used with an elapse time of about 7min.
- this explain that consumption has increase suddenly from 67 to 204.

In version 2.11.42, you should not have anymore this sudden increase.
Still it would be good to look at the raw data in the frames sent by the ESC to understand why frames where discarded.

mstrens, 2024-02-24:
To get the raw frame data on the usb/serial, I should make a version for you or you could use 2.11.42 and change a few lines in esc.cpp and compile your self.
The lines to change are: 
//for (uint8_t i=0; i< escMaxFrameLen ; i++){
//     printf("%2X ",escRxBuffer[i] );   
//}
//printf("\n");

Just remove the // in front of the lines to activate them.

Klippeneck, 2024-02-24:
Hello Mstrens,
I have the following data on the USB interface with SW 2.11.42 without any changes 
received. I am not able to change 2.11.42 myself at the moment with my knowledge.
I would have to familiarise myself first.
[ESC_Test2.odt](https://github.com/mstrens/oXs_on_RP2040/files/14393639/ESC_Test2.odt)

mstrens, 2024-02-24:
I have an issue when I try to open the file in Word. I get a message saying the file is corrupted.
In Wordpad, I can read the beginning up to 17:32:43.999
Is this the end of the file?

This first part seems ok.

Klippeneck, 2024-02-24:
From 17;31:58.361 i conect the Accu.
From 17:32:04.797 to 17:32:10.140 I Schwitch Motor On.
From 17:32:04.140 to 17:32:39.291 i Motor Stop
From 17:32:39.291 to 17:32:43.999 sekond Run
Jürgen

mstrens, 2024-02-24:
With 2.11.42 the messages where printed when esc data could be processed (after some checks).
When motor is swith off, there is no "valid" messages.

I made version 2.11.43 (now on github) with more debug messages (all raw data's).
I hope we can see more.

Klippeneck, 2024-02-24:
Test with 2.11.43 ( Run-Pause-Run)
[ESC_Test2.11.43.odt](https://github.com/mstrens/oXs_on_RP2040/files/14394114/ESC_Test2.11.43.odt)
Jürgen

Klippeneck, 2024-02-24:
Hello Mstrens,
In-house tests with SW 2.11.44 are successful.
No more jumps in capacity.
Thank you very much.
Juergen

mstrens, 2024-02-24:
Thanks for the feedback.
I can also make one more change:
- fill the current field with 0 when the interval between 2 ESC frames is more than 0.8 sec
This would avoid having the handset displaying the current from the last frame.

mstrens, 2024-02-24:
Sorry, sending a telemetry field with current=0 when ESC does not provide a frame withing 0.8sec would be to difficult because the logic is currently to react on incoming frame only.

Klippeneck, 2024-02-24:
Hello Mstrens,
No problem, as the BEC consumption is not measured anyway and is not included in the capacity calculation.
With my model, this is approx. 200mAh/hour, which I add manually depending on the flight time.
Then the recharged capacity corresponds well with the capacity displayed in the transmitter.
Jürgen

closed 2024-02-25 by Klippeneck