mstrens/oXs_on_RP2040 #125: config.h

secretman28, 2024-02-25:
if i want to set my gpio pin fora all in one flash exemple

 #define _pinChannels_1  0XFF
 #define _pinChannels_2  0XFF
 #define _pinChannels_3  0XFF
 #define _pinChannels_4  0XFF
 #define _pinChannels_5  0XFF
 #define _pinChannels_6  0XFF
 #define _pinChannels_7  0XFF
 #define _pinChannels_8  0XFF
 #define _pinChannels_9  0XFF
 #define _pinChannels_10  0XFF
 #define _pinChannels_11  0XFF
 #define _pinChannels_12  0XFF
 #define _pinChannels_13  0XFF
 #define _pinChannels_14  0XFF
 #define _pinChannels_15  0XFF
 #define _pinChannels_16  0XFF

did 0xff will be the gpio pin? or i need to find the right hex?
can i do 👍 
 #define _pinChannels_1  0
 #define _pinChannels_2  1
 #define _pinChannels_3  2
 #define _pinChannels_4  3
 #define _pinChannels_5  4
 #define _pinChannels_6  5
 etc...
??

mstrens, 2024-02-25:
If you want to avoid having to send commands via usb/serial to setup your oXs, you can use the defines in config.h
So you can use
#define _pinChannels_1 0
#define _pinChannels_2 1
#define _pinChannels_3 2
#define _pinChannels_4 3
#define _pinChannels_5 4
#define _pinChannels_6 5
...
Still this is taken into account only if you flash a new rp2040 (one with a totally clean flash memory).
If you already saved some parameters in the rp2040, it will used the saved parameters (and not those defines from the config.h).
You can clear totally the rp2040 flash memory using the flash_nuke.uf2 file that is available in the doc folder.
Upload this file before uploading  oXs program.

secretman28, 2024-02-25:
perfect tanks!

secretman28, 2024-02-25:
an other question, with new release enter dont get a list of what is already set
you change the command its not  enter we need to input a command?

secretman28, 2024-02-25:
sry just saw other post my bad

mstrens, 2024-02-25:
Enter alone should display the current setup.
? should display the list of commands
Do not forget to apply "save" command after one or several changes.
After "save", you should best make a power off/on

closed 2024-02-27 by secretman28