mstrens/oXs_on_RP2040 #126: Version 2.12.0 Debug message ???

Satcomix, 2024-03-02:
Hello Mstrens,
I want to make some tests with the new -test and -main and get after pushing some buttons for the sequencer this messages in terminal window. What is the meaning of the messages?

processing cmd

Version = 2.12.0 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =   21  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =  255  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =  255  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)
RGB led . . . . . . . . . =   16  (RGB    = 0, 1, 2, ..., 29)
Logger  . . . . . . . . . =  255  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =  255  (ESC_PIN= 0, 1, 2, ..., 29)
Locator CS  . . . . . . . =  255  (SPI_CS = 0, 1, 2, ..., 29)
        SCK . . . . . . . =  255  (SPI_SCK= 10, 14, 26)
        MOSI  . . . . . . =  255  (SPI_MOSI=11, 15, 27)
        MISO  . . . . . . =  255  (SPI_MISO=8, 12, 24, 28)
Esc type is not defined

Protocol is Fbus(Frsky)
CRSF baudrate   = 420000
Logger baudrate = 115200
PWM is generated at = 50 Hz
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is not detected
Airspeed sensor is not detected
    No Vspeed compensation channel defined; oXs uses default settings
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox (configured by oXs) :GPS is not (yet) detected
Led color is normal (not inverted)
Failsafe uses predefined values
     Chan 1...4  =  1500  1500  1500  1500
     Chan 5...8  =  1500  1500  1500  1500
     Chan 9...12 =  1500  1500  1500  1500
     Chan 13...16=  1500  1500  1500  1500
Acc/Gyro is not detected

Gyro is not configured

Number of: sequencers=16   sequences=48   steps= 48
Sequencer = [  Gpio  Type(0=servo,1=analog) Clock(msec) ChannelNr Default Min Max ]
Sequence  = (  RC_value(-100...100)  to_Repeat  Uninterrupted   Only_priority_interrupted   is_a_Priority_seq  )
Step      = {  Smooth(clocks) Pwm(-100...100) Keep(clocks)  }
SEQ=
[ 0 1 20 8 0 0 100 ] 
  ( -100 O )
    {0 0 1} 
  ( -15 P )
    {0 0 1} 
  ( 100 O )
    {0 100 1} 
[ 1 1 20 8 0 0 100 ] 
  ( -100 O )
    {0 0 1} 
  ( -20 P )
    {0 0 1} 
  ( 95 O )
    {0 100 1} 
[ 2 1 20 8 0 0 100 ] 
  ( -100 O )
    {0 0 1} 
  ( -25 P )
    {0 0 1} 
  ( 90 O )
    {0 100 1} 
[ 3 1 20 8 0 0 100 ] 
  ( -100 O )
    {0 0 1} 
  ( -30 P )
    {0 0 1} 
  ( 85 O )
    {0 100 1} 
[ 4 1 20 8 0 0 100 ] 
  ( -100 O )
    {0 0 1} 
  ( -35 P )
    {0 0 1} 
  ( 80 O )
    {0 100 1} 
[ 5 1 20 8 0 0 100 ] 
  ( -100 O )
    {0 0 1} 
  ( -40 P )
    {0 0 1} 
  ( 75 O )
    {0 100 1} 
[ 6 1 20 8 0 0 100 ] 
  ( -100 O )
    {0 0 1} 
  ( -45 P )
    {0 0 1} 
  ( 70 O )
    {0 100 1} 
[ 7 1 20 8 0 0 100 ] 
  ( -100 O )
    {0 0 1} 
  ( -50 P )
    {0 0 1} 
  ( 65 O )
    {0 100 1} 
[ 8 1 20 8 0 0 100 ] 
  ( -100 O )
    {0 0 1} 
  ( -55 P )
    {0 0 1} 
  ( 60 O )
    {0 100 1} 
[ 9 1 20 8 0 0 100 ] 
  ( -100 O )
    {0 0 1} 
  ( -60 P )
    {0 0 1} 
  ( 55 O )
    {0 100 1} 
[ 10 1 20 8 0 0 100 ] 
  ( -100 O )
    {0 0 1} 
  ( -65 P )
    {0 0 1} 
  ( 50 O )
    {0 100 1} 
[ 11 1 20 8 0 0 100 ] 
  ( -100 O )
    {0 0 1} 
  ( -70 P )
    {0 0 1} 
  ( 45 O )
    {0 100 1} 
[ 12 1 20 8 0 0 100 ] 
  ( -100 O )
    {0 0 1} 
  ( -75 P )
    {0 0 1} 
  ( 40 O )
    {0 100 1} 
[ 13 1 20 8 0 0 100 ] 
  ( -100 O )
    {0 0 1} 
  ( -80 P )
    {0 0 1} 
  ( 35 O )
    {0 100 1} 
[ 14 1 20 8 0 0 100 ] 
  ( -100 O )
    {0 0 1} 
  ( -85 P )
    {0 0 1} 
  ( 30 O )
    {0 100 1} 
[ 15 1 20 8 0 0 100 ] 
  ( -100 O )
    {0 0 1} 
  ( -90 P )
    {0 0 1} 
  ( 25 O )
    {0 100 1} 

Config parameters are OK
Press ? + Enter to get help about the commands
i=0 lo=0 s=21 c=6291456 m=-14680065 v1=8388607 v2=16777215
i=1 lo=0 s=18 c=786432 m=-1835009 v1=7340031 v2=16777215
i=2 lo=0 s=15 c=98304 m=-229377 v1=7208959 v2=16777215
i=3 lo=0 s=12 c=12288 m=-28673 v1=7192575 v2=16777215
i=4 lo=0 s=9 c=1536 m=-3585 v1=7190527 v2=16777215
i=5 lo=0 s=6 c=192 m=-449 v1=7190271 v2=16777215
i=6 lo=0 s=3 c=24 m=-57 v1=7190239 v2=16777215
i=7 lo=0 s=0 c=3 m=-8 v1=7190235 v2=16777215
i=8 lo=0 s=21 c=6291456 m=-14680065 v1=7190235 v2=8388607
i=9 lo=0 s=18 c=786432 m=-1835009 v1=7190235 v2=7340031
i=10 lo=0 s=15 c=98304 m=-229377 v1=7190235 v2=7208959
i=11 lo=0 s=12 c=12288 m=-28673 v1=7190235 v2=7192575
i=12 lo=0 s=9 c=1536 m=-3585 v1=7190235 v2=7190527
i=13 lo=0 s=6 c=192 m=-449 v1=7190235 v2=7190271
i=14 lo=0 s=3 c=24 m=-57 v1=7190235 v2=7190239
i=15 lo=0 s=0 c=3 m=-8 v1=7190235 v2=7190235
i=0 lo=0 s=21 c=6291456 m=-14680065 v1=8388607 v2=16777215
i=1 lo=0 s=18 c=786432 m=-1835009 v1=7340031 v2=16777215
i=2 lo=0 s=15 c=98304 m=-229377 v1=7208959 v2=16777215
i=3 lo=0 s=12 c=12288 m=-28673 v1=7192575 v2=16777215
i=4 lo=0 s=9 c=1536 m=-3585 v1=7190527 v2=16777215
i=5 lo=0 s=6 c=192 m=-449 v1=7190271 v2=16777215
i=6 lo=0 s=3 c=24 m=-57 v1=7190239 v2=16777215
i=7 lo=0 s=0 c=3 m=-8 v1=7190235 v2=16777215
i=8 lo=0 s=21 c=6291456 m=-14680065 v1=7190235 v2=8388607
i=9 lo=0 s=18 c=786432 m=-1835009 v1=7190235 v2=7340031
i=10 lo=0 s=15 c=98304 m=-229377 v1=7190235 v2=7208959
i=11 lo=0 s=12 c=12288 m=-28673 v1=7190235 v2=7192575
i=12 lo=0 s=9 c=1536 m=-3585 v1=7190235 v2=7190527
i=13 lo=0 s=6 c=192 m=-449 v1=7190235 v2=7190271
i=14 lo=0 s=3 c=24 m=-57 v1=7190235 v2=7190239
i=15 lo=0 s=0 c=3 m=-8 v1=7190235 v2=7190235
i=0 lo=0 s=21 c=6291456 m=-14680065 v1=8388607 v2=16777215
i=1 lo=0 s=18 c=786432 m=-1835009 v1=7340031 v2=16777215
i=2 lo=0 s=15 c=98304 m=-229377 v1=7208959 v2=16777215
i=3 lo=0 s=12 c=12288 m=-28673 v1=7192575 v2=16777215
i=4 lo=0 s=9 c=1536 m=-3585 v1=7190527 v2=16777215
i=5 lo=0 s=6 c=192 m=-449 v1=7190271 v2=16777215
i=6 lo=0 s=3 c=24 m=-57 v1=7190239 v2=16777215
i=7 lo=0 s=0 c=3 m=-8 v1=7190235 v2=16777215
i=8 lo=0 s=21 c=6291456 m=-14680065 v1=7190235 v2=8388607

mstrens, 2024-03-02:
Those are some debug messages I had in sequencer.cpp to debug.
you can put line 169 
            printf("i=%i lo=%i s=%i c=%i m=%i v1=%i v2=%i\n" , i, lastOutput , shift, code , mask , value1 , value2); 
as comment and you will not get those messages anymore.

Satcomix, 2024-03-02:
Hello Mstrens,
Thank you for your reply, it works with comment.
best regards,
Torsten

closed 2024-03-02 by Satcomix