mstrens/oXs_on_RP2040 #128: Adoption of XGZP6897D Differential Pressure Sensor

jasc76, 2024-03-05:
Datasheet

[XGZP6897D-Pressure-Sensor-V2.7.pdf](https://github.com/mstrens/oXs_on_RP2040/files/14493394/XGZP6897D-Pressure-Sensor-V2.7.pdf)

In Test is the 3.3V 500Pa Variant. Currently the sensor is recognized, but no measurements are displayed when using FV. In parallel a GY-63 is connected so I²c should be terminated without using pullups

> Baro sensor is detected using MS5611
> Sensitivity min = 100 (at 100) , max = 300 (at 1000)
> Hysteresis = 5
> Airspeed sensor is detected using XGZP....

jasc76, 2024-03-05:
Just downloaded the 2.12.4 Version and will try to get some debug data

If you like I can send you one of these sensors (if you are in the EU... US takes ages) . I got 2 :-)

mstrens, 2024-03-05:
Perhaps we can find the issue with the debug messages.
In a second stage, if I do not find the issue, then getting a sensor could help.
I am in EU (Belgium).

jasc76, 2024-03-05:
looks good

rawPres=66695
rawPres=172799
rawPres=33783
rawPres=3823
rawPres=16777142
rawPres=115830
rawPres=205088
rawPres=320911
rawPres=486299
rawPres=16737849
rawPres=16747550
rawPres=16740163
rawPres=16736876
rawPres=783477
rawPres=16728544
rawPres=531232
rawPres=1093841
rawPres=1124545
rawPres=1056067
rawPres=649749
rawPres=16735820
rawPres=16743526
rawPres=16740288

mstrens, 2024-03-06:
In the version you tested I made 2 changes:
- one to add a debug message
- one to fix a minor bug. It is perhaps this fix that solved the issue.

mstrens, 2024-03-06:
I made a new version 2.12.5 in order to manage negative pressure (the sensor is differential and so some values can be negative.
Without this correction, negative values were processed as very big values and could generate wrong results.
I hope it help to get more realistic airspeed.

jasc76, 2024-03-06:
you're a wizard :-)
depending, which port gets the air pressure the value is pos or neg now. I guess ignoring pos/neg makes the most sense since its irrelevant for the application

rawPres=-30229
rawPres=-28695
rawPres=-1945446
rawPres=-40881
rawPres=-51391
rawPres=-27898
rawPres=-88071
rawPres=-73877
rawPres=-324429
rawPres=-889948
rawPres=898187
rawPres=773595
rawPres=-70024
rawPres=79785
rawPres=1257551
rawPres=700077
rawPres=-64377
rawPres=-61067
rawPres=1076767
rawPres=-61751
rawPres=-35367
rawPres=-30116
rawPres=-29381

jasc76, 2024-03-06:
hmm. at the same time there seems to be a lot of noise

rawPres=1733
rawPres=3086
rawPres=-218
rawPres=1984
rawPres=4579
rawPres=1337
rawPres=175
rawPres=-1214
rawPres=-2869
rawPres=-1598
rawPres=-1104
rawPres=1779
rawPres=1744
rawPres=862
rawPres=335
rawPres=-1252
rawPres=-547
rawPres=103
rawPres=246
rawPres=1398
rawPres=1641
rawPres=-8
rawPres=-348

mstrens, 2024-03-06:
Yes but this is perhaps not so abnormal.
A  raw value like 2000 means in fact a pressure of 0.25 Pa which is equivalent to a speed of 2km/h

Best is to check the airspeed calculated by oXs and see if it is realistic.

Do not expect an realistic value when speed is less than about 15 km/h because this result of very low difference of pressure.
When speed get higher, then accuracy/noise of airspeed becomes better.

mstrens, 2024-03-06:
FYI tTaking into account the negative values is not bad:
- when program starts, oXs takes the average of several measurements to calculate an offset that is applied to the following measurements
- when oXs calculate the airspeed, if the pressure is negative, oXs converts it to a positive value. This allows to connect the pitot tube in both directions on the sensor.

jasc76, 2024-03-06:
ok cool

Currently this is all a crude test setup botched together with some wires. I now ordered stuff to actually mount it to my glider. I would assume I put it in the plane, hook up SBUS and tweak the compensation adjustment channel until I get a consistent result. Is this correct !?

Which oXs version should I use now? The master or still test?

mstrens, 2024-03-06:
Use always test branch. I put all new functions there. Still take care that there can be some not yet known bugs.

Some users tested the airspeed while running with a car. One man drives the car, another hold the pitot tube outside. The value is read on the handset.

jasc76, 2024-03-11:
Do I have to change the k value in the code or is the calibration done solely done with the compensation channel ?

mstrens, 2024-03-11:
You have to change manually in config.h the value for XGZP_K_FACTOR
It depends on the max pressure of the model you selected.
Here the text in config.h
#define XGZP_K_FACTOR 8192.0 ;  // See datasheet : K depends on the sensor model (related to the max pos pressure being measured)
                              // 8192.0 is the value when max sensor pressure is 1000Pa (1kPa) like XGZP6897D001KPDPN

mstrens, 2024-03-11:
The compensation is a parameter used only for the compensated aispeed. it does not matter to calculate the airspeed.

jasc76, 2024-03-12:
I dont really care for airspeed, I'm just interested in the vario compensation. So my thought was not to bother with the k value. Also in the datasheet, all models below 1kPa have the same k value.

jasc76, 2024-04-01:
hmm
I wanted to do the calibration, I used the current test branch 2.13.1 and I get changing raw values. However with FV airspeed is always 0. I did not hook up a receiver and just used usb

`rawPres=-683676
rawPres=-734410
rawPres=-800049
rawPres=-25184
rawPres=-217883
rawPres=-1596020
rawPres=-1711642
rawPres=-1651593
rawPres=-28809
rawPres=-33805
rawPres=-1309954
rawPres=-635123
rawPres=-1202301
rawPres=-690018
rawPres=-769427
rawPres=-567340
rawPres=-958961
rawPres=-1552390
rawPres=-1172935
rawPres=-1505920
rawPres=-1097496
rawPres=-78826
rawPres=-33733
rawPres=-2576
rawPres=-6549
rawPres=-4949
rawPres=-8274
rawPres=-7904
rawPres=-4487
rawPres=-22924
rawPres=-32619
rawPres=-3220531
rawPres=-2690748
rawPres=-3060192
rawPres=-3926
rawPres=-4940
rawPres=-1513
rawPres=-2688
rawPres=-6399
rawPres=-7482
rawPres=-5073
rawPres=-6046
rawPres=-8062
rawPres=-5920`

`Version = 2.13.1
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   10  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   11  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255  255  255 (V1 / V4 = 26, 27, 28, 29)
RGB led . . . . . . . . . =   16  (RGB    = 0, 1, 2, ..., 29)
Logger  . . . . . . . . . =  255  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =  255  (ESC_PIN= 0, 1, 2, ..., 29)
Locator CS  . . . . . . . =  255  (SPI_CS = 0, 1, 2, ..., 29)
        SCK . . . . . . . =  255  (SPI_SCK= 10, 14, 26)
        MOSI  . . . . . . =  255  (SPI_MOSI=11, 15, 27)
        MISO  . . . . . . =  255  (SPI_MISO=8, 12, 24, 28)
Output level High . . . . =  255  (HIGH = 0, 1, 2, ..., 29)
Output level Low  . . . . =  255  (LOW  = 0, 1, 2, ..., 29)

Esc type is not defined

Protocol is Sport (Frsky)
CRSF baudrate   = 420000
Logger baudrate = 115200
PWM is generated at = 50 Hz
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611 at I2c adress 77
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5
Airspeed sensor is detected using XGZP....
    No Vspeed compensation channel defined; oXs uses default settings
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox (configured by oXs) :GPS is not (yet) detected
Led color is normal (not inverted)`

mstrens, 2024-04-06:
What do you mean with "calibration"?

At power on oXs makes several (32) measurements and uses the average as offset.
So this is an automatic calibration.

jasc76, 2024-04-19:
ok 
I get now also Airspeed. 
But I guess you need to show the absolute value, because I get neg values :-)

processing cmd

Cmd to execute:   FV

Vspeed = 3 cm/s
Baro Rel altitude = 24 cm
Airspeed = -1235 cm/s
Compensated Vspeed = -1529 cm/s
pwmTop= 20000
rawPres=-733749  sumPa=-86.113831  count=1
rawPres=-1275134  sumPa=-152.200867  count=1
frawPres=-22982  sumPa=0.649719  count=1
v
processing cmd

Cmd to execute:   FV

Vspeed = 1 cm/s
Baro Rel altitude = 30 cm
Airspeed = 28 cm/s
Compensated Vspeed = 187 cm/s
pwmTop= 20000
rawPres=-34739  sumPa=-0.785461  count=1
rawPres=-304483  sumPa=-33.713196  count=1
rawPres=-2581819  sumPa=-311.708313  count=1
fvrawPres=-3587696  sumPa=-434.496033  count=1

processing cmd

Cmd to execute:   FV

Vspeed = 1 cm/s
Baro Rel altitude = 32 cm
Airspeed = -2541 cm/s
Compensated Vspeed = -1600 cm/s
pwmTop= 20000
rawPres=-3573326  sumPa=-432.741882  count=1
rawPres=-27387  sumPa=0.112000  count=1
rawPres=-27745  sumPa=0.068298  count=1
rawPres=-18979  sumPa=1.138367  count=1
rawPres=-11486  sumPa=2.053040  count=1
rawPres=-12270  sumPa=1.957336  count=1
rawPres=-11181  sumPa=2.090271  count=1

mstrens, 2024-04-19:
I use negative values because they can help during the calibration process (to calculate offset)
Do you test with a pitot tube or just applying a pressure on one of the entry of the sensor?
With a pitot tube, you should try to invert the 2 tubes (this should invert the sign.