mstrens/oXs_on_RP2040 #131: Gyro Subsystem

doloebig, 2024-04-16:
Hallo,
i try to activate the Gyro Subsystem on Zerro. From point off view everything is setup but there is noch movement
Version = 2.11.6 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =    1  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =    2    3    4    6 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255  255  255 (V1 / V4 = 26, 27, 28, 29)
RGB led . . . . . . . . . =   16  (RGB    = 0, 1, 2, ..., 29)
Logger  . . . . . . . . . =  255  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =  255  (ESC_PIN= 0, 1, 2, ..., 29)Esc type is not defined

Protocol is CRSF (=ELRS)
CRSF baudrate   = 420000
Logger baudrate = 115200
PWM is generated at = 50 Hz
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is not detected
Airspeed sensor is not detected
    No Vspeed compensation channel defined; oXs uses default settings
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox (configured by oXs) :GPS is not (yet) detected
Led color is normal (not inverted)
Failsafe type is HOLD
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = -42 , 119 , 2582
     Gyro offsets         X, Y, Z = -46 , 92 , 28
     Orientation          Horizontal is UP(Z+)     Vertical is BACK(X-)

Gyro configuration is:
   Channels for :                     mode/gain=10  ,  Ail stick=1  ,  Elv stick=2  ,  Rud stick=4
   Gain per axis (-128/127):                           Roll=127          Pitch=127         Yaw=127
   Gain on throw :                    1    (1=on full throw, 2=on half, 3=on quater)
   Max rotate    :                    3    (1=Very low , 2=low , 3=medium , 4=high)
   Stick rotate enabled in rate mode: 1    (1=disabled , 2=enabled)
   Stabilize mode :               ON    (Hold mode is disabled)
    PID         ---Roll(aileron)---         --Pitch(elevator)--        ----Yaw(rudder)----
   Mode          Kp      Ki      Kd          Kp      Ki      Kd         Kp      Ki      Kd
   Normal  PIDN= 500     0       500         500     0       500        500     0       500  
   Hold    PIDH= 500     500     500         500     500     500        500     500     500  
   Stab.   PIDS= 500     0       500         500     0       500        500     0       500  

Gyro mixers are calibrated:
Sticks centered at:    Ail=0%   Elv=1%   Rud=0%
Gyro corrections (from center pos in %) on:

No sequencers are defined

Config parameters are OK
Press ? + Enter to get help about the commands

Regards
Doro

mstrens, 2024-04-16:
The way gyro is implemented in oXs is a little unusual.
There is no need to manually define mixers/limits in the receiver/gyro because those are automatically detected during a "learning" process.
Still to achieve this, you have to let the handset transmit stick positions (for Ail, Elv and Rud).
Imagine that when you do not use the gyro, your setup would be:
Ail = Channel 1 = oXs gpio 2 => PWM to AIL servo
ELV = Channel 2 = oXs gpio 3  => PWM to ELV servo
Rud = Channel 4 = oXs gipo 6  => PWM to Rud servo

In order to use the gyro you also have to let the handset transmit the original stick positions on separate channels.
So in the handset you should add some mixers to transmit also 
Ail stick position on e.g. channel 11
ELV stick position on e.g. channel 12
Rud stick position on e.g. channel 13
Best is to exclude trim, expo in those mixers (if you handset allow it).
You also have to transmit on channel 10 (in your case) a switch (to select the gyro mode Normal/off, stabilize)). Depending on the position of the switch, the transmitted value on this channel should be positive (e.g. +100%), 0, negative (e.g. -100%).
During the "learning" process, oXs will discover that when channel 11 is at max (stick in a corner), then channel 1 is e.g. at position -80%. oXs will take care of this to apply the gyro corrections.

Be careful when you use the gyro functionalities. I used it and it seems to work but I made only a few tests.

aeropic, 2024-04-16:
I use it in NORMAL or HOLD mode both work very well. Be very careful with STABILIZE... :-)

doloebig, 2024-04-16:
In order to use the gyro you also have to let the handset transmit the original stick positions on separate channels.
So in the handset you should add some mixers to transmit also
Ail stick position on e.g. channel 11
ELV stick position on e.g. channel 12
Rud stick position on e.g. channel 13

################################################
where this chanels need to configured on the RP2040?
 
In my case i have 5 Mixers.
AIL channel 1
ELV channel 2
Thro channel 3
Rud channel 4
Gyro changel 10
no ad. mixer all +-100% no subtrim

on the zerro
GMG = 10
AIL=1
ELV=2
Rud=4

aeropic, 2024-04-16:
you have to set them here :
Gyro configuration is:
Channels for : mode/gain=10 , Ail stick=1 , Elv stick=2 , Rud stick=4

send th GSA GSE GSR commands:
Gyro Stick Aileron      GSA = YY            Gyro only : Original aileron stick (without mix/limits/trim)
    Gyro Stick Elevator     GSE = YY                                elevator 
    Gyro Stick Rudder       GSR = YY                                rudder

mstrens, 2024-04-16:
keep the commands
C1=2   AIL
C2=3   ELV
C3=4  THro
C4=6   Rud

Add e.g. commands
GSA=11
GSE=12
GSR=13
This setup supposes that in your handset you also add some mixers in order to let 
Aileron stick generate a signal on channel 11 with a range -100%/+100% and NO trim, no expo
Elevator stick generate a signal on channel 12 with a range -100%/+100% and NO trim, no expo 
Rudder stick generate a signal on channel 13 with a range -100%/+100% and NO trim, no expo
If you want, you can use other channels than 11,12,13.
Those channels and channels 10 do not have to generate a PWM output (so no need to have command like C11=x, C12=y, C13=z;
Then apply:
- the mpu calibration to let oXs know the orientation of the mpu in the model
- the learning process to let oXs know the mixers to output the PWM signals for ail, elv, rud.
Did you read the documentation that is in the doc forlder about the gyro?
If it is still not clear, just ask here or on rcgroups forum.

mstrens, 2024-04-16:
Those messages (Write error for MPU6050) are strange.
It looks like a hardware defect.
Do you use long wires between MPU6050 and RP2040?

doloebig, 2024-04-16:
12-15cm

mstrens, 2024-04-16:
This is perhaps too long.
Can you reduce the length (E.g. 2 or 3 cm)?

mstrens, 2024-04-18:
Reset means just power OFF  and ON again.
Nuke erase all flash. So it is normally never used.

After reset, please just press ENTER and oXs should display the config that is saved in memory.
This should allow to check if everything is OK.
Post here a copy/paste of what is displayed.

doloebig, 2024-04-18:
i know the def. nuke and reset.
On the new version the led ist constant on, the old version was flashing
my fault
will try to calibrate tomorrow
Thx for now
Doro

doloebig, 2024-04-19:
get this after cal. H and V
MPUCAL =h
processing cmd

Cmd to execute:   MPUCAL=H
Before calibration:
Offset Values in config:
Acc. X = 0, Y = 0, Z = 0
Gyro. X = 0, Y = 0, Z = 0
Upper face is BACK(X-)
Acc & gyro after new gyro calibration

Offset Values in config:
Acc. X = 0, Y = 0, Z = 0
Gyro. X = -59, Y = 97, Z = 61
Horizontal calibration done: use SAVE command to save the config!!
save
processing cmd

Cmd to execute:   SAVE
size of config is 244
Write error for MPU6050
config has been saved
Device will reboot but it could be that a reset or a (power down + power on) is required

crsfRxQueue try add error
crsfRxQueue try add error
crsfRxQueue try add error
crsfRxQueue try add error
crsfRxQueue try add error
crsfRxQueue try add error
crsfRxQueue try add error
crsfRxQueue try add error
crsfRxQueue try add error
crsfRxQueue try add error
crsfRxQueuecrsfRxQueue try add error
crsfRxQueue try add error

Total config
processing cmd

Version = 2.13.3 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =    1  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =    2    3    4    6 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255  255  255 (V1 / V4 = 26, 27, 28, 29)
RGB led . . . . . . . . . =   16  (RGB    = 0, 1, 2, ..., 29)
Logger  . . . . . . . . . =  255  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =  255  (ESC_PIN= 0, 1, 2, ..., 29)
Locator CS  . . . . . . . =  255  (SPI_CS = 0, 1, 2, ..., 29)
        SCK . . . . . . . =  255  (SPI_SCK= 10, 14, 26)
        MOSI  . . . . . . =  255  (SPI_MOSI=11, 15, 27)
        MISO  . . . . . . =  255  (SPI_MISO=8, 12, 24, 28)
Output level High . . . . =  255  (HIGH = 0, 1, 2, ..., 29)
Output level Low  . . . . =  255  (LOW  = 0, 1, 2, ..., 29)

Esc type is not defined

Protocol is CRSF (=ELRS)
CRSF baudrate   = 420000
Logger baudrate = 115200
PWM is generated at = 50 Hz
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is not detected
Airspeed sensor is not detected
    No Vspeed compensation channel defined; oXs uses default settings
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox (configured by oXs) :GPS is not (yet) detected
Led color is normal (not inverted)
Failsafe type is HOLD
Acc/Gyro is detected using MP6050
     Acceleration param: ACC= 0.000000 0.000000 0.000000
                              1.000000 0.000000 0.000000
                              0.000000 1.000000 0.000000
                              0.000000 0.000000 1.000000
     Gyro offsets         X, Y, Z = -59 , 97 , 61
     Orientation          Horizontal is BACK(X-)     Vertical is BACK(X-)

Gyro configuration is:
   Channels for :                     mode/gain=10  ,  Ail stick=11  ,  Elv stick=12  ,  Rud stick=13
   Gain per axis (-128/127):                           Roll=127          Pitch=127         Yaw=127
   Gain on throw :                    1    (1=on full throw, 2=on half, 3=on quater)
   Max rotate    :                    3    (1=Very low , 2=low , 3=medium , 4=high)
   Stick rotate enabled in rate mode: 1    (1=disabled , 2=enabled)
   Stabilize mode :               ON    (Hold mode is disabled)
    PID         ---Roll(aileron)---         --Pitch(elevator)--        ----Yaw(rudder)----
   Mode          Kp      Ki      Kd          Kp      Ki      Kd         Kp      Ki      Kd
   Normal  PIDN= 500     0       500         500     0       500        500     0       500  
   Hold    PIDH= 500     500     500         500     500     500        500     500     500  
   Stab.   PIDS= 500     0       500         500     0       500        500     0       500  
Gyro mixers must be calibrated

No sequencers are defined
Error in parameters: gyro horizontal (1) and vertical (1) orientations are not compatible

Config parameters are OK
Press ? + Enter to get help about the commands

mstrens, 2024-04-19:
Thanks for providing your config.
It shows that there are still 2 issues.
First is given with "Error in parameters: gyro horizontal (1) and vertical (1) orientations are not compatible
To solve this, you have to run a command mpucal=V.
The principle is the following:
oXs has to know the orientation of the mpu6050 in the model in order to calculate roll and pitch.
There are 24 possible orientations.
To find the orientation, there are 2 steps required.
First is done with command mpucal=H. You did it (take care to keep the model horizontal and still just before running the command!) . After this test, oXs knows which axis (x, y or z) of the MPU6050 is vertically aligned and and if it is pointing up or down. Still at this stage, there are still 4 orientations possible and oXs can't make the difference between the roll and pitch axis.
Second command mpucal=V has to be done while the nose of the model is vertically up.

The second error you have is that "Gyro mixers must be calibrated"
After having solved the first issue, you must apply all the steps of the learning process" (sticks in right corners, elevator up corner, move gyro switch, ...).

Please do this and post here again your config.

Note: getting some error messages (like crsfRxQueue try add error , crsfRxQueuecrsfRxQueue try add error) when oXs saves the config is not abnormal. They must disappear after the reset.

doloebig, 2024-04-19:
i run in the next problem during "learning"
The learning process started
Rud right Ch1=1513   Ch2=1512    Ch3=996    Ch4=1492
Error in Gyro setup: missing all 3 sticks centered
Error in Gyro setup: missing one corner position for aileron alone
Error in Gyro setup: missing one corner position for elevator alone
Error in Gyro setup: missing one corner position for rudder alone
Error :  oXs will stop handling receiver and sensors; power off/on is required !!!!!!

you do a great job on the oXt and have 6 zerros running als telemtrie dongle also i like to use as ELRS PWM dongle. But the gyrosubsystem is to komplex for me, i will use a commercial product. THX for patience and sorry for waisting you time.

Doro

closed 2024-04-19 by doloebig