mstrens/oXs_on_RP2040 #132: Failsafe uses predefined values do not work on Version 2.12.0

meyer1111, 2024-04-25:
When I switch off the transmitter, the sevos do not move to predefined values.
The outputs are in HOLD mode.
Have I made a mistake?

My setup: ELRS & RP2040
------------------------------------------
Version = 2.12.0 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    9  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    8  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =    0    1    2    3 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255    4    5    6
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26  255  255  255 (V1 / V4 = 26, 27, 28, 29)
RGB led . . . . . . . . . =   16  (RGB    = 0, 1, 2, ..., 29)
Logger  . . . . . . . . . =  255  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =  255  (ESC_PIN= 0, 1, 2, ..., 29)
Locator CS  . . . . . . . =  255  (SPI_CS = 0, 1, 2, ..., 29)
        SCK . . . . . . . =  255  (SPI_SCK= 10, 14, 26)
        MOSI  . . . . . . =  255  (SPI_MOSI=11, 15, 27)
        MISO  . . . . . . =  255  (SPI_MISO=8, 12, 24, 28)
Esc type is not defined
Protocol is CRSF (=ELRS)
CRSF baudrate   = 420000
Logger baudrate = 115200
PWM is generated at = 50 Hz
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using BMP280
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Airspeed sensor is not detected
    No Vspeed compensation channel defined; oXs uses default settings
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox (configured by oXs) :GPS is not (yet) detected
Led color is normal (not inverted)
Failsafe uses predefined values
     Chan 1...4  =   988  1498  1605  1552
     Chan 5...8  =  1000  1550  1045  1923
     Chan 9...12 =  1500   881   881   881
     Chan 13...16=   881   881  1992  2012
Acc/Gyro is not detected
Gyro is not configured
No sequencers are defined
Config parameters are OK

mstrens, 2024-04-25:
I do not see an error in your config.
I checked the code from the last version of oXs in test branch (2.13.6).
I did not see the bug but I do not have an ELRS receiver currently to make a real test.

Could you try to upload version 2.13.6. Let me know if this solve the issue.
If it still does not work, I could make a version adding some debug messages to the PC to find the bug.

mstrens, 2024-04-25:
I think I understand the issue.
I will try to solve it for next day.
So, no need to make new test currently.

meyer1111, 2024-04-29:
Have 2.14.2 tested.
No improvement.

RF connected: LED flashes green.
Transmitter off: LED flashes blue.

Failsafe uses predefined values do not work.

meyer1111, 2024-04-29:
Very good! Congratulations!
And thank you!

closed 2024-04-29 by meyer1111