mstrens/oXs_on_RP2040 #133: Is it possible to add G-Force Sensor GY-291 ADXL345 for MPX, HoTT, ELRS protocol to RP2040

Astor87, 2024-05-25:
Hi mstrens, 
sorry it is not an issue rather than another Add-on to your marvelous oXs-on-RP2040 project.
I have build quite some qty for friends mainly with Alt/Vario MS5611 and GPS Breitian BN2020 for our gliders.
My question is, is there a possibility to connect an GY-291 ADXL345 G-Force Sensor to the RP2040 via SDA, SCL and log G-Forces during flight similar to a G-Force measurement in human piloted Gliders Fox, Swift etc. 
The cheap GY-291 ADXL345 is capable upto +/- 16G (!!)   
Would be great to connect it backpacked with an RP2040 via HoTT, MPX, ELRS protocol to the relevant telemetry RX and read (max + plus) and (max - negative)  G-Force after flight or (should not happen !) after an impact.
Thanks for a short review of my thoughts
And further many thanks for your superb project 
BR
Wolfram
-ASTOR87-

mstrens, 2024-05-25:
I suggest that you use a KX134 sensor.
It supports up to 64g and is already managed by oXs (with the version in test branch).

Please note that if you look at the values logged on the handset, you will probably not have the real max/min because not all values are transmitted to the handset (because the bandwidth for telemetry data is quite limitted and accelerations change very quickly). In order to get more data, you can build an oXs logger that will store the data directly on a sd card at a much higher rate. This (oXs logger)  is a project on my github too. It requires to use another RP2040 connected to oXs.

Astor87, 2024-05-25:
Thanks a lot for your quick answer!
I thought about GY–291  ADXL345 because it is just at arround 2US$  and with upto +/– 16G is already quite good range.
This was used by a JETI DIY G–Force sensor when packed on an arduino mini or nano.  Ohne reason for my raised question.
Your mentioned KX134 is significant  higher priced. Costs are above an original complete MPX G–Rate Sensor I would normaly use but RP2040 is my favourite!! 
However I will have a review on your SW in your testbench !
Anyway many many thanks for your superfast response and I will check on your oXs logger project!  You are doing an increddible great Job !! 
I am still using many times / flights your last years addon: cumulated distance during flight on MPX protocol!! 
BR Wolfram