mstrens/oXs_on_RP2040 #136: putty interface not working

gugr05, 2024-07-16:
Hi
I have been trying to get the program working but it wont respond, pls can someone help!
I followed the steps to the important note but then i didn't get a response for pressing enter.

mstrens, 2024-07-17:
Several users had also issues with putty.
Can you try another serial terminal program.
The arduino ide serial terminal works. The Visual code serial monitor works also.
I tested also with cooltermwin that does not require an installation (just unzip and run the exe).

In all cases, you have to activate an option that add automatically CR+LF at the end of each command.
For coolterm, this option is on a tab transmit (in the settings icon); it is named "Send string option".
To send commands in coolterm, select the "send string..." item in the menu connexion.

gugr05, 2024-07-17:
THX for a fast response, and it worked i just had to send CR+LF manually by pressing control M and control J. Tip add it to the readme for a easy start for noobs like me :/.

mstrens, 2024-07-17:
some serial terminal programs can add automatically CR+LF when you press ENTER