mstrens/oXs_on_RP2040 #137: GUI remark

Merga189, 2024-07-27:
Playing a bit with the GUI.

Loaded the config from my yellow Delta here. 

That is the config in a terminal

[Pibros oXs RP2040 2.14.13.txt](https://github.com/user-attachments/files/16401247/Pibros.oXs.RP2040.2.14.13.txt)

and that is the saved file generated by the GUI here

[Pibros.ini.txt](https://github.com/user-attachments/files/16401251/Pibros.ini.txt)

Several questions come up, but I would like to start with that one here.

Want to transfer Pibros.ini to a second RP2040, so I 'load the config from the pc' and have a look at it in the GUI, but I do see this here

[image]

'Use gyro' is NOT selected, but pins are set. 

Could that be a fault??

br Karl

mstrens, 2024-07-27:
It was a bug.
I just saw it and normally fix it.
New version is on github but I forgot to update the version nummer (that appears in the gui main window)

Merga189, 2024-07-27:
That was (as usual ;-)) fast. I'll keep on testing with the GUI and come back here, not in RCG.

mstrens, 2024-07-27:
Ok thanks.

There are probably some other bugs because I had quite many changes in order to add the new fields and I had a lot of copy/paste.

About gyro, I did not yet implemented the parameters about the PID's. I can do it later on.

mstrens, 2024-07-28:
I also fixed this morning a bug about inverted led when loading oXs config
Now version is 0.0.3

Merga189, 2024-07-28:
At the moment I am puzzled what is happening here.

Just a simple test on a breadboard. RP2040 and SPL-06.

On (Linux) terminal I read 

[Simple #1.txt](https://github.com/user-attachments/files/16402793/Simple.1.txt)

In the GUI 003 I see this here

[image]

Then I save the settings on the desktop here, I get this

[Simple #1a.ini.txt](https://github.com/user-attachments/files/16402800/Simple.1a.ini.txt)

and when I go into that window and select 'load from oXs' the GUI crashes

[image]

which those messages in terminal here

[crash.txt](https://github.com/user-attachments/files/16402820/crash.txt)

And this crash I can reproduce ....

Quite confusing for me at the moment.

Karl

mstrens, 2024-07-28:
There was still a bug for inerted led.
I hope it is now OK in version 0.0.4 (on github)

Merga189, 2024-07-28:
I'll test 0.0.4 later, but let me ask ....

I nuked the RP2040 and loaded 3.0.6/0.0.3

[image]

Now I can make the settings in the GUI for that simple setup. ELRS 420000, pin 9,10, I2C pin 14,15. But I am not seeing a way to directly get those settings in the GUI onto the connected RP2040??

I could save the 'default' (nuked) settings onto the PC (*.ini), modify the values with an editor there and then load the modified settings with the GUI back to the RP2040.

But is there a direct way? Not seeing the way (-;

Karl

Merga189, 2024-07-28:
Have several remarks, but I think the best is to show those points in a short video.

https://youtu.be/atZTSVZ3s3Q

Start with no.1 - Desktop with 'oXs_on_RP2040-test 004', 'New File' is the setting done in terminal and 'test.ini' for saving the setting from the GUI.

I open the folder, open the sub folder 'gui' and start the GUI with a short bash script.

Start windows  'Config' and I switch to 'USB commands'. Connect to the RP2040, 'Display current **correct** configuration and the setting shows up!

Then switch to 'Receiver Handset' and this windows shows me a 'Default' setting, BUT NOT THE current config I just saw in the USB windows.

For me personally that is a bit strange. To get the actual values from the RP2040 I have to go into 'Config' and load those by pushing a button.

THAT is a little confusing in my eyes. Connect, values show up in the GUI, but obviously the values are not in the GUI and it needs a second step.  Maybe it is better, if NO VALUES at all show up a this 1.st stage, when 2 stages for getting the values into the GUI are required. 

Hope it is OK for you to do it in that way with a short video.

Karl

mstrens, 2024-07-29:
Thanks for the comments.

In fact this was the expected behaviour.

For the GUI, I did not make changes to oXs. I just try to reuse some usb commands that oXs already supports (at least latest versions).

oXs can provide his setup with 2 commands
- ENTER : it provides all setups in clear text with some details on the code to use to change some parameters. 
- DUMP: it provides only the setups that are different from the default. The format is the same as usb command and so it is possible to use it for direct copy/paste.

The first command is very difficult to decode in order to preset the GUI. It is still very usefull because it gives all setup in clear text and all errors detected by oXs.
To preset the GUI, I use only the second command (when the user press "Load config from oXs" button.

To preset the GUI with oXs setup, you have just to press this button (after having establish the connection).

I just made some changes in version 0.0.5
- in order to move the buttons from the config tab to the usb command tab
- add a button that directly upload and save the config into oXs. In previous version, you had to let the GUI generate the command to upload the setup into oXs, then to send this command. Finally, you had also to send a SAVE cmd to store definitively the setup into oXs. This option still exist but I now add a button to do all the actions.
Still please note that the after a SAVE, the serial port is disconnected (because oXs perform a reset). So it is usually better practice to first send the setup commands (to upload parameters), then send ENTER (to check for errors) and only when there is no errors anymore, then send the SAVE cmd. I could perhaps add such buttons.

Merga189, 2024-07-29:
Thank you for the comprehensive answer. 

Things clear a little up here. So the GUI will be **also** a complete substitute for the terminal and it will cover other features?!

The information when using 'DUMP' and "Load config from oXs"  is on first look from the content identical, but a little different displayed. What is the use of two commands for one information?

For your information, to get your python program running here under Debian 12, I used only those Debian packages for installation. 

python3-pyqt6
pyqt6-dev
pyqt6-dev-tools
python3-serial
python3-pyqt6.qtserialport

The problem was, that at start everything pointed in the direction of using 'pip' and with actual Linux distros 'pipx' and that way things got messed up ....

mstrens, 2024-07-29:
Yes gui is a complete substitute for the terminal.
It does not support other features.

The advantage of DUMP is that the response from oXs can be used with copy/paste for sending commands to oxs (in other device or after some edit.

I made some more minor changes in 0.0.6 to the usb command tab. Hope it is clearer.

closed 2024-07-29 by Merga189