mstrens/oXs_on_RP2040 #143: GPS and I2C sensors issue

CodeWorksCZ, 2024-08-20:
Hello, 
I have got a small PCB board which has GPS(BN220)  connected  and it works fine 

```
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox (configured by oXs) :GPS is detected but has not (yet) a fix
Led color is inverted
```
 
There are also sensors connected to it
**BME280, MPU6050, XGZP, ADS1115.** 

They also work **but as soon as I  connect BN220** and  power on oXs , serial monitor is clogged by these messages:

**Write error for BMP280: -1
Write error ads1115
Write error for MPU6050
Write error for BMP280: -1
Write error ads1115Write error for MPU6050**

It only happens when  all sendors and gps are connected o oXs

My susspision falls on GPS handling which somehow  breaks  i2c bus

Have anyone encountered this behaviour?

- 3.0.6 version
- same happens on old 2.8. version. 

[Parameters inc. debug without GPS connected]

```
 DEBUG

Config parameters are OK
Press ? + Enter to get help about the commands
Trying to detect MS5611 sensor at I2C Addr=76
error in CRC of calibration for MS5611
Trying to detect MS5611 sensor at I2C Addr=77
error write reset MS5611: -2
SPL06 has wrong device id
write error spl06
Trying to detect BMP280 sensor at I2C Addr=76
BMP280  sensor is successfully detected
ADS1115 sensor I2C Addr=48
Set up Ads1115 done and OK.
ADS1115 sensor I2C Addr=49
Write error for ads1115
Error in Set up of ads1115.
Trying to detect MPU6050 sensor at I2C Addr=68
error read MS4525 (airspeed sensor) at startup: -2 
error write command to sdp3x: -2
Setup1 takes 149333 usec
Automatic gyro calibration failed; oXs uses gyro offsets saved during horizontal calibration
rawPres=-34370  sumPa=1.637825  count=1
rawPres=-33166  sumPa=1.931770  count=1
rawPres=-34738  sumPa=1.547981  count=1
rawPres=-34327  sumPa=1.648323  count=1
rawPres=-32170  sumPa=2.174934  count=1
---- Sent utf8 encoded message: "\r\n" ----
processing cmd

Version = 3.0.6 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =  255  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =    5  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =    4  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =    2  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =    3  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255  255  255 (V1 / V4 = 26, 27, 28, 29)
RGB led . . . . . . . . . =   16  (RGB    = 0, 1, 2, ..., 29)
Logger  . . . . . . . . . =  255  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =  255  (ESC_PIN= 0, 1, 2, ..., 29)
Locator CS  . . . . . . . =  255  (SPI_CS = 0, 1, 2, ..., 29)
        SCK . . . . . . . =  255  (SPI_SCK= 10, 14, 26)
        MOSI  . . . . . . =  255  (SPI_MOSI=11, 15, 27)
        MISO  . . . . . . =  255  (SPI_MISO=8, 12, 24, 28)
        BUSY  . . . . . . =  255  (SPI_BUSY=0, 1, 2, ..., 29)
Output level High . . . . =  255  (HIGH = 0, 1, 2, ..., 29)
Output level Low  . . . . =  255  (LOW  = 0, 1, 2, ..., 29)

Esc type is not defined

Protocol is Sport (Frsky)
CRSF baudrate   = 420000
Logger baudrate = 115200
PWM is generated at = 50 Hz
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
**Baro sensor is detected using BMP280 at I2c adress 76**
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
**Airspeed sensor is detected using XGZP....**
    No Vspeed compensation channel defined; oXs uses default settings
**First analog to digital sensor is detected using ads1115**
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox (configured by oXs) :GPS is not (yet) detected
Led color is inverted
Failsafe type is HOLD
**Acc/Gyro is detected using MP6050**
     Acceleration param: ACC= 0.000000 0.000000 0.000000
                              1.000000 0.000000 0.000000
                              0.000000 1.000000 0.000000
                              0.000000 0.000000 1.000000
     Gyro offsets         X, Y, Z = 0 , 0 , 0
     Orientation          Horizontal is UP(Z+)     Vertical is FRONT(X+)

Gyro is not configured (Rc channel used to select the gyro mode/gain is not defined)

No sequencers are defined
```

CodeWorksCZ, 2024-08-27:
Thanks Torsten for your answer. I need to dig deeper. 
I have also ordered another GPS module to see if the one I have is not faulty.

THis is a custom PCB i designed and this time I  added MPU6050 but not as a GY board module but  as on board components.

Previous version worked but did not have MPU and XGZP module. 

I am glad that your "full house" works. At least I know that I have look at the hardware  aspect.

Michal

romoloman, 2024-09-02:
My suspect looking to your hardware  (the oxs stack) is that you rely too much on the LDO of the raspaberry,
The raspberry ldo only supports very low input voltage (5.5V) and very low current. Your stack may ask too much.
I suggest you to add an additional LDO like AMS1117-3.3 to supply enough current not regarding INPUT voltage.

CodeWorksCZ, 2024-09-02:
Thanks @romoloman. I though about that as well and I even tried to power up the GPS directly from 3.3V bread board  and no luck.

I made a small test rig using POGO pins to test both PCB before soldering them together. I am sure I will crack it soon or later.
In the future version I might add this LDO if i find some space on the board

Michal

romoloman, 2024-09-02:
I also have designed my own board to host some sensor but I have choose a different approach from your:
[image]
[image]
I do not have cell measuring but I have optical RPM sensor, current (using a WCS1500) and temperature, plus online variometer (MS5611) and GPS, but if you look to the board there is the LDO