mstrens/oXs_on_RP2040 #147: Add field RPM and Temperature for ELRS protocol

Gjeremie, 2024-09-17:
Hello,

first thanks for your work.

Is it possible to add option in the config.h file in order to replace  the GPS fields by the RPM and Temperatures.

I know ELRS doesnt transmit these fields.
But I dont need altitude or Vspeed or GPS data, but VOLT3 for motor and VOLT4 for esc temperature and RPM will be interesting

I see in older release you do this with RPM

thank you

Gjeremie, 2024-09-18:
I've just checked and in the GPS data what fields we have:
https://github.com/crsf-wg/crsf/wiki/CRSF_FRAMETYPE_GPS

- We could use the GPS altitude (unsigned integer 16 bit, up to 65000 is enough) **for RPM**
- And the  ground speed in km/h ((integer 16 bit / 10, up to 3200 is enough) **for Temperature**

romoloman, 2024-09-19:
I think is not a good idea, 
ELRS already foreseen additional sensors but nothing is standard like you can see in rotorflight https://github.com/rotorflight/rotorflight-firmware/issues/81
It would be better a standardization on ELRS/CLRS side instead of an hack on oxs one.
After a standardization is done on ELRS side then adding the proper field on oxs would be tricky...