mstrens/oXs_on_RP2040 #149: GPS route in FRSKY always shows 0,.. does not change the value.

Matorra, 2024-10-04:
I have built one snsor whit a GPS M10 series, and I have connected it to a FRSKY receiver and Tx X20S. The data of Lat, Lon, Alt, Speed ​​work correctly but the route always marks zero. Does anyone have experience with this?

Satcomix, 2024-10-05:
> I have built one snsor whit a GPS M10 series, and I have connected it to a FRSKY receiver and Tx X20S. The data of Lat, Lon, Alt, Speed ​​work correctly but the route always marks zero. Does anyone have experience with this?

Hello Matorra,
Here is my oXs_RP2040 Version with Beitian BE250 M10 with FrSky X20S and TD-R10 with FBUS Protocol.
I can see every telemetry sensor and the values on my X20S, that are in the fields of protocol.txt for the S.Port/FBUS protocol.
https://github.com/mstrens/oXs_on_RP2040/blob/test/doc/fields%20per%20protocol.txt
If you want to show gps data on the display with a gps map widget in ETHOS with this one: https://ethosmap.hobby4life.nl/
you have to look in the instructions of the downloaded map file and enter the following coordinates in the ETHOS GPS Map widget with double N (north) data and double E (east) data for example.
best regards,
Torsten
processing cmd

Version = 3.0.0 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =   21  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    4  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255    7  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)
RGB led . . . . . . . . . =   18  (RGB    = 0, 1, 2, ..., 29)
Logger  . . . . . . . . . =  255  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =  255  (ESC_PIN= 0, 1, 2, ..., 29)
Locator CS  . . . . . . . =  255  (SPI_CS = 0, 1, 2, ..., 29)
        SCK . . . . . . . =  255  (SPI_SCK= 10, 14, 26)
        MOSI  . . . . . . =  255  (SPI_MOSI=11, 15, 27)
        MISO  . . . . . . =  255  (SPI_MISO=8, 12, 24, 28)
        BUSY  . . . . . . =  255  (SPI_BUSY=0, 1, 2, ..., 29)
Output level High . . . . =  255  (HIGH = 0, 1, 2, ..., 29)
Output level Low  . . . . =  255  (LOW  = 0, 1, 2, ..., 29)

Esc type is not defined

Protocol is Fbus(Frsky)
CRSF baudrate   = 420000
Logger baudrate = 115200
PWM is generated at = 50 Hz
Voltage parameters:
    Scales : 10.000000 , 10.000000 , 0.100000 , 0.100000 
    Offsets: 0.000000 , 0.000000 , 1.500000 , 2.000000 
    Temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611 at I2c adress 77
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Aispeed sensor is detected using MS4525
    Vspeed compensation channel = 7
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Foreseen GPS type is Ublox (configured by oXs) :GPS is detected and has a fix
Led color is normal (not inverted)
Failsafe uses predefined values
     Chan 1...4  =  1500  1500  1500  1500
     Chan 5...8  =  1500  1500  1500  1500
     Chan 9...12 =  1500  1500  1500  1500
     Chan 13...16=  1500  1500  1500  1500
Acc/Gyro is detected using MP6050
     Acceleration param: ACC= 0.000000 0.000000 0.000000
                              1.000000 0.000000 0.000000
                              0.000000 1.000000 0.000000
                              0.000000 0.000000 1.000000
     Gyro offsets         X, Y, Z = 4 , -19 , -24
     Orientation          Horizontal is UP(Z+)     Vertical is FRONT(X+)

Gyro is not configured (Rc channel used to select the gyro mode/gain is not defined)

No sequencers are defined

Config parameters are OK
Press ? + Enter to get help about the commands
processing cmd

Cmd to execute:   FV

GPS Latitude = XX.X224096 degree
GPS Longitude = X.X171192 degree
GPS Groundspeed = 12 cm/s
GPS Heading = 247.680000 degree
GPS Altitude = 8549 cm
GPS Num sat. = 106
GPS Date J M A = 5 10 24 
GPS Time H M S = 12 33 7 
GPS Pdop = 384 
GPS Home bearing = 219 degree
GPS Home distance = 28 m
Volt 1 = 6675 mVolt
Current (Volt 2) = 6813 mA
Capacity (using current) = 346 mAh
Temp 1 (Volt 3) = 23 degree
Temp 2 (Volt 4) = 23 degree
Vspeed = -3 cm/s
Baro Rel altitude = 198 cm
Pitch = 6.000000 degree
Roll = -7.000000 degree
RPM = 1007 Hertz
Ads 1 1 = 119 mVolt
Ads 1 2 = 119 mVolt
Ads 1 3 = 119 mVolt
Ads 1 4 = 119 mVolt
Ads 2 1 = 119 mVolt
Ads 2 2 = 119 mVolt
Ads 2 3 = 119 mVolt
Ads 2 4 = 119 mVolt
Airspeed = 168 cm/s
Compensated Vspeed = 35 cm/s
Gps cumulative distance = 43
Acc X = 0.212000g
Acc Y = 0.007000g
Acc Z = 0.813000g
Reserve 5 = 614
Vspeed compensation = 1.10
pwmTop= 20000

Matorra, 2024-10-12:
Thank you very much for your information, but my problem is that the value is always "0.1º" on the screen display during the flight, it does not change value according to the direction of the plane. The rest of the values ​​are correct, this makes me think that there are no problems with the sensor configuration. I think that the GPS is not working correctly. I will try with another GPS ( different model ) next week and I will tell you about the experience.

Satcomix, 2024-10-17:
> Thank you very much for your information, but my problem is that the value is always "0.1º" on the screen display during the flight, it does not change value according to the direction of the plane. [...]

Hallo Matorra,
Which value, telemetry sensor shows zero° or 0.1°? 
Have you a look at the "fields of protocol.txt" ducument?
Can please post a FV (Field View) from your oXs_RP2040 config. So i have a look to your telemetry sensors.
best regards,
Torsten

Matorra, 2024-10-17:
17:03:43.771 -> processing cmd
17:03:43.771 -> 
17:03:43.771 -> Version = 3.0.0 
17:03:43.771 ->     Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
17:03:43.771 -> Primary channels input    =  255  (PRI     = 5, 9, 21, 25)
17:03:43.771 -> Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
17:03:43.771 -> Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
17:03:43.771 -> GPS Rx  . . . . . . . . . =    4  (GPS_RX  = 0, 1, 2, ..., 29)
17:03:43.771 -> GPS Tx  . . . . . . . . . =    3  (GPS_TX  = 0, 1, 2, ..., 29)
17:03:43.771 -> Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
17:03:43.771 -> RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
17:03:43.771 -> SDA (I2C sensors) . . . . =   26  (SDA     = 2, 6, 10, 14, 18, 22, 26)
17:03:43.771 -> SCL (I2C sensors) . . . . =   27  (SCL     = 3, 7, 11, 15, 19, 23, 27)
17:03:43.771 -> PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
17:03:43.771 -> PWM Channels 5, 6, 7 ,8   =  255  255  255  255
17:03:43.771 -> PWM Channels 9,10,11,12   =  255  255  255  255
17:03:43.771 -> PWM Channels 13,14,15,16  =  255  255  255  255
17:03:43.818 -> Voltage 1, 2, 3, 4        =  255  255  255  255 (V1 / V4 = 26, 27, 28, 29)
17:03:43.818 -> RGB led . . . . . . . . . =   16  (RGB    = 0, 1, 2, ..., 29)
17:03:43.818 -> Logger  . . . . . . . . . =  255  (LOG    = 0, 1, 2, ..., 29)
17:03:43.818 -> ESC . . . . . . . . . . . =  255  (ESC_PIN= 0, 1, 2, ..., 29)
17:03:43.818 -> Locator CS  . . . . . . . =  255  (SPI_CS = 0, 1, 2, ..., 29)
17:03:43.818 ->         SCK . . . . . . . =  255  (SPI_SCK= 10, 14, 26)
17:03:43.818 ->         MOSI  . . . . . . =  255  (SPI_MOSI=11, 15, 27)
17:03:43.818 ->         MISO  . . . . . . =  255  (SPI_MISO=8, 12, 24, 28)
17:03:43.818 ->         BUSY  . . . . . . =  255  (SPI_BUSY=0, 1, 2, ..., 29)
17:03:43.818 -> Output level High . . . . =  255  (HIGH = 0, 1, 2, ..., 29)
17:03:43.818 -> Output level Low  . . . . =  255  (LOW  = 0, 1, 2, ..., 29)
17:03:43.818 -> 
17:03:43.818 -> Esc type is not defined
17:03:43.818 -> 
17:03:43.818 -> Protocol is Sport (Frsky)
17:03:43.818 -> CRSF baudrate   = 420000
17:03:43.818 -> Logger baudrate = 115200
17:03:43.818 -> PWM is generated at = 50 Hz
17:03:43.818 -> Voltage parameters:
17:03:43.818 ->     Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
17:03:43.818 ->     Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
17:03:43.818 ->     No temperature sensors are connected on V3 and V4
17:03:43.818 -> RPM multiplier = 1.000000
17:03:43.818 -> Baro sensor is not detected
17:03:43.818 -> Airspeed sensor is not detected
17:03:43.818 ->     No Vspeed compensation channel defined; oXs uses default settings
17:03:43.818 -> First analog to digital sensor is not detected
17:03:43.818 -> Second analog to digital sensor is not detected
17:03:43.818 -> Foreseen GPS type is Ublox (configured by oXs) :GPS is detected and has a fix
17:03:43.818 -> Led color is normal (not inverted)
17:03:43.818 -> Failsafe type is HOLD
17:03:43.818 -> Acc/Gyro (MP6050) is not detected
17:03:43.818 -> 
17:03:43.818 -> Gyro is not configured (Rc channel used to select the gyro mode/gain is not defined)
17:03:43.818 -> 
17:03:43.818 -> No sequencers are defined
17:03:43.818 -> 
17:03:43.818 -> Config parameters are OK
17:03:43.818 -> Press ? + Enter to get help about the commands
17:06:46.999 -> processing cmd
17:06:46.999 -> 
17:06:46.999 -> Cmd to execute:   FV
17:06:46.999 -> 
17:06:46.999 -> GPS Latitude = 40.6438560 degree
17:06:46.999 -> GPS Longitude = -3.1491294 degree
17:06:46.999 -> GPS Groundspeed = 78 cm/s
17:06:46.999 -> GPS Heading = 33.310000 degree
17:06:46.999 -> GPS Altitude = 72063 cm
17:06:46.999 -> GPS Num sat. = 115
17:06:46.999 -> GPS Date J M A = 17 10 24 
17:06:46.999 -> GPS Time H M S = 15 6 48 
17:06:46.999 -> GPS Pdop = 145 
17:06:46.999 -> GPS Home bearing = 94 degree
17:06:46.999 -> GPS Home distance = 15 m
17:06:46.999 -> Gps cumulative distance = 131
17:06:46.999 -> pwmTop= 20000

Indeed here you can see that it marks a value different from zero ,  but the telemetry seen in X20s always marks zero or as it has decimals a value of zero with decimals.

Matorra, 2024-10-18:
This is what the data bus for the route looks from the transmitter. It seems like it should be multiplied by 100 to be correct.
The Tx divides by 100 and that's why I only see small values.

Segundo - Referencia-   binario-   decimal
15-0840-00000000000000000000000000011101-29
14-0840-00000000000000000000000000011101-29
14-0840-00000000000000000000000000011101-29
14-0840-00000000000000000000000000011101-29
14-0840-00000000000000000000000000011101-29
13-0840-00000000000000000000000000011101-29
13-0840-00000000000000000000000000011101-29
13-0840-00000000000000000000000000011101-29
12-0840-00000000000000000000000000011101-29

On the other hand, it is a shame that the values ​​of    " PS Home bearing = 94 degree"
and  " GPS Home distance = 15 m"     are sent without decimals, they would be more interesting given that, for example, at 400 meters one degree is a lot of meters. Perhaps they could be sent with two decimals. The integer of the value multiplied by 100.

romoloman, 2024-11-02:
I suppose is a similar problem as for sport https://github.com/mstrens/oXs_on_RP2040/pull/97 in my fork i have fixed s-port if you want to test I can try to fix also FBUS.

Matorra, 2024-11-06:
Thank you very much, I had not seen your message until now. I am going to fly today but I will try it tomorrow and confirm the result.