mstrens/oXs_on_RP2040 #151: Problem GPS

antlerhanger, 2024-11-20:
Having problem with GPS ...RX=8 , TX=7 ..I can hook up Matek gps and it works fine .. I hook up A HGLRC m100 mini and I get not detected ..I can't figure out what is going on ..I tried several Matek and they all work . I tried several hglrc m100 mini and none work . I can hook the hglrc up to a matek fc and get telem. from it so I know they work .I let itrun for over 30 minutes to make sure it had Sats. baud rate according to hglrc is 115200bps..M10 chip . Any help would be appreciated

mstrens, 2024-11-20:
M10 gps are supposed to use 38400 bps per default.
For a M10 gps, oXs sent some commands at 38400 bps to force the GPS to provide some specific types of messages.

Is it possible to configure the HGLRC m100 mini in such a way that it works at 38400 at power up?

antlerhanger, 2024-11-20:
I will give it a try .. It should be possible in u center2

Thanks
Allen

mstrens, 2024-11-20:
ucenter allows to change the baudrate. The question is : is your gps able to keep a config after a power down/up?

antlerhanger, 2024-11-20:
I dropped it to 38400 and  GPS held it . It now works with OXS ..Thanks so much .. I'm learning ..just slow .
Thanks
Allen

antlerhanger, 2024-11-20:
It now says GPS is detected and has a fix
Thanks again
Allen