mstrens/oXs_on_RP2040 #152: Fix Gyro with MPU6050 and GY63 on SDA, SCL

Eisbaeeer, 2024-12-01:
With version 2.8 the MPU6050 will be detected with GY63 on the same bus. All versions above will not recognize the MPU6050 and GY63 on same bus.

Satcomix, 2024-12-02:
> With version 2.8 the MPU6050 will be detected with GY63 on the same bus. All versions above will not recognize the MPU6050 and GY63 on same bus.

Hello Eisbaeeer,
Here is **Ver. 2.13.1** with single GY-63 MS5611 and single MP6050 at one I2C Bus. Normally i use a GY-86 with all in one sensor board. My next test will be Ver. 3.0
processing cmd

Version = 2.13.1 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =   21  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255    7  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255  255   29 (V1 / V4 = 26, 27, 28, 29)
RGB led . . . . . . . . . =   18  (RGB    = 0, 1, 2, ..., 29)
Logger  . . . . . . . . . =  255  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =    1  (ESC_PIN= 0, 1, 2, ..., 29)
Locator CS  . . . . . . . =  255  (SPI_CS = 0, 1, 2, ..., 29)
        SCK . . . . . . . =  255  (SPI_SCK= 10, 14, 26)
        MOSI  . . . . . . =  255  (SPI_MOSI=11, 15, 27)
        MISO  . . . . . . =  255  (SPI_MISO=8, 12, 24, 28)
Output level High . . . . =  255  (HIGH = 0, 1, 2, ..., 29)
Output level Low  . . . . =  255  (LOW  = 0, 1, 2, ..., 29)

Esc type is BLH (BlHeli)

Protocol is Fbus(Frsky)
CRSF baudrate   = 420000
Logger baudrate = 115200
PWM is generated at = 50 Hz
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 60.000000
**Baro sensor is detected using MS5611 at I2c adress 77**
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Aispeed sensor is detected using MS4525
    Vspeed compensation channel = 7
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox (configured by oXs) :GPS is detected and has a fix
Led color is normal (not inverted)
Failsafe uses predefined values
     Chan 1...4  =  1500  1500   992  1500
     Chan 5...8  =  1500  1500   988  1500
     Chan 9...12 =  1500  1500  1500  1500
     Chan 13...16=  1500  1500  1500  1500
**Acc/Gyro is detected using MP6050**
     Acceleration param: ACC= 0.000000 0.000000 0.000000
                              1.000000 0.000000 0.000000
                              0.000000 1.000000 0.000000
                              0.000000 0.000000 1.000000
     Gyro offsets         X, Y, Z = 0 , 0 , 0
     Orientation          Horizontal is UP(Z+)     Vertical is FRONT(X+)

Gyro is not configured

No sequencers are defined

Config parameters are OK
Press ? + Enter to get help about the commands
processing cmd

Cmd to execute:   FV

GPS Latitude = XX.XX25824 degree
GPS Longitude = X.XXX2952 degree
GPS Groundspeed = 2 cm/s
GPS Heading = 174.890000 degree
GPS Altitude = 5738 cm
GPS Num sat. = 116
GPS Date J M A = 2 12 24 
GPS Time H M S = 7 36 15 
GPS Pdop = 254 
GPS Home bearing = 235 degree
GPS Home distance = 2 m
Volt 4 = 249 mVolt
Vspeed = 0 cm/s
Baro Rel altitude = -1 cm
Pitch = -43.000000 degree
Roll = 22.000000 degree
Ads 1 1 = 119 mVolt
Ads 1 2 = 119 mVolt
Ads 1 3 = 119 mVolt
Ads 1 4 = 119 mVolt
Airspeed = -16 cm/s
Compensated Vspeed = -21 cm/s
Gps cumulative distance = 10
Acc X = -0.706000g
Acc Y = 0.283000g
Acc Z = 0.691000g
Reserve 5 = -4300
Vspeed compensation = 1.10
pwmTop= 20000

Satcomix, 2024-12-02:
**Next test with Ver. 3.0.0-main** and GY-63 MS5611 and MP6050 at one I2C bus.

processing cmd

Version = 3.0.0 
    Function                GPIO  Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =   21  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255    7  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255  255   29 (V1 / V4 = 26, 27, 28, 29)
RGB led . . . . . . . . . =   18  (RGB    = 0, 1, 2, ..., 29)
Logger  . . . . . . . . . =  255  (LOG    = 0, 1, 2, ..., 29)
ESC . . . . . . . . . . . =    1  (ESC_PIN= 0, 1, 2, ..., 29)
Locator CS  . . . . . . . =  255  (SPI_CS = 0, 1, 2, ..., 29)
        SCK . . . . . . . =  255  (SPI_SCK= 10, 14, 26)
        MOSI  . . . . . . =  255  (SPI_MOSI=11, 15, 27)
        MISO  . . . . . . =  255  (SPI_MISO=8, 12, 24, 28)
        BUSY  . . . . . . =    0  (SPI_BUSY=0, 1, 2, ..., 29)
Output level High . . . . =  255  (HIGH = 0, 1, 2, ..., 29)
Output level Low  . . . . =  255  (LOW  = 0, 1, 2, ..., 29)

Esc type is BLH (BlHeli)

Protocol is Fbus(Frsky)
CRSF baudrate   = 420000
Logger baudrate = 115200
PWM is generated at = 50 Hz
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 60.000000
Baro sensor is detected using MS5611 at I2c adress 77
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Aispeed sensor is detected using MS4525
    Vspeed compensation channel = 7
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox (configured by oXs) :GPS is detected and has a fix
Led color is normal (not inverted)
Failsafe uses predefined values
     Chan 1...4  =  1500  1500   992  1500
     Chan 5...8  =  1500  1500   988  1500
     Chan 9...12 =  1500  1500  1500  1500
     Chan 13...16=  1500  1500  1500  1500
Acc/Gyro is detected using MP6050
     Acceleration param: ACC= 0.000000 0.000000 0.000000
                              1.000000 0.000000 0.000000
                              0.000000 1.000000 0.000000
                              0.000000 0.000000 1.000000
     Gyro offsets         X, Y, Z = 0 , 0 , 0
     Orientation          Horizontal is UP(Z+)     Vertical is FRONT(X+)

Gyro is not configured (Rc channel used to select the gyro mode/gain is not defined)

No sequencers are defined

Config parameters are OK
Press ? + Enter to get help about the commands
processing cmd

Cmd to execute:   FV

GPS Latitude = XX.XX25760 degree
GPS Longitude = X.XXX3264 degree
GPS Groundspeed = 5 cm/s
GPS Heading = 32.500000 degree
GPS Altitude = 5164 cm
GPS Num sat. = 120
GPS Date J M A = 2 12 24 
GPS Time H M S = 7 43 19 
GPS Pdop = 133 
GPS Home bearing = 22 degree
GPS Home distance = 2 m
Volt 4 = 249 mVolt
Vspeed = -9 cm/s
Baro Rel altitude = 13 cm
Pitch = -2.000000 degree
Roll = 45.000000 degree
Ads 1 1 = 119 mVolt
Ads 1 2 = 119 mVolt
Ads 1 3 = 119 mVolt
Ads 1 4 = 119 mVolt
Airspeed = 137 cm/s
Compensated Vspeed = -15 cm/s
Gps cumulative distance = 2
Acc X = -0.702000g
Acc Y = 0.279000g
Acc Z = 0.699000g
Reserve 5 = -139
Vspeed compensation = 1.10
pwmTop= 20000

Satcomix, 2024-12-02:
> With version 2.8 the MPU6050 and GY63 is deteced on same bus. All versions above will not recognize the units. EDIT: The 3.0 recognize the I2C devices, if there is no PWM on channels defined.

Hello Eisbaeeer,
Can you please give us some more informations about your hardware. Which RP2040 board???, protocol???, can you please make a copy of your settings(see mine) and with FV command. Thank you.
best regards,
Torsten

Eisbaeeer, 2024-12-02:
> Can you please give us some more informations about your hardware. Which RP2040 board???, protocol???, can you please make a copy of your settings(see mine) and with FV command. Thank you.

For sure. That was not proffessional from my side, sorry. I will post if Im at home.
My setup:
Receiver: Frsky s.bus + s.port. 
PWM output for servos (C1 = Aileron, C2=Elevator, C3=Throttle, C4=Rudder)
Stabi mode with C5.
The RP2040 should work with MPU6050 + GY63 on same bus. MPU6050 should be stabilize the plane and GY63 giving the altitude.
Is this possible?
In version 3.0 I saw all the telemetry (altitude + X-Y axis of MPU6050, ..) After configure the PWM Cx channels, the I2C modules wasn´t recognized. Nothing changend on hardware.
[image]
Lars

Satcomix, 2024-12-02:
Hello Lars,
You want to use your FrSky with S.Port and SBUS 16CH, ok. But we have a problem with the gyro stabilization for your plane.
I almost only have FrSky equipment and work with the FBUS protocol on the oXs RP2040. From version 2.14.x, I think, there is a problem with the SBUS out interface. I also believe that the gyro flight system has not been sufficiently tested. I can provide you with a 2.13.1 version, maybe this will work. When you speak german, you can look at RC Network for Infos.
best regards,
Torsten

oXs_RP2040 Ver.: 2.13.1
[oXs.zip](https://github.com/user-attachments/files/17976191/oXs.zip)

closed 2025-06-12 by Eisbaeeer